olikraus/u8g2 #583: Fonts take too much space

kichibaba, 2018-04-18:
Hi i am trying to use 2 fonts
u8g2_font_logisoso58_tr and u8g2_font_courB24_tr
but whenever i try to compile for the arduino nano it says that 103% global variables used. How can i reduce this?

RobotGB, 2018-04-18:
Do you mind posting your code?

olikraus, 2018-04-18:
Which U8g2 constructor did you use?

olikraus, 2018-04-18:
And of course: 103% global variables means RAM, which has nothing todo with the fonts.

kichibaba, 2018-04-18:
Oh ok it means RAM. Thanks.
 I used U8G2_SH1106_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);
as my constructor

kichibaba, 2018-04-18:
`#include <Arduino.h>
#include <U8g2lib.h>
#include <Wire.h>

U8G2_SH1106_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);

void setup(void) {
  u8g2.begin();
  u8g2.clearBuffer();          
  u8g2.setFont(u8g2_font_logisoso58_tr);  
  u8g2.drawStr(45,64,"V");
  u8g2.drawLine(56,6,35,6);
  u8g2.drawLine(71,6,92,6);
  u8g2.sendBuffer();          
  delay(3000); 
  
  u8g2.clearBuffer();         
  u8g2.setFont(u8g2_font_logisoso58_tr);  
  u8g2.drawStr(10,64,"TVM");
  u8g2.drawLine(58,6,37,6);
  u8g2.drawLine(73,6,94,6); 
  u8g2.sendBuffer();          
  delay(1000); 
}`
Thats my code with an empty loop

olikraus, 2018-04-18:
Not sure what eats up the RAM here, but I guess you need to use the page buffer approach, e.g.
```
U8G2_SH1106_128X64_NONAME_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);
```

kichibaba, 2018-04-18:
Now it doesn't print properly. only the top portion of the text shows

RobotGB, 2018-04-18:
does your code look like this?

```
u8g2.begin();
  do {
    //u8g2.clearBuffer();
    u8g2.setFont(u8g2_font_logisoso58_tr);
    u8g2.drawStr(45, 64, "V");
    u8g2.drawLine(56, 6, 35, 6);
    u8g2.drawLine(71, 6, 92, 6);
    //u8g2.sendBuffer();
    delay(3000);
  } while (u8g2.nextPage());
  do {
    //u8g2.clearBuffer();
    u8g2.setFont(u8g2_font_logisoso58_tr);
    u8g2.drawStr(10, 64, "TVM");
    u8g2.drawLine(58, 6, 37, 6);
    u8g2.drawLine(73, 6, 94, 6);
    //u8g2.sendBuffer();
    delay(1000);
  } while (u8g2.nextPage());
```

kichibaba, 2018-04-18:
no

kichibaba, 2018-04-18:
It works now but how do i clear the page?

RobotGB, 2018-04-18:
what happens when you put the code in the setup loop?
overall it should look like this:
```
#include <Arduino.h>
#include <U8g2lib.h>
#include <Wire.h>

U8G2_SH1106_128X64_NONAME_2_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);

void setup(void) {
  u8g2.begin();
  do {
    //u8g2.clearBuffer();
    u8g2.setFont(u8g2_font_logisoso58_tr);
    u8g2.drawStr(45, 64, "V");
    u8g2.drawLine(56, 6, 35, 6);
    u8g2.drawLine(71, 6, 92, 6);
    //u8g2.sendBuffer();
    delay(3000);
  } while (u8g2.nextPage());
  do {
    //u8g2.clearBuffer();
    u8g2.setFont(u8g2_font_logisoso58_tr);
    u8g2.drawStr(10, 64, "TVM");
    u8g2.drawLine(58, 6, 37, 6);
    u8g2.drawLine(73, 6, 94, 6);
    //u8g2.sendBuffer();
    delay(1000);
  } while (u8g2.nextPage());
}
void loop(void) {

}
```

kichibaba, 2018-04-18:
yes thats what it looks like. So the big V prints slowly(i changed the delay to make it faster and put the 3sec delay after the loop)  but then it doesnt print the TVM it just prints some rectangles at the bottom of the dispay. like its printing the TVM but the V is not being cleared

kichibaba, 2018-04-18:
OK THANKS I GOT IT TO WORK. I just added  `u8g2.clear();` after each loop and it works

kichibaba, 2018-04-18:
THANKS FOR MAKING THIS LIBRARY

kichibaba, 2018-04-19:
one last thing.
if a have a String l = "Hello World";
`u8g2.drawStr(0,38,l);` doest work. However,
`u8g2.setCursor(0,38);`
`u8g2.print(l);` works but if my String is very big it doesnt automatcally bring it to a newline.

`
String l ="TestLongString";
int y = 38;
int count1 = 0;
do {
  u8g2.setFont(u8g2_font_courB12_tr);  
  u8g2.drawStr(0,13,"Hello");
  u8g2.drawStr(0,26,"World");
   u8g2.setFont(u8g2_font_courB12_tr);
  u8g2.setCursor(0, y);
  u8g2.print(l);
    //u8g2.sendBuffer();
    delay(50);
    count1++;
    if(count1>10){
      y+=12;
      }
  } while (u8g2.nextPage());`

olikraus, 2018-04-19:
Yes

closed 2018-04-21 by olikraus