olikraus/u8g2 #586: 96x68 HX1230 speed

mithat, 2018-04-23:
My sample of those cute [96x68 HX1230](https://www.aliexpress.com/item/96x68-inch-HX1230-LCD-screen-better-than-5110-resolution-picture-text-display/32810161178.html) displays in a test setup involving a 3.3V 8MHz Pro Mini clone is incredibly slow. A simple [sketch](https://mithatkonar.com/wiki/doku.php/arduino/displays_for_classic_arduinos/display_96x68_hx1230.ino) with a single large counter yields about 1.2 updates per second with visible updating artifacts. The [equivalent sketch](https://mithatkonar.com/wiki/doku.php/arduino/displays_for_classic_arduinos/display_nokia_5110_u8g2_counter.ino) with a Nokia 5110 display gets about 10 updates per second with imperceptible artifacts.

Do I have a bad sample or is this typical of all these displays? I note that the 96x68 HX1230 runs 3-wire SW SPI and the Nokia 4-wire HW SPI. Are there any library-level changes that might be implemented to speed things up?

olikraus, 2018-04-23:
These are my results with a SSD1306 and 3-Wire SW SPI (page mode, FPS.ino example)
```
U8G2_SSD1306_128X64_NONAME_1_3W_SW_SPI 	Uno			Clip=2.7 Box=2.9  @=1.8 Pix=2.2
```

So I guess, I can confirm your results (maybe you can post the measurements with FPS example from page buffer section). BTW: What is your Arduino Board?

There are actually two problems:
1. There is no optimization for the 3-wire software driver, so all the I/O goes via Arduino digitalWrite(), which is known to be very slow.
2. There is no 3-wire hardware driver, because SPI.h does not support 9-Bit transfers.

On the Uno, there is an optimized version for the SW SPI:
```
U8G2_SSD1306_128X64_NONAME_1_4W_SW_SPI	Uno			Clip=21.7 Box=46.8  @=4.0 Pix=7.1			
```
So I guess, that the 3-Wire SPI could be 8 times faster if the SW I/O would be improved.

The 3-wire hardware driver is much more complicated. Maybe it might be possible to emulate 9 bit transfer via 8 bit SPI, but this would require some analysis. Then still a lot of bit level conversion is required, so I guess this does not make sense. 

So finally the only option seems to be a AVR optimized 3-Wire SW SPI, taken over from the existing 4-Wire SPI. In such a case, only AVR based Boards (Uno, Nano, Mega) would benefit from the optimization.

Does this make sense for your? How much is the demand for having such optimization?
Are there any other Graphics Libs for the HX1230?

mithat, 2018-04-23:
Running _page_buffer > FPS_ using `U8G2_HX1230_96X68_1_3W_SW_SPI` I get:

    draw clip test
    001.6

    clear screen
    001.7

    draw @
    001.1

    draw pixel
    001.4

This is  on an 8MHz Pro Mini clone (3.3V), so I suspect ~2x better results with a 16MHz 'duino. Just for reference, the same test on a Nokia 5110 using `U8G2_PCD8544_84X48_1_4W_HW_SPI` yields

    draw clip test
    021.2

    clear screen
    0077.6

    draw @
    005.5

    draw pixel
    010.5

My need for faster 3W_SW rendering is not at all urgent at the moment as Nokia 5110 displays are still in good supply and comparably sized 128x64 OLED displays are almost as cheap.

How hard would it be to refactor the code to use direct port manipulation rather than digitalWrite when possible?

olikraus, 2018-04-29:
```
  U8G2_SSD1306_128X64_NONAME_1_3W_SW_SPI 	Uno			Clip=2.7 Box=2.9  @=1.8 Pix=2.2		issue 586, before optimization
  U8G2_SSD1306_128X64_NONAME_1_3W_SW_SPI 	Uno			Clip=23.4 Box=40.8  @=4.4 Pix=8.1	issue 586, after optimization
```

olikraus, 2018-04-29:
tested with SSD1306...

olikraus, 2018-04-29:
Can you test this with the HX1230 display? Will it work?

Download U8g2 Beta 2.22.17 from here:
https://github.com/olikraus/U8g2_Arduino/archive/master.zip

Install via "Add Zip Library" menu of your Arduino IDE.

mithat, 2018-04-30:
Muuuch better. Running _page_buffer > FPS_ using `U8G2_HX1230_96X68_1_3W_SW_SPI` with U8g2 Beta 2.22.17 I get:

    draw clip test
    011.5

    clear screen
    023.0

    draw @
    002.7

    draw pixel
    004.8

The clear screen test produced some visible artifacts that I can best describe as vertically scrolling tiger stripes. It looks like the display's innards need about 15ms to complete the state change but electronically it can handle faster updates. In any event, I'm not concerned as the test represents such a corner case.

olikraus, 2018-04-30:
> The clear screen test produced some visible artifacts that I can best describe as vertically scrolling tiger stripes.

This is normal, the LCD has a refresh rate of about 60 FPS, which interfears with the display RAM update.

Thanks for the feedback.

closed 2018-04-30 by olikraus