olikraus/u8g2 #587: Can I print Korean fonts?

skyldh89325, 2018-04-24:
Hello!

I created u8g2 font by converting multilingual font to bdf file from u8g2 library, but I have some output problems and leave a question.

Is there an api that can output multiple languages in the u8g2 function, such as Hangul?

olikraus, 2018-04-24:
If you already have a bdf file, you just need to use bdfconv to convert this to the internal format. Thats it. 
What are your specific problems?

skyldh89325, 2018-04-24:
Hangul is a 2-byte character format, but api functions can only be printed in 1 byte format.

olikraus, 2018-04-24:
U8g2 only supports UTF8. I mean, you can not use the 2-byte char format, but instead you have to use the corresponding UTF8 representation (which is automatically done by the Arduino IDE).

Hangul is already supported by u8g2_font_unifont_korean1 and u8g2_font_unifont_korean2.
Probably not all glyphs are there, but the most popular glyphs should be included.

olikraus, 2018-04-30:
Can i close this?