olikraus/u8g2 #591: device crashing using u8g2 inside class

treii28, 2018-04-29:
I saw there was another closed issue similar but not identical to this but it was closed. I'm trying to use U8g2 inside of another class to handle all my UI rendering in one place. I first ran into the issue in the other post, but solved it with a similar fix. But now, whenever I try to draw anything in the do...while loop, the device crashes and reboots. The code I'm using was working when in a single arduino sketch, but after adding it to a class and making some progress, it suddenly started crashing and I've been trimming it down bit by bit and eventually substituted my code with some of the example code and still get the same result. As soon as I try to draw in the while loop, crash->reboot

I'm trying to create an independent class to handle drawing all of the necessary graphics for an SSD1305 display over either SPI or i2c.

When nothing else would work, I pulled the drawLogo() function from the example sketches and used that code to create a method in my object to call in the do...while loop and it still crashes. (it doesn't seem to matter if I define for mini logo or not)

Any help finding the problem would be appreciated!

relevant code: (file RpuiSSD1306.h)

Code: [Select]

```
#ifndef Arduino_h
#include <Arduino.h>
#endif

#ifndef _U8G2LIB_HH
#include <U8g2lib.h>
#endif

//#define MINI_LOGO

class RPUISSD1306 {
private:
    // private variables
    U8G2 u8g2;

public:
    // constructors
    RPUISSD1306(void) {};

    void initDisplay()
    {
        U8G2_SSD1306_128X64_NONAME_1_4W_SW_SPI u8g2(U8G2_R0, 3, 11, 10, 9, 8);
        u8g2.begin();
    };

    void drawLogo()
    {
        uint8_t mdy = 0;
        if ( u8g2.getDisplayHeight() < 59 )
            mdy = 5;
  

        u8g2.setFontMode(1);	// Transparent
        u8g2.setDrawColor(1);

#ifdef MINI_LOGO

        u8g2.setFontDirection(0);
        u8g2.setFont(u8g2_font_inb16_mf);
        u8g2.drawStr(0, 22, "U");
    
        u8g2.setFontDirection(1);
        u8g2.setFont(u8g2_font_inb19_mn);
        u8g2.drawStr(14,8,"8");
    
        u8g2.setFontDirection(0);
        u8g2.setFont(u8g2_font_inb16_mf);
        u8g2.drawStr(36,22,"g");
        u8g2.drawStr(48,22,"\xb2");
    
        u8g2.drawHLine(2, 25, 34);
        u8g2.drawHLine(3, 26, 34);
        u8g2.drawVLine(32, 22, 12);
        u8g2.drawVLine(33, 23, 12);

#else

        u8g2.setFontDirection(0);
        u8g2.setFont(u8g2_font_inb24_mf);
        u8g2.drawStr(0, 30-mdy, "U");
    
        u8g2.setFontDirection(1);
        u8g2.setFont(u8g2_font_inb30_mn);
        u8g2.drawStr(21,8-mdy,"8");
        
        u8g2.setFontDirection(0);
        u8g2.setFont(u8g2_font_inb24_mf);
        u8g2.drawStr(51,30-mdy,"g");
        u8g2.drawStr(67,30-mdy,"\xb2");
    
        u8g2.drawHLine(2, 35-mdy, 47);
        u8g2.drawHLine(3, 36-mdy, 47);
        u8g2.drawVLine(45, 32-mdy, 12);
        u8g2.drawVLine(46, 33-mdy, 12);
    
#endif

    };

    void drawScreen() {
        u8g2.firstPage();
        do {
            drawLogo();
        } while( u8g2.nextPage() );
    };
}
```

Compiling/building with platformio and mingw gcc/g++ on windows 10 for arduino nano with profile nanoatmega328 btw

olikraus, 2018-04-29:
hmm.. again, I do not know whether I can support.
Can you provide a complete example which crashes. I mean something, which I can execute on my boards.

olikraus, 2018-04-30:
... if not, can I close this issue?