olikraus/u8g2 #594: Zero Padding with Print?

starwolf73, 2018-05-01:
Hi,
First I really want to thank you Oliver for hard work you did for this amazing library and support you provide. 

I'm building a simple clock style display with ESP8266 and Wemos D1 Oled and would like to have a clock display format like this "HH:MM:SS.D". I look at github and your documentation which mention that it can support he Arduino Print function but didn't find any option for Zero padding like other function printf ("%02d", variable). 

Is there a possibility to add this functionality to Print?

Thanks

olikraus, 2018-05-01:
I can not modify "Print". I just call the original Arduino function. So the request for a modified Print has to go to Arduino Group.

But instead you could use an internal u8x8 function:

.print(u8x8_u8toa(hh, 2));
.print(":");
.print(u8x8_u8toa(mm, 2));

starwolf73, 2018-05-02:
Thanks, that work! I understand this is a internal function and If I can suggest, to add it to your documentation on print function. 

Finally even if I use your library for hobby, I want to support your work. Do you have any way to send contribution?

olikraus, 2018-05-02:
I have my e-mail connected to paypal (see source code), but there is no contribution required. I am happy if you like u8g2 :-)

olikraus, 2018-05-02:
Todo: Add u8x8_u8toa to u8g2reference

starwolf73, 2018-05-02:
Contribution sent! Thanks

olikraus, 2018-05-02:
Thanks a lot!

olikraus, 2018-05-02:
I will keep this issue open, until I have updated the documentation.

olikraus, 2018-08-23:
docu updated

closed 2018-08-23 by olikraus