olikraus/u8g2 #595: New font blob format, speed up unicode glyph lookup, backward compatible

Adam5Wu, 2018-05-01:
Goal:
1. Allow font build-time fixed point index generation for unicode glyphs
    Speeds up unicode symbol lookup without costing RAM
2. Introduce versioning to font blob format, open to future extensions

Format:
```
<U8G2_FONT_DATA_STRUCT>

<ASCII Glyph Data>

: start_pos_unicode
0x00 0x01 // WORD, font extension version 1 (extendable up to version 255)
0xAA      // BYTE, size of extension block
0xBB 0xCC // WORD, Number of unicode glyphs
0xDD      // BYTE, number of index entries
0xEE 0xFF 0xKK 0xLL 0xMM 0xNN // Index entry: Symbol WORD 0xEEFF, at offset DWORD 0xKKLLMMNN relative to start_pos_unicode
...       // More indices ...
: end_font_extension_block

0xGG 0xHH // WORD, the first unicode symbol
0xJJ      // BYTE, size of glyph data
...
```

Effectively, this proposal inserts a font extension info block at the original location of the first unique glyph data.
1. The info block contains pre-computed index so unicode lookup can be improved by a order of magnitude or more
2. The info block contains a version identifier and is open to future extension
3. The info block is backward compatible with existing code, because:
    1. The initial three bytes format matches that of unicode glyph data, and
    2. The version field is an impossible unicode symbol, so existing code which does not understand the extension block will simply pass to the next glyph block, which is the actual first symbol, so everything falls into place.

olikraus, 2018-05-01:
I also think, that it is better to extend the font format.

olikraus, 2018-05-01:
I have implemented the new font format. It is already checked in. I have closed the PR.
Also note: The new font format is not backward compatible and a new bdfconv.exe is required.
Changes to u8g2 are minimal...

Can we close this issue and continue the discussion in #596 ?

Adam5Wu, 2018-05-01:
Cool!

closed 2018-05-01 by Adam5Wu