olikraus/u8g2 #597: KS0108 controller no sign of life

Pauperman66, 2018-05-01:
I am using an STM32 board called Black Pill. It works fine and I am upgrading an Arduino sketch -perfectly working with 16x2 LCD (HD44180 based) module- toward Graphic Display, with this powerful Library u8g2. I am using graphic display module Winstar WG12864B with built in NT7107/NT7108 controllers, if I am not wrong one of the KS0108 clones, like Hitachi HD6120x. So I have used ports PA and PB (avoiding PA9 and PA10 used for serial link connections with ST-LINK V2, I am not using USB bootloader) for electrical connections.

I have set up "software pinout nomenclature" reflecting above electrical connections as follows:

U8X8_KS0108_128X64 u8x8(PA0, PA1, PA2, PA3, PA4, PA5, PA6, PA7, /*enable=*/ PB3, /*dc=*/ PB4, /*cs0=*/ PB0, /*cs1=*/ PB1, /*cs2=*/ U8X8_PIN_NONE, /* reset=*/  PB6); 	// Set R/W to low!

and respecting comment i have pulled down R/W line. Wiring is pretty clean and many times double checkced.

Vdd is tied at 5V from the ST-LINK output and backlight pin 19 to 3,3V available from the Black Pill board.

Well using Hello World example I do not see anything. What is wrong?! May I kindly ask you what should I do for discovering why Murphy was not sleeping on today?! Thanks so much.

Pauperman66, 2018-05-01:
Forgot to mention V0 input for contrast clearly is used tied to output of a 10KOhm potentiometer as I did for Character LCD Module. Backlight is working.

Pauperman66, 2018-05-01:
Played with Oscilloscope. All signals from PortA are swinging from 0V to 3,36V. RES is always at 3,36V (during micro hard reset goes up to 5V). R/W is pulled down and no shift. Surprisingly D/C from PB4 (steadily at 3,36V) and E from PB3 (steadily at 0V) ARE NOT CHANGING AT ALL!

olikraus, 2018-05-01:
Your display is a 5V display. It might have problems to read from the 3.3V signals from your controller.

Pauperman66, 2018-05-01:
Thanks for your super-fast comment. I will build up a level shifter: effectively CMOS min threshold for "1" is 3.5V... on W-E I will update this topic.

olikraus, 2018-05-11:
feel free to reopen...

closed 2018-05-11 by olikraus