olikraus/u8g2 #598: Slow nextPage() patch? 440 ms -> 50 ms

ttww, 2018-05-01:
Hi,
I'm using the u8g2 lib version 2.22.18 on an ESP32 (like https://robotzero.one/heltec-wifi-kit-32/) with the  following constructor `U8G2_SSD1306_128X64_NONAME_F_SW_I2C u8g2(U8G2_R0, 15, 4, 16)`

The u8g2.nextPage() call needs around **~440 ms** for updating the little screen.

**_The HW i2c mode is not working without some hacking:_**
Using the constructor `U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0)` and remap the ESP32 pins by hand with `Wire.begin(4, 15)` leads to **~50 ms** update time, which is usable for a good interactive display feedback.

Is it possible to add this "behavior" to the library if an user is using a constructor like `U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, 15, 4)` ?
It would lead to a "working out of the box" experience :-)

Btw: Thanks for the nice little graphic library 👍

olikraus, 2018-05-01:
Hmm.. This should be implemented already, however the syntax is different (https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#ssd1306-128x64_noname):

```
U8G2_SSD1306_128X64_NONAME_1_HW_I2C(rotation, [reset [, clock, data]])
```
In your case:

```
U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, U8X8_PIN_NONE, 15, 4)
```

olikraus, 2018-05-01:
kein Ding...