olikraus/u8g2 #600: A question, why not separate Interface and Lcd Driver?

ghosoft, 2018-05-03:


olikraus, 2018-05-03:
What exactly do you mean?

ghosoft, 2018-05-04:
some what like: 
//cpp code:
U8g2 u(ST7920/*LCD Driver*/, P8080/*interface*/);

olikraus, 2018-05-04:
Ah, ok: not all combinations are valid.

olikraus, 2018-05-11:
but by the way: This is already implemented like this... 
Anyhow... no feedback since 8 days, so I close this...

closed 2018-05-11 by olikraus

ghosoft, 2018-05-14:
I work in a LCD/LCM factory, and my job is making LCD/LCM tester, 
so I will face a lot LCD Resolution X LCD combinations, 
and a lot of the combinations are not in your LIB, so what can i do? 
by the way, how can i make a new LCD driver?

olikraus, 2018-05-14:
> and a lot of the combinations are not in your LIB

Obvdiously. This is open source, freely available software. There is no budget to support all kind of displays. 

> so what can i do?

You can add the missing combinations by yourself. If the display is available for the public, then I could be convinced to add the missing combination, but as I said: There is no budget to do this. I have to do this in my spare time. 

Sometimes, I receive donation for the implementation of a new LCD, which helps to add missing combinations.

> by the way, how can i make a new LCD driver?

Here is a brief description:
https://github.com/olikraus/u8g2/wiki/internal#new-displays