olikraus/u8g2 #601: Compatibility with Adafruit nRF52 board

Moving-Electrons, 2018-05-03:
Hi Oliver,

I've been trying to have the U8G2 library working with Adafruit nRF52 board in a generic 0.96 in OLED display (SSD1306 controller over I2C). I got it working with Adafruit SSD1306 library, but performance is erratic (sometimes I need to reset the board for it to detect the OLED screen). 

I'm trying with your library as I've heard is very stable, but I can't get it to work. This is the constructor I'm using:

`U8G2_SSD1306_128X64_NONAME_F_SW_I2C u8g2(U8G2_R0, /* clock=*/ 26, /* data=*/ 25, /* reset=*/ U8X8_PIN_NONE);   // All Boards without Reset of the Display`

I'm assuming the board's SCL and SDA pins are 26 and 25 respectively (based on the board's pinout). I'm trying to run the default _HelloWorld_ sketch. I just commented out the constructor section as indicated above and revised the _setup_ function as follows:

`
void setup(void) {

  u8g2.setI2CAddress(0x3C);
  u8g2.begin();
}
`

Thanks in advance!

olikraus, 2018-05-03:
Will the HW I2C constructor work?
U8G2_SSD1306_128X64_NONAME_F_SW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);

Moving-Electrons, 2018-05-05:
No, unfortunately none of those worked. I think you meant:

`U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);`

Also, I tried `u8g2.setI2CAddress(0x3C*2);` and also just deleting that line (as it's not on the examples), and didn't work either.

Am I reading correctly [this](https://cdn-learn.adafruit.com/assets/assets/000/046/248/original/microcontrollers_Feather_NRF52_Pinout_v1.2-1.png?1504885794) pinout?. Pin 25 is SDA and pin 26 is SCL, right?

olikraus, 2018-05-05:
> Am I reading correctly this pinout?. Pin 25 is SDA and pin 26 is SCL, right?

looks correct

Moving-Electrons, 2018-05-05:
I can't figure out why such a simple sketch won't work. Any ideas?

olikraus, 2018-05-05:
As I said, for me it works. Maybe you can send a picture of the display and the setup.

Moving-Electrons, 2018-05-05:
Sure, you can see a picture of the setup [here](https://forums.adafruit.com/viewtopic.php?f=57&t=135204). It's actually a post on Adafruit's forums where I'm asking for help on resolving some issues related to their library. As you can see, the OLED is working (kind of) with their library, but it's very unreliable.

I really want to use your library as it seems to work with a wider variety of screens.

Thanks in advance.

olikraus, 2018-05-05:
These breadboards are also not very reliable. Maybe change the position of the OLED.

olikraus, 2018-05-11:
no further comments... closing...

closed 2018-05-11 by olikraus