olikraus/u8g2 #602: Does u8g2 rotation work when used with a MAX7219 array?

al-ang, 2018-05-03:
I'm using u8g2 with four 8x8 MAX7219 combined (1x4 configuration).  Library working well, except I cannot rotate the display. 

When I initialize with 
U8G2_MAX7219_32X8_F_4W_SW_SPI u8g2(U8G2_R0,13,11,10,U8X8_PIN_NONE,U8X8_PIN_NONE)
everything works well,  If I change U8G2_R0 to U8G2_R2, nothing is displayed.  Is the rotation argument functional when using it with MAX7219?

al-ang, 2018-05-03:
I thought that idea might help, but when I use any arg other than 0 in u8g2.setFontDirection(0) the display remains blank. 

I'd like to say this is not that important -- I can rewire my project and flip the display around.  I was mainly wondering if the functionality was there, but I wasn't using it correctly.  This is kinda a limited display :)  I'm happy that the library gives me fonts to choose!

olikraus, 2018-05-03:
Remember to change the reference point when a different font direction is applied

al-ang, 2018-05-04:
oh, yeah, I think that's what happened to me.  I used 0,0 as reference point, so a rotated font would be in a different quadrant, off the screen.  I think it's time to tear things down and re-assemble with longer cables to get the orientation that my simple upper-right quadrant mind can follow.

Thank you for the suggestions! I like your library.

closed 2018-05-04 by al-ang

olikraus, 2018-05-04:
:-)