olikraus/u8g2 #603: Drawing bitmap on heltec WiFi Lora 32 ESP.

dmamalis, 2018-05-04:
I am having some problems trying to print an XBM file on the screen on the Heltec WiFi Lora 32 module.
Any box I define seems to be printed half size with the bottom half not showing on the screen.
I am using: 
```
    #include <U8g2lib.h>
```
The bitmap is:
```
    #define btm_1bit_width 32
    #define btm_1bit_height 32
static unsigned char btm_1bit_1bit_bits[] = {
   0x07, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF,	0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF,
   0x07, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF,	0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF,
   0x07, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF,	0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF,
   0x07, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF,	0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF,
   0x07, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF,	0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF,
   0x07, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF,	0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF,
   0x07, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF,	0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF,
   0x07, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF,	0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF,
  };
```

The constructor is:
```
U8G2_SSD1306_128X64_NONAME_2_HW_I2C display(U8G2_R0,/*rst*/ 16, /*scl*/ 15, /*sda*/ 4);
```
The display sequence is:
```
display.clearBuffer();
display.drawXBMP(0,0,btm_1bit_width,btm_1bit_height,btm_1bit_bits);
display.sendBuffer();
```
I assume that this is a wrong configuration issue. Any tips?

olikraus, 2018-05-04:
You must not use the "2" Constructor with sendBuffer. Instead use the "F" constructor: U8G2_SSD1306_128X64_NONAME_F_HW_I2C

olikraus, 2018-05-04:
oh, hmm... i need to reproduce the problem. Can you send me the complete code which causes the problem?

dmamalis, 2018-05-05:
The following code prints a box of with height=width/2 instead of a square.

```
#define bit_width 32
#define bit_height 32
static unsigned char bit_bits[] = {
   0x07, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF,	0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF,
   0x07, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF,	0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF,
   0x07, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF,	0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF,
   0x07, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF,	0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF,
   0x07, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF,	0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF,
   0x07, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF,	0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF,
   0x07, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF,	0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF,
   0x07, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF,	0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF,
  };

U8G2_SSD1306_128X64_NONAME_F_HW_I2C display(U8G2_R0,/*rst*/ 16, /*scl*/ 15, /*sda*/ 4);

void setup() {
    Serial.begin(115200);
    Serial.println(F("Starting"));

    #ifdef VCC_ENABLE
    // For Pinoccio Scout boards
    pinMode(VCC_ENABLE, OUTPUT);
    digitalWrite(VCC_ENABLE, HIGH);
    delay(1000);
    #endif

	display.begin();
	display.setFont(u8x8_font_victoriamedium8_r);   //Optional, more fonts are available.
}

void draw(void) {
	// graphic commands to redraw the complete screen should be placed here
	display.drawXBMP(0,0,bit_width, bit_height,bit_bits);
}

void loop() {
	display.clearBuffer();
	draw();
	display.sendBuffer();
}
```

dmamalis, 2018-05-05:
I did a bit more testing and I start to realize that my problem might have a different source: it looks like the screen is printing on every second line. For example, printing a 10x10 frame leaves the frame bottom open while a 11x11 frame looks complete.

The same applies in my code causing similar unexpected results.

```
#include <U8g2lib.h>

#define bit_width 32
#define bit_height 32
static unsigned char bit_bits[] = {
   0x80, 0x80, 0x80, 0x80, 0x80,  0x80,  0x80,  0x80, 	0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF,
   0x80, 0x80, 0x80, 0x80, 0x80,  0x80,  0x80,  0x80, 	0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF,
   0x80, 0x80, 0x80, 0x80, 0x80,  0x80,  0x80,  0x80, 	0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF,
   0x80, 0x80, 0x80, 0x80, 0x80,  0x80,  0x80,  0x80, 	0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF,
   0x80, 0x80, 0x80, 0x80, 0x80,  0x80,  0x80,  0x80, 	0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF,
   0x80, 0x80, 0x80, 0x80, 0x80,  0x80,  0x80,  0x80, 	0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF,
   0x80, 0x80, 0x80, 0x80, 0x80,  0x80,  0x80,  0x80, 	0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF,
   0x80, 0x80, 0x80, 0x80, 0x80,  0x80,  0x80,  0x80, 	0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF,
  };

U8G2_SSD1306_128X64_NONAME_F_HW_I2C display(U8G2_R0,/*rst*/ 16, /*scl*/ 15, /*sda*/ 4);

void setup() {
    Serial.begin(115200);
    Serial.println(F("Starting"));

    display.begin();
    display.setFont(u8x8_font_victoriamedium8_r);   //Optional, more fonts are available.
}

void draw(void) {
	// graphic commands to redraw the complete screen should be placed here
	display.drawXBMP(0,0,bit_width, bit_height,bit_bits);
	//display.drawFrame(0, 0, 11, 11);
}

void loop() {
	display.clearBuffer();
	draw();
	display.sendBuffer();
}
```

Could it be the wrong controller?

olikraus, 2018-05-05:
I just tested this on my own display (Uno+SSD1306). All looks fine. Of course I have to replace drawXBMP with drawXBM, but I think it does not matter on your system. Nevertheless, I think it is better to use drawXBM because drawXBMP is only valid for AVR architectures.

dmamalis, 2018-05-05:
What could be the cause of not viewing the odd number lines as I described in the example with the frame then?

olikraus, 2018-05-05:
Broken OLED?

dmamalis, 2018-05-05:
Not really cause the fw that they where shipped with, to which I don't have access now but definitely doesn't use this library, prints every line! :)

olikraus, 2018-05-05:
Maybe you need to use U8G2_SSD1306_128X64_ALT0_F_HW_I2C

dmamalis, 2018-05-05:
The good news is that the ALT0 constructor does seem to solve the "every second line" problem but I am wondering if it is doing that by just skipping a line. 

I am now running the library test file. Running with NONAME skips a line while running with ALT0 images are chopped on the bottom... :|

olikraus, 2018-05-05:
After seeing this pictures, I would still say, this display is broken. Odd and even are driven by different parts of the controller. It happens from time to time, that one half of the controller does not work any more. This will lead to the pictures above.

dmamalis, 2018-05-06:
Well, I managed to load the original firmware and the screen looks fine. I till think that this must be a configuration/constructor issue. Could it be a not-supported OLED?

[image]

olikraus, 2018-05-06:
> Could it be a not-supported OLED?

Hmm indeed. Are the sources for the firmware available online? I mean, is the init sequence for the display available somewhere?

olikraus, 2018-05-11:
no further feedback... closing...

closed 2018-05-11 by olikraus

sanfordjd, 2019-11-25:
actually, i'm seeing similar-ish issues.  
(found this post when searching for a solution).  
In my case, it looks like parts of the image are written "right to left" instead of "left to right".  
I'd assumed this was an "endian-ness" issue, but switching it didn't seem to help.  
I previewed the "Heltec" logo and the logo's bitmap does indeed seem to be written strangely.  
(checked here 'http://javl.github.io/image2cpp/')
My bmp (which should be an oval with the word "test" behind it):

>

>  #define logo_width 128
>    #define logo_height 53
> 
>     const unsigned char testlogo[] = {
>       0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 
>       0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 
>       0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 
>       0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 
>       0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 
>       0xfe, 0x04, 0x11, 0x81, 0xc0, 0x82, 0x30, 0x38, 0x10, 0x46, 0x07, 0xff, 0xff, 0xff, 0xff, 0xff, 
>       0xff, 0xfd, 0xf7, 0xff, 0xff, 0xbe, 0xff, 0xff, 0xf7, 0xdf, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 
>       0xff, 0xfd, 0xf7, 0xff, 0xff, 0xbe, 0xff, 0xff, 0xf7, 0xdf, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 
>   0xff, 0xfc, 0x33, 0xff, 0xff, 0x86, 0x7f, 0xff, 0xf0, 0xcf, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 
>   0xe0, 0x7d, 0xfc, 0xff, 0xff, 0xbf, 0x9f, 0xff, 0xf7, 0xf3, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 
>   0xf9, 0xfd, 0xfc, 0xff, 0xff, 0xbf, 0xdf, 0xff, 0xf7, 0xfb, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 
>   0xf9, 0xc1, 0x88, 0x7f, 0xff, 0xbf, 0xdf, 0xf0, 0x00, 0x00, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 
>   0xf9, 0x98, 0x10, 0xff, 0xff, 0x82, 0x00, 0x00, 0x00, 0x00, 0x00, 0xff, 0xff, 0xff, 0xff, 0xff, 
>   0xf9, 0x81, 0x1c, 0xff, 0xff, 0xfc, 0x00, 0x00, 0x00, 0x00, 0x00, 0x07, 0xff, 0xff, 0xff, 0xff, 
>   0xf9, 0x9f, 0x8c, 0xff, 0xff, 0xc0, 0x00, 0x1f, 0xff, 0xff, 0x00, 0x00, 0x7f, 0xff, 0xff, 0xff, 
>   0xf9, 0x9f, 0xcc, 0xff, 0xfe, 0x00, 0x1f, 0xff, 0xff, 0xff, 0xff, 0x00, 0x07, 0xff, 0xff, 0xff, 
>   0xf9, 0xc1, 0x1e, 0x7f, 0xf0, 0x03, 0xff, 0xff, 0xff, 0xff, 0xff, 0xf8, 0x01, 0xff, 0xff, 0xff, 
>   0xff, 0xff, 0xff, 0xff, 0x80, 0x3f, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0x80, 0x3f, 0xff, 0xff, 
>   0xff, 0xff, 0xff, 0xfe, 0x01, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xf0, 0x0f, 0xff, 0xff, 
>   0xff, 0xff, 0xff, 0xf8, 0x0f, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xfe, 0x03, 0xff, 0xff, 
>   0xff, 0xff, 0xff, 0xf0, 0x3f, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0x81, 0xff, 0xff, 
>   0xff, 0xff, 0xff, 0xc0, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xe0, 0x7f, 0xff, 
>   0xff, 0xff, 0xff, 0x83, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xf8, 0x3f, 0xff, 
>   0xff, 0xff, 0xff, 0x07, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xfc, 0x1f, 0xff, 
>   0xff, 0xff, 0xff, 0x0f, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xfe, 0x1f, 0xff, 
>   0xff, 0xff, 0xfe, 0x1f, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0x0f, 0xff, 
>   0xff, 0xff, 0xfe, 0x3f, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0x8f, 0xff, 
>   0xff, 0xff, 0xfc, 0x3f, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xc7, 0xff, 
>   0xff, 0xff, 0xfc, 0x7f, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xc7, 0xff, 
>   0xff, 0xff, 0xfc, 0x7f, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xc7, 0xff, 
>   0xff, 0xff, 0xfc, 0x7f, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xc7, 0xff, 
>   0xff, 0xff, 0xfc, 0x3f, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xc7, 0xff, 
>   0xff, 0xff, 0xfe, 0x3f, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0x8f, 0xff, 
>   0xff, 0xff, 0xfe, 0x1f, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0x0f, 0xff, 
>   0xff, 0xff, 0xff, 0x0f, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xfe, 0x1f, 0xff, 
>   0xff, 0xff, 0xff, 0x07, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xfc, 0x1f, 0xff, 
>   0xff, 0xff, 0xff, 0x83, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xf8, 0x3f, 0xff, 
>   0xff, 0xff, 0xff, 0xc0, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xe0, 0x7f, 0xff, 
>   0xff, 0xff, 0xff, 0xf0, 0x3f, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0x81, 0xff, 0xff, 
>   0xff, 0xff, 0xff, 0xf8, 0x0f, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xfe, 0x03, 0xff, 0xff, 
>   0xff, 0xff, 0xff, 0xfe, 0x01, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xf8, 0x0f, 0xff, 0xff, 
>   0xff, 0xff, 0xff, 0xff, 0x80, 0x3f, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0x80, 0x3f, 0xff, 0xff, 
>   0xff, 0xff, 0xff, 0xff, 0xf0, 0x03, 0xff, 0xff, 0xff, 0xff, 0xff, 0xf8, 0x00, 0xff, 0xff, 0xff, 
>   0xff, 0xff, 0xff, 0xff, 0xfc, 0x00, 0x1f, 0xff, 0xff, 0xff, 0xff, 0x00, 0x07, 0xff, 0xff, 0xff, 
>   0xff, 0xff, 0xff, 0xff, 0xff, 0xc0, 0x00, 0x1f, 0xff, 0xff, 0x80, 0x00, 0x3f, 0xff, 0xff, 0xff, 
>   0xff, 0xff, 0xff, 0xff, 0xff, 0xf8, 0x00, 0x00, 0x00, 0x00, 0x00, 0x03, 0xff, 0xff, 0xff, 0xff, 
>   0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xe0, 0x00, 0x00, 0x00, 0x00, 0x7f, 0xff, 0xff, 0xff, 0xff, 
>   0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xe0, 0x00, 0x00, 0x7f, 0xff, 0xff, 0xff, 0xff, 0xff, 
[... cut ...]
> 	0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 
> 	0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 
> 	0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 
> 	0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 
> 	0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff
> };
>

olikraus, 2019-11-26:
strange...

sanfordjd, 2019-11-26:
yuuuup :)  
have you seen this before?

olikraus, 2019-11-26:
It would be better to create a separate issue. Maybe add pictures of your problem. Can the   problem shown with a standard hello world example?

jamrogers, 2021-06-28:
I solved this after a lot of reading - the file should not be a bitmap it should be a XBM file. Use gimp to convert your bmp (width 128 x height 64) or you can convert using this website https://convertio.co/fr/png-xbm/.
Once you have saved as XBM - look inside the file you will see the width and height in the first 2 rows as below.

`#define 2486c42fc3f14af5d7643dbfec9a4a7cxeU8wE1AEb5PAlZu_width 128
#define 2486c42fc3f14af5d7643dbfec9a4a7cxeU8wE1AEb5PAlZu_height 64
static char 2486c42fc3f14af5d7643dbfec9a4a7cxeU8wE1AEb5PAlZu_bits[] = {
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x60, 0x00, 0x00, 0x00, 0x00, 
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x68, 
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
  0x00, 0x00, 0x00, 0xF8, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x78, 0xF0, 0x00, 0x00, 0x00, 0x00, 
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x78, 0xC1, 
  0x06, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
  0x00, 0x00, 0x7D, 0x40, 0x02, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
  0x00, 0x00, 0x00, 0x00, 0x80, 0x00, 0x7B, 0x40, 0x00, 0x00, 0x00, 0x00, 
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x80, 0x01, 0x3F, 0x00, 
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
  0x80, 0x81, 0x2F, 0x00, 0x40, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
  0x00, 0x00, 0x00, 0x00, 0xC0, 0x03, 0x0F, 0x00, 0x40, 0x00, 0x00, 0x00, 
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x05, 0x07, 0x00, 
  0xF0, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
  0x40, 0x00, 0x03, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
  0x00, 0x00, 0x00, 0x00, 0xC0, 0x00, 0x01, 0x00, 0x00, 0x00, 0x00, 0x01, 
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x80, 0x01, 0x00, 0x00, 
  0x00, 0x10, 0x04, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
  0x80, 0x13, 0x00, 0x00, 0x00, 0x00, 0x81, 0x01, 0x00, 0x00, 0x00, 0x00, 
  0x00, 0x00, 0x00, 0x00, 0x80, 0x0F, 0x00, 0x00, 0x00, 0xA0, 0x80, 0x03, 
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x7D, 0x00, 0x00, 
  0x00, 0xA0, 0x41, 0x03, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
  0x80, 0xF9, 0x00, 0x00, 0x00, 0xE0, 0xC1, 0x03, 0x00, 0x00, 0x00, 0x00, 
  0x00, 0x00, 0x00, 0x00, 0x80, 0xFB, 0x01, 0x00, 0x00, 0x80, 0xE1, 0x03, 
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xE7, 0x01, 0x00, 
  0x00, 0x80, 0x81, 0x03, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
  0x80, 0x7F, 0x00, 0x00, 0x00, 0x10, 0x86, 0x03, 0x00, 0x00, 0x00, 0x00, 
  0x00, 0x00, 0x00, 0x00, 0x00, 0x3E, 0x00, 0x00, 0x00, 0xE8, 0x10, 0x03, 
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x1E, 0x00, 0x00, 
  0x00, 0x90, 0x49, 0x05, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
  0x00, 0xDE, 0x07, 0x80, 0x7F, 0x66, 0xE0, 0x06, 0x00, 0x00, 0x00, 0x00, 
  0x00, 0x00, 0x00, 0x00, 0x80, 0x7F, 0x1C, 0xC0, 0xC0, 0x38, 0x08, 0x03, 
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xC0, 0x3D, 0x00, 0x00, 
  0x80, 0xB7, 0x9C, 0x03, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
  0xC0, 0x3F, 0x30, 0x20, 0x04, 0xD7, 0x63, 0x05, 0x00, 0x00, 0x00, 0x00, 
  0x00, 0x00, 0x00, 0x00, 0x00, 0x7F, 0x40, 0x00, 0xD0, 0xCF, 0x3A, 0x07, 
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x40, 0xFC, 0x1F, 0x00, 
  0x61, 0x96, 0xE0, 0x06, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
  0x40, 0xFB, 0x2F, 0xC0, 0xBF, 0x8F, 0xE7, 0x02, 0x00, 0x00, 0x00, 0x00, 
  0x00, 0x00, 0x00, 0x00, 0x40, 0xCB, 0x01, 0x00, 0x7E, 0xD0, 0xE8, 0x03, 
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x80, 0x16, 0x00, 0x00, 
  0x30, 0xC8, 0xD2, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
  0x00, 0x05, 0x00, 0x00, 0x00, 0xE8, 0xC5, 0x01, 0x00, 0x00, 0x00, 0x00, 
  0x00, 0x00, 0x00, 0x00, 0x80, 0x09, 0x00, 0x00, 0x00, 0xDC, 0xE5, 0x03, 
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x80, 0x01, 0x00, 0x00, 
  0x00, 0xE4, 0xF9, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
  0x80, 0x01, 0x00, 0x00, 0x00, 0x74, 0xF8, 0x01, 0x00, 0x00, 0x00, 0x00, 
  0x00, 0x00, 0x00, 0x00, 0x40, 0x10, 0x00, 0x00, 0x00, 0xFE, 0xF8, 0x00, 
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x10, 0x40, 0x04, 
  0x00, 0xBF, 0xF9, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
  0x00, 0x10, 0xE0, 0x3F, 0x00, 0xFF, 0xFD, 0x03, 0x00, 0x00, 0x00, 0x00, 
  0x00, 0x00, 0x00, 0x00, 0x00, 0x3C, 0xC0, 0x0B, 0xC0, 0xBF, 0xFB, 0x03, 
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x7C, 0x00, 0x00, 
  0x40, 0xFF, 0xF8, 0x03, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
  0x00, 0x3C, 0x00, 0x00, 0x40, 0xFF, 0xFF, 0x01, 0x00, 0x00, 0x00, 0x00, 
  0x00, 0x00, 0x00, 0x00, 0x00, 0x6C, 0xC0, 0x12, 0xA0, 0xFF, 0xFF, 0x00, 
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x7C, 0xF0, 0x7F, 
  0x80, 0xFF, 0x7F, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
  0x00, 0x70, 0x78, 0xF0, 0x40, 0xFF, 0x3F, 0x00, 0x00, 0x00, 0x00, 0x00, 
[... 119 lines cut ...]
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xFC, 0xFF, 
[... cut ...]
  0x1F, 0x02, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
  0x00, 0x00, 0xF8, 0xFF, 0x07, 0x02, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xF4, 0xFF, 0x03, 0x01, 0x00, 0x00, 
  0x00, 0x00, 0x00, 0x00, };

void setup(void) {
  u8g2.begin();
}

void loop(void) { 
  u8g2.clearBuffer();
  //uint8_t frame_size = 28;
  u8g2.drawXBMP( 0, 0, u8g_logo_width, u8g_logo_height, u8g_logo_bits);
  u8g2.sendBuffer();
  
  // delay between each page
  delay(100);

}
`