olikraus/u8g2 #604: Support for vietnamese unicode font

emptyhb, 2018-05-06:
I'm planing using u8g2 for display content in Vietnamese, but i dont know how to add new font? or can you buid one of theme? example in windows os, Arial and Tahoma font

olikraus, 2018-05-06:
First of all, I need to know the unicode ranges, which make up Vietnamese language. 
Is this the full set of characters: http://vietunicode.sourceforge.net/charset/ ?
This would mean, the ranges are:
0x0020...0x0080
0x00c0...0x00ff
0x0102
0x0103
0x0110
0x0111
0x0128
0x0129
0x0168
0x0169
0x01a0
0x01a1
0x01af
0x01b0
0x1ea0...0x1ef9

A more extended set is mentioned here: http://vietunicode.sourceforge.net/charset/vietalphabet.html
It claims, that this should contain the following unicode ranges:
Basic Latin {U+0000..U+007F}
Latin-1 Supplement {U+0080..U+00FF}
Latin Extended-A, -B {U+0100..U+024F}
Latin Extended Additional {U+1E00..U+1EFF}
Combining Diacritical Marks {U+0300..U+036F}

So, I could create two different fonts, let me call them vietnamese1 (first, smaller version) and vietnamese2 (second, more extended version).
http://unifoundry.com/png/plane00/uni0001.png
http://unifoundry.com/png/plane00/uni0002.png (partly)
http://unifoundry.com/png/plane00/uni0003.png (partly)
and
http://unifoundry.com/png/plane00/uni001E.png

correct? Please confirm.

Both versions will be based on the unifont font (http://unifoundry.com/unifont.html).
For unifont_t_vietnamese2 I would include:

emptyhb, 2018-05-06:
Yes! that correct. I'm verry with rapid answer of your! Thank you

olikraus, 2018-05-06:
Ok, I have created the following two fonts:
[image]
and
[image]

Hope this will be ok for you. I have not investigated other fonts, whether Vietnamese glyphs are included, so unless there is some other information, there will be only the unifont font.

emptyhb, 2018-05-06:
After apply new libray and tested. I can report only "đ" and "Đ" display correct.
"á Á" still display as: "a A"
"ô Ô" => "o O"
"ơ Ơ" => "o O"

I also folowing tutorial, conver ttf font to bdf and convert bdf to .c font then apply add to u8g2_fonts.c and u8g2.h have same result!

emptyhb, 2018-05-06:
Here is info of Vietnamese font in windows OS

https://image.ibb.co/gs3Ew7/Capture.png

With this font i'm doing convert with folowing command:

> ./otf2bdf -r 72 -p 8 UVNBaiSau_R.TTF -o unvbaisau_8.bdf

> ./bdfconv -f 1 -m '32-128, $c0-$c1' -n unvbaisau_8 -o unvbaisau_8.c unvbaisau_8.bdf

I'm testing with char "Á" and "À" at Unicode position 00C0 and 00C1

emptyhb, 2018-05-07:
Can you create sketch and print out "Xin Chào" // Vietnamese Hello. Maybe I forgot something

olikraus, 2018-05-07:
Will U8G2_R0 Work better?

emptyhb, 2018-05-07:
With U8G2_R0 my display display Mirror, and charactor still same above (in mirror)

olikraus, 2018-05-07:
What about other chars? Can you send a picture?

emptyhb, 2018-05-07:
Here is result in Arduino Uno

[image]

olikraus, 2018-05-07:
indeed strange.
Will what about drawing this to 0,60 or 0,58? Will the accent appear?

emptyhb, 2018-05-07:
Here is when using:   u8g2.drawUTF8(0,50,"Xin Chào Á À đ Đ"); 

[image]

olikraus, 2018-05-07:
Ok, I give up. I have no idea. For me it is working. So finally I need to be able to reproduce for further analysis.

emptyhb, 2018-05-07:
[image]

Here is my "Thank you" from Vietnamese

Problem: with Arduino IDE 1.9.0 beta, i'm trying new feauture of IDE is auto codecomple. Going back to version 1.8.5 and issue solved!

olikraus, 2018-05-08:
Wow, thanks for the feedback.

closed 2018-05-11 by olikraus