olikraus/u8g2 #605: u8g2 with Greek characters support?

mikekgr, 2018-05-06:
Dear @olikraus ,
I am not sure if the current release of your dreamed library, support Greek language / characters?
Please tell me and if it does, how can use Greek.

Thanks and Best Regards,
Mike Kranidis

olikraus, 2018-05-06:
There are several greek fonts available:

u8g2_font_10x20_t_greek  https://github.com/olikraus/u8g2/wiki/fntgrpx11
u8g2_font_cu12_t_greek https://github.com/olikraus/u8g2/wiki/fntgrpcu12
u8g2_font_unifont_t_greek  https://github.com/olikraus/u8g2/wiki/fntgrpunifont
Several other greek fonts are available here (look for _all fonts):
https://github.com/olikraus/u8g2/wiki/fntgrpnbp

mikekgr, 2018-05-06:
Dear @olikraus ,
sorry for my ignorance about this.
Please close the issue.
Thanks

olikraus, 2018-05-06:
:)

closed 2018-05-06 by olikraus