olikraus/u8g2 #606: u8log print "\n" does not scroll

makbas, 2018-05-06:
Hi,
Even after setting u8x8log.setRedrawMode(0); the print("\n") command does not make it scroll.
In a loop It writes from top row down to bottom, and starts writing on the most bottom line over and over with new values. The above lines stops the same.

Any help?

Thanks

olikraus, 2018-05-06:
Can you provide the code?

makbas, 2018-05-06:
`#include <Arduino.h>
#include <U8x8lib.h>

// U8x8 constructor for OLED display
U8X8_SSD1306_128X64_NONAME_SW_I2C OLED(22, 21);

// Create a U8x8log object
U8X8LOG DEBUG; 

// Define the dimension of the U8x8log window
#define U8LOG_WIDTH 16
#define U8LOG_HEIGHT 8

// Allocate static memory for the U8x8log window
uint8_t u8log_buffer[U8LOG_WIDTH*U8LOG_HEIGHT];

/* u8x8.begin() is required and will sent the setup/init sequence to the display */
void setup(void)
{
    OLED.begin();
    OLED.setFont(u8x8_font_pressstart2p_f);

    // Start U8x8log, connect to U8x8, set the dimension and assign the static memory
    DEBUG.begin(OLED, U8LOG_WIDTH, U8LOG_HEIGHT, u8log_buffer);
  
    // Set the U8x8log redraw mode
    DEBUG.setRedrawMode(0);    // 0: Update screen with newline, 1: Update screen for every char  
    
}

void loop(void)
{

    DEBUG.print(millis());
    DEBUG.print("\n");

    delay(100);
    
}`

olikraus, 2018-05-06:
ok, I see the problem. 
It works fine in DEBUG.setRedrawMode(1); but does not work with DEBUG.setRedrawMode(0);

makbas, 2018-05-06:
Thanks for your prompt reply, have a nice one...

olikraus, 2018-05-06:
I leave it open, until the bug is fixed... I just created a test file for now.

olikraus, 2018-05-06:
I think I have fixed the problem. Please download beta 2.23.2 from here:
https://github.com/olikraus/U8g2_Arduino/archive/master.zip

Add the library via the "add zip library" menu of your Arduino IDE.

makbas, 2018-05-06:
Lighting fast solution!

Works great, thank you.

olikraus, 2018-05-06:
:-)

closed 2018-05-11 by olikraus