olikraus/u8g2 #607: Bug in E-Ink Display Waveshare 1.54

bobybobybob, 2018-05-06:
Hello Codemaster,

i use u8g2 for a Day Calendar and every new day the display refresh new and after that the Arduino go deep sleep and wake up after get interrupt signal from DS3231 RTC Alarm. 

It work perfect every day and it is done in some seconds but sometime it look like it is prision in the do while loop and only get finish in about 5 Minutes. The Problem is very rare and not reproduceable.  

Is there a little bug for refresh?

olikraus, 2018-05-07:
I have not touched e-ink code for a while. Actually I can not imagine that there is something which causes such delay.

bobybobybob, 2018-05-07:
It refresh and refresh and refresh total 5 Minutes long. Dont know what u8g2 need if display talk "yes now is ok" Most time it only take 2-3 Seconds.

bobybobybob, 2018-05-07:
I dont know. I am not the Codemaster :-)  Sometime when it hapen (and that is very rare... maybe 1 time in 30 actions) ( 1 Time = 1 day) it seem to help touching the breadboard cable.

olikraus, 2018-05-07:
Well, I can say that such a problem never happend with u8g2 so far...

olikraus, 2018-06-17:
no further comment, closing...

closed 2018-06-17 by olikraus

bobybobybob, 2018-06-17:
Sorry, cant write more. Only can describe what i see