olikraus/u8g2 #608: € symbol

pppeixoto, 2018-05-07:
Hello

I couldn't find any font with the symbol €, can this be included manually ? I'm using the font u8g2 font fub20_tf, but i can use other same size.

Char | Dec | Hex
€ | 8364 | 20AC

Thanks

olikraus, 2018-05-07:
The euro symbol is available in the following fonts:
u8g2_font_6x12_t_symbols
u8g2_font_6x12_m_symbols
u8g2_font_7x13_t_symbols
u8g2_font_7x13_m_symbols
u8g2_font_8x13_t_symbols
u8g2_font_8x13_m_symbols
u8g2_font_9x15_t_symbols
u8g2_font_9x15_m_symbols
u8g2_font_tom_thumb_4x6_t_all
u8g2_font_t0_11_t_all
u8g2_font_samim_10_t_all
u8g2_font_samim_12_t_all
u8g2_font_samim_14_t_all
u8g2_font_samim_16_t_all
u8g2_font_samim_fd_10_t_all
u8g2_font_samim_fd_12_t_all
u8g2_font_samim_fd_14_t_all
u8g2_font_samim_fd_16_t_all
u8g2_font_iranian_sans_8_t_all
u8g2_font_iranian_sans_10_t_all
u8g2_font_iranian_sans_12_t_all
u8g2_font_iranian_sans_14_t_all
u8g2_font_iranian_sans_16_t_all
u8g2_font_mozart_nbp_t_all
u8g2_font_mozart_nbp_h_all
u8g2_font_glasstown_nbp_t_all
u8g2_font_shylock_nbp_t_all
u8g2_font_roentgen_nbp_t_all
u8g2_font_roentgen_nbp_h_all
u8g2_font_calibration_gothic_nbp_t_all
u8g2_font_mercutio_basic_nbp_t_all
u8g2_font_mercutio_sc_nbp_t_all
u8g2_font_nine_by_five_nbp_t_all
u8g2_font_rosencrantz_nbp_t_all
u8g2_font_guildenstern_nbp_t_all
u8g2_font_cu12_t_symbols
u8g2_font_cu12_h_symbols
u8g2_font_unifont_t_symbols
u8g2_font_unifont_h_symbols
u8g2_font_pxplustandynewtv_t_all
u8g2_font_pxplustandynewtv_8_all
u8g2_font_pxplusibmvga9_t_all
u8g2_font_pxplusibmvga9_m_all
u8g2_font_pxplusibmvga8_t_all
u8g2_font_pxplusibmvga8_m_all

... not sure whether this helps...

pppeixoto, 2018-05-07:
Thanks for the reply.
Can i get any of this fonts size 20 ?

olikraus, 2018-05-07:
U8g2 only has pixel fonts... If fub20 contains the euro sign, then of course you could create your own font with the euro sign... let me check.

olikraus, 2018-05-07:
FreeUniversal-Bold does not include the euro symbol, so it can not be generated for fub20.
But it is available in logososo...
[image]

Code: 
```
/*
  Fontname: -FreeType-Logisoso-Medium-R-Normal--30-300-72-72-P-38-ISO10646-1
  Copyright: Created by Mathieu Gabiot with FontForge 2.0 (http://fontforge.sf.net) - Brussels - 2009
  Glyphs: 97/527
  BBX Build Mode: 0
*/
const uint8_t u8g2_font_logisoso20_tx[2261] U8G2_FONT_SECTION("u8g2_font_logisoso20_tx") = 
  "a\0\4\4\5\5\4\6\6\24\37\0\373\25\373\25\0\2\347\5\313\10\225 \6\0 \350\2!\14\243"
  ".\210\302\17\14\205(\1\0\42\12\206p\354\2\213 \261\0#/\253&\330\316\220\61C\306\14\31\63"
  "d\314\210A#\6\215\30\361\340\301\220\21\203F\14y\360`\304\220\61C\4\215\30\64b\314\220\61C"
  "\306\0$ Kg\327\322\300\311\14%\31\63b\224\205#\311\25\244\303\221\246L\211 \63\4\221\271\201"
  "\23\1%(\253&\330\306 \21EF\10\11\62\242\210\230!\3\5N(pB\201\323\15\34(d\314"
  "\20\42#FHCb\314\20\0&#\253&\330\16\261\63H\6\215\30\64b\344\300\201\344\210\231\62\65"
  "b\204\211\21F\262AC$\15\242\23\0'\12\203h\214F\211 !\0(\27\245\62\350N\230!C"
  "\204\14\21#d\376F\314\220\61C\306\204\0)\31\246\62\350J\230Ac\6\11\32\63H\242\61\263\21"
  "\63\223\61CF\205\1*\25(\355\353\216\230 BF\210PBD\205\210!B\302\210\1+\16\10-"
  "\351\312\250i\36$\31\65\6\0,\12\203\354\207F\210\42!\0-\10h\354\351\302\203\4.\11c,"
  "\210F\210\22\0/\30\252*\350\336\260q\223\215\233l\334d\343\306\211\233N\334d\343\246\3\60\25\253"
  "&\330N\251\63)\6\215\30e\377\217\206\244\71U\6\0\61\14\245\66\330\312\24/\206\314\377\17\62\34"
  "\253*\350N\251\63)\6\215\30eU\251q\344\246#\67\35\271\11\307=x\60\0\63\34\253*\350\302"
  "\7\343&\34\67\216\334\300b\6G\16\234\13A#J\20Ad\6\0\64\34\253&\350\222\300\31\12\234"
  "\335\300)\6\11\31\63d\314\220\61\17\36\214\32\70\33\0\65\33\253&\330\302\7\6g\210&\211\212A"
  "#F\15\234\241)CC\322\234*\3\0\66\35\253&\330N\251\63)\6\215\30epB\64IT\14"
  "\32\61\312\36\15Is\252\14\0\67\30\253&\330\302\7\206F\14\32\70n\340\354\6\316n\340\354\6N"
  "\5\0\70\36\253*\350N\251\63)\6\215\30e\253\63C\322\234I\61h\304(;\32\222\346\24!\0"
  "\71\33\253&\330N\251\63)\6\215\30e\217Hh\203pBS\206\206\244\71U\6\0:\16\203\355\210"
  "\202\210\21\341A\205(\1\0;\17\243\255\210\202\210\21\341A\205(\22\2\0<\21\207\61\351\226\240\61"
  "C\250\21%\212\20\251\251\4=\13\11\351\330\302\3\366`\36\60>\21\207\61\351B\60Q\244f\64f"
  "\22\62\203\202\1\77\33\253&\330N\251\63)\6\215\30\65p\272q\304\250\33\70\357A\205\34(\12\0"
  "@>\63\347V\233Y\205\255\310\221!\71f\214\221\21c\316\210\30sF\304\220i\214Lcd\32#"
  "\323\30\231\306\310\64F\246\61\62\215\221i\314\260\30\303d\14\11B\344A\220\7\321\216\245\32\0A\42"
  "\252*\350\316\270q\304\250\252\225\210A\42\306\214\30\63D\314\20\61C\206`\221b\220\210A\245J\15"
  "B&\253&\330\302!\64I\6\215\30\64b\224\210A#\6\215H\222\311\30\22\203F\214\262\321\210A"
  "#\222\240\71\4\0C\27\253&\330\16\261\63)\6\215\30ep\376W\206\206\244\71U\6\0D\25\253"
  "&\330\2\232$*\6\215\30e\377\217F\250H\202\6\0E\22\253&\330\302\7\6\347!\32l\6\316"
  "\303\7\17\6F\20\253&\330\302\7\6\347!\32l\6\316\37\2G\27\253&\330\16\261\63)\6\215\30"
  "ep\236\344\312\216H(I\203\0H\17\253&\330\302(\373\253\7\17R\331_\15I\10\243.\210\302"
  "\37\14J\20\253*\350\342\374\377\320\224\241!iN\225\1K!\252*\350\302 \63gF\14\231\213\61"
  "g\14\225*E\254T)CgF\214\31\62\67#\6\31\32L\15\253&\330\302\300\371\377\207*\36\34"
  "M\35\253&\330\302(C\207\320 Q\241\342\301\27&F\230\20bB\210\221 \246\354\325\0N\32\253"
  "&\330\302(S\207.B\203&I&&F\230\30a$\67\30\335j\0O\25\253&\330N\251\63)"
  "\6\215\30e\377\217\206\244\71U\6\0P\33\253&\330\302!\64I\6\215\30\64b\224\215F\214!\221"
  "\4M\261\201\363C\0Q\25\253&\330N\251\63)\6\215\30e\377\217\206\244At\2\0R)\253&"
  "\330\302!\64I\6\215\30\64b\224\215F\214!\221\4\15\232!c\306\210\231d\314\220AB\6\215\30"
  "\64b\224\10\0S\32\253&\330N\251\63)\6\215\30ep \275\234\241)CC\322\234*\3\0T"
  "\15\253&\350\302\7\203\6\316\377\77\2U\20\253&\330\302(\373\377GC\322\234*\3\0V \252*"
  "\350\202\250R\206\14\211\30$D\314\20\61C\346\211 \21\203D\14\262\250\30\275\23\4\0W\60\253&"
  "\330\302\240C#\6\215\30\64b\320\210A#\6\215\30\22b\210\220\20CD\210\30\42\302\210\10#y"
  "\202\206\304\30\22c\206\214\31\62\4\0X)\252*\350\202(Cf\206\210\31\62\23\61#\6\211\30T"
  "\252\30\251\252D\214\31\61f\210\230!C\304\214\30$bP\251\1Y!\252*\350\202(C\42\306\214"
  "\30\63D\214\230\21cF\214\61U\25\271q\342\246\23\67\331\270\351\0Z\34\253&\350\302\7\343\310\215"
  "#\67\216\334\70r\343\310\215#\67\216\334\70r\17\36\14[\15\246\62\350\302\3\63\363\377\315\203\1\134"
  "\31\252*\350\302\270\201\343&\34\67\341\270\11\307M\70n\240\300q\3\5\16]\15\246\62\330\302\203\61"
  "\363\377\315\3\3^\13\210\254\355\312\240\62\42&\31_\10N\240\347\302\203\6`\12\205\60\355\302\230\61"
  "\202\4a \14&\350N\261CiD\11\31\65b\324\310AHT\254\30\65b\324\210Q#X\254)"
  "\42\0b\27\253&\330\302\300\71D\223D\305\240\21\243\354\217F\250H\202\6\0c\26\13&\330N\251"
  "\63)\6\215\30ep\276\62\64$\315\251\62\0d\22\253&\330\342\274A\222\242\321)\373#\22J\322"
  " e\30\13&\330N\251\63)\6\215\30e\352\301\203\204\63\32\222\346T\31\0f\25\252&\330\22\251"
  "\63G\306\10\31\67\253\24\331\214\233\177\5\0g\37\253\346\326\16\222\24\215F\214\262\352\314\220\64\207\210"
  "\15\34i(\205\212Q\246\232\244\71\2\0h\23\253&\330\302\300\71D\223D\305\240\21\243\354\177\65\0"
  "i\20\244*\210\202\220!AG\314\377\5\211!\2j\24H\357\326\226\260a\341\301\215\232\377_\225Q"
  "q\206\10\0k!\252&\330\302\270yd\315\210\61#\206L\61f\304\30C\206\316\234!\61d\314\210"
[... cut ...]
  "\26'\63\347\226\30J\6\315\67\23\211\42\64j\320<\242hX\0|\10cw\327\302\77\60}\31'"
  "\63\347\202\250A\204F\15\232\257H\211!\62h\276\31CfP\60\0~\16\213\244\351\206\230!+\324"
  "\10\12\1\0\177\6\0 \350\2\0\0\0\4\377\377 \254#\255&\10\33\71Ti\6\215\31%f\350"
  "\300%JV\15L\224F\331\320\251\304\14\32\224\352\34\31\0\0";
```

reference: 

```
otf2bdf/otf2bdf -a -r 72 -p 30 -o tmp.bdf ../ttf/Logisoso.ttf
bdfconv -v -f 1 -m '32-127,8364' tmp.bdf -o u8g2_font_logisoso20_tx.c  -n u8g2_font_logisoso20_tx -d ../bdf/helvB18.bdf 
```

pppeixoto, 2018-05-07:
Thanks for your help, this should work.

olikraus, 2018-05-07:
ok, but remember to use u8g2 beta 2.23.2: https://github.com/olikraus/U8g2_Arduino

closed 2018-05-11 by olikraus

pppeixoto, 2018-05-24:
Hello

I'm sorry, for newbie question, how can i use this code to add the font?
Any guide how to do it ?
Thanks

olikraus, 2018-05-24:
> I'm sorry, for newbie question, how can i use this code to add the font?

Just mark the c array and copy the same to your .ino (or .c or .cpp) file. Then use "u8g2_font_logisoso20_tx" as an argument for u8g2.setFont():

```
u8g2.setFont(u8g2_font_logisoso20_tx);
```

pppeixoto, 2018-05-25:
Hello
Can i get the font 8g2 font fub20_tf in a format compatible with Fony software ?
I donwnload the font file in the site, but can't find a way to convert it.

olikraus, 2018-05-25:
You have to work with the intermediate .bdf file, which was generated in this way (see above):
```
otf2bdf/otf2bdf -a -r 72 -p 30 -o tmp.bdf ../ttf/Logisoso.ttf
```

otf2bdf is available here: https://www.math.nmsu.edu/~mleisher/Software/otf2bdf/

pppeixoto, 2018-06-29:
hello

Another update on this, i was able to export one font with € characeter using otf2bdf, but is missing other characters like £. Is this related with th option for the otf2bdf comand ?
Thanks

MMartinezV, 2018-08-08:
Hi,

Following your instructions I've managed to generate many sizes of logisoso to use the € symbol on an e-ink 2.13" display using U8g2_for_Adafruit_GFX.

After generating it I simply add it to u8g2_fonts.c and the header to U8g2_for_Adafruit_GFX.h. 

I've found two problems doing it:

1) If I use the command to generate the logisoso20+€ as described here (using -m "32-127,8364") it doesn't work for me. Otherwise, if I use (-m "32-8364") then I'm able to use the € symbol. It is all right as it works but this way the c array is bigger so I would prefer to select less symbols and keep the array smaller.

The command I've used is:

`otf2bdf/otf2bdf -a -r 72 -p 30 -o tmp.bdf ./ttf/Logisoso.ttf`
`bdfconv/bdfconv -v -f 1 -m '32-8364' tmp.bdf -o u8g2_font_logisoso20_tx.c  -n u8g2_font_logisoso20_tx -d ./bdf/helvB18.bdf`

2) I've been able to generate logisoso 20pt, 24 and 28 and they work fine, but I have problems with 30, 32, 34 and 42pt. After generating them, when I try to use these fonts on an ESP32 device it hangs. I've tought that the problem could be the size of the array but I'm able to use a standard (without € symbol) u8g2_font_logisoso50_tf (9491 elements in the font array) that has more elements than the 34pt font.

For more information these are the commands that I'm using to build the fonts:

`otf2bdf/otf2bdf -a -r 72 -p 43 -o tmp.bdf ./ttf/Logisoso.ttf`
`bdfconv/bdfconv -v -f 1 -m '30-8364' tmp.bdf -o u8g2_font_logisoso30_tx.c  -n u8g2_font_logisoso30_tx -d ./bdf/helvB18.bdf`

`otf2bdf/otf2bdf -a -r 72 -p 45 -o tmp.bdf ./ttf/Logisoso.ttf`
`bdfconv/bdfconv -v -f 1 -m '32-8364' tmp.bdf -o u8g2_font_logisoso32_tx.c  -n u8g2_font_logisoso32_tx -d ./bdf/helvB18.bdf`

`otf2bdf/otf2bdf -a -r 72 -p 49 -o tmp.bdf ./ttf/Logisoso.ttf`
`bdfconv/bdfconv -v -f 1 -m '8364,32-127' tmp.bdf -o u8g2_font_logisoso34_tx.c  -n u8g2_font_logisoso34_tx -d ./bdf/helvB18.bdf`

I've also tested using ./bdf/helvB24.bdf with same results.

Can you help me to build at least th 34pt font?

Thanks,

  Manuel Martínez

MMartinezV, 2018-08-09:
ups, now I've seen that this issue is closed. I've opened another one.