olikraus/u8g2 #610: Support for PaPiRus HAT Display

Donderda, 2018-05-09:
Hi Oli (and everyone else),
can you tell me whether u8g2 supports the PaPiRus HAT Display (see https://github.com/PiSupply/PaPiRus/)? In those repos I found this document: https://github.com/PiSupply/PaPiRus/blob/master/hardware/PaPiRus%20HAT/Latest%20Version%20-%20v1.9/4P018-00_03_G2_Aurora%20Mb_COG_Driver_Interface_Timing_for_small%2Bsize_2015....pdf
Does this help?

Any reply would be appreciated. :)

Donderda, 2018-05-09:
To be more precise: I want to run the 2.7" Display on a PaPiRus v1.9 board. :)

olikraus, 2018-05-09:
Unfortunately, there is no support and probably there also will not be support for this display.
U8g2 was developed with a specific kind of target device (semi-intelligent OLED screens and GLCDs). The SSD16xx based e-paper controller somehow fits, but the controller mentioned above deviates too much from the target device. This means: U8g2 can not control such a display without huge rework. At the end, it would be simpler to create a new library for such kind of display.

Donderda, 2018-05-09:
Too bad, but thank you for your reply :(

closed 2018-05-09 by Donderda