olikraus/u8g2 #611: Font  6x11  u8g2_font_t0_11_t_all

vadpol, 2018-05-09:
Is everything OK with it? Is this correct size of 6x11 pixels as declared? As to me it seems to be bigger than  u8g2_font_6x12_tf. Could you, please, check it? Many thanks!

olikraus, 2018-05-09:
Here are the two fonts:
[image] [image]

The font name (6x11 vs 6x12) is the name or the original font. It does not reflect the size of the letters. Instead you can look at the height of captial A, which is 8 for the 6x11 font but 7 for the 6x12 font (see the screenshots above), so yes: the 6x12 font is "smaller" than the 6x11 font.

vadpol, 2018-05-10:
Hi Oliver, many thanks for your fast reaction, now it's more clear indeed! In such a case I have the following question which really needs an answer. 
As you probably know your u8g2 library is used in many projects, namely in webradio Ka-Radio32 based on ESP32 platform. Originally it has just a latinic set of fonts to be displayed on OLED SSH1306 128x64 screen as an option. The font u8g2_font_6x12_tf is a default one in there, however it doesn't contain cyrillic letters which are quite needed in my case. It was some advice on how to make it by replacing of some library files and fonts which have cyrillic letters, namely to replace the u8g2_font_6x12_tf with u8g2_font_t0_11_t_all. It works well with cyrillic BUT due to these fonts size difference some information (date and time) is overlapped due to bigger default font u8g2_font_t0_11_t_all on the screen. If I apply smaller font like u8g2_font_5x8_cyrillic as default for example, then everything is fine however the font appearance on the screen is too small... 
Would you, please, suggest which cyrillic font would be the same size replacement to native u8g2_font_6x12_tf? I could add the issue pics if needed.
Many thanks!

olikraus, 2018-05-10:
So, you need the u8g2_font_6x12_t_cyrillic, right? The cyrillic glyphs are included in the original bdf so the generation of the same should be no problem.

vadpol, 2018-05-10:
Thanks for your answer! Could you, please tell in steps what I need to do in order to generate the u8g2_font_6x12_t_cyrillic font then as I am not really too deep in that so far? Thank you very much!

olikraus, 2018-05-10:
Looks great :-)