olikraus/u8g2 #612: U8g2 and Ucg combine error

Devilscave, 2018-05-10:
Hello,
I need the U8g2 and Ucg libs in the same sketch, but I get errors like:

_libraries\U8g2\clib\u8g2_polygon.c.o (symbol from plugin): In function `pg_ClearPolygonXY':
(.text+0x0): multiple definition of `pg_ClearPolygonXY'
libraries\Ucglib\clib\ucg_polygon.c.o (symbol from plugin):(.text+0x0): first defined here
libraries\U8g2\clib\u8g2_polygon.c.o (symbol from plugin): In function `pg_ClearPolygonXY':
(.text+0x0): multiple definition of `pg_AddPolygonXY'
libraries\Ucglib\clib\ucg_polygon.c.o (symbol from plugin):(.text+0x0): first defined here
libraries\U8g2\clib\u8g2_polygon.c.o (symbol from plugin): In function `pg_ClearPolygonXY':
(.text+0x0): multiple definition of `pg_DrawPolygon'
libraries\Ucglib\clib\ucg_polygon.c.o (symbol from plugin):(.text+0x0): first defined here
collect2.exe: error: ld returned 1 exit status
exit status 1_

can you fix this?

olikraus, 2018-05-10:
Thanks for reporting. I have fixed https://github.com/olikraus/u8g2/blob/master/csrc/u8g2_polygon.c
You could just replace this file in your local installation.

olikraus, 2018-05-11:
:)

closed 2018-05-11 by olikraus