olikraus/u8g2 #613: Using UTF-8 with u8g2log

RobotGB, 2018-05-12:
Hi,
Your library is amazing and it has really helped me with my coding.
The only thing is that I want to be able to use non-ASCII text with u8g2log.
Do you have a way to do this?
This is my code (it works with ASCII but not UTF-8):
```
#include <U8x8lib.h>
#include <U8g2lib.h>

// U8g2 constructor
U8G2_SSD1306_128X64_NONAME_F_SW_I2C u8g2(U8G2_R0, /* clock=*/ PIN_WIRE_SCL, /* data=*/ PIN_WIRE_SDA, /* reset=*/ U8X8_PIN_NONE);

// Create a U8g2log object
U8G2LOG u8g2log;

// assume 4x6 font, define width and height
#define U8LOG_WIDTH 10
#define U8LOG_HEIGHT 5

// allocate memory
uint8_t u8log_buffer[U8LOG_WIDTH*U8LOG_HEIGHT];

//String message = "日本語（にほんご、にっぽんご[注 1]）は、主に日本国内や日本人同士の間で使われている言語である。日本は法令によって公用語を規定していないが、法令その他の公用文は全て日本語で記述され、各種法令[6]において日本語を用いることが定められ、学校教育においては「国語」として教えられるなど、事実上、唯一の公用語となっている。使用人口について正確な統計はないが、日本国内の人口、および日本国外に住む日本人や日系人、日本がかつて統治した地域の一部住民など、約1億3千万人以上と考えられている[7]。統計によって前後する可能性はあるが、この数は世界の母語話者数で上位10位以内に入る人数である。日本で生まれ育ったほとんどの人は、日本語を母語とする[注 2]。日本語の文法体系や音韻体系を反映する手話として日本語対応手話がある。2017年4月現在、インターネット上の言語使用者数は、英語、中国語、スペイン語、アラビア語、ポルトガル語、マレー語に次いで7番目に多い[8]。"; //Japanese Wikipedia for Japanese
String message = "Hello World! I am testing this on my SSD1306 OLED. I hope I'll be able to get it to work with UTF8!"; //Testing with ASCII works

void setup(void) {
  u8g2.begin();    
  u8g2.setFont(u8g2_font_b12_t_japanese3);  // set the font for the terminal window, supports all characters
  u8g2log.begin(u8g2, U8LOG_WIDTH, U8LOG_HEIGHT, u8log_buffer); // connect to u8g2, assign buffer
  u8g2log.setLineHeightOffset(0); // set extra space between lines in pixel, this can be negative
  u8g2log.setRedrawMode(0);   // 0: Update screen with newline, 1: Update screen for every char  
  u8g2.enableUTF8Print();
}

void loop(void) {
  u8g2.enableUTF8Print();
  // Print a number on the U8g2log window
  u8g2log.print(message);
  // Print a new line, scroll the text window content if required
  // Refresh the screen
  u8g2log.print("\n");

  u8g2.sendBuffer();
  
  delay(500);
}
```
Many thanks.

olikraus, 2018-05-12:
Difficult. It requires effort to pass utf-8 through Arduino print function and reconstruct the Unicode.

RobotGB, 2018-05-12:
That's fine, I'll create a function to do the scrolling instead of the inbuilt one, then draw UTF8 with u8g2.
Do you know where the existing u8log code is stored?
Many thanks.

olikraus, 2018-05-12:
Core procedures are here:
https://github.com/olikraus/u8g2/blob/master/csrc/u8log.c

Interface for u8g2 is here:
https://github.com/olikraus/u8g2/blob/master/csrc/u8log_u8g2.c

RobotGB, 2018-05-12:
Updated code:
```cpp
#include <U8x8lib.h>
#include <U8g2lib.h>

// U8g2 constructor
U8G2_SSD1306_128X64_NONAME_F_SW_I2C u8g2(U8G2_R0, /* clock=*/ PIN_WIRE_SCL, /* data=*/ PIN_WIRE_SDA, /* reset=*/ U8X8_PIN_NONE);

const int line_width = 20; //How many characters can fit on a line
const int line_height = 16; //Pixel height of the line
int current_line = 1; //The line being drawn in the for loop
int voffset = 0; //The vertical offset

const int scrollDelta = 2; //How many characters to skip at once.
const int bytesPerChar = 1; //1 for English, 3 for other languages e.g. Arabic, Japanese etc. I'm not sure about characters like ë or ñ.

String message = "English is a West Germanic language that was first spoken in early medieval England and is now a global lingua franca. Named after the Angles, one of the Germanic tribes that migrated to England, it ultimately derives its name from the Anglia (Angeln) peninsula in the Baltic Sea. It is closely related to the Frisian languages, but its vocabulary has been significantly influenced by other Germanic languages, particularly Norse (a North Germanic language), as well as by Latin and French.";
//String message = "日本語（にほんご、にっぽんご）は、主に日本国内や日本人同士の間で使われている言語である。日本は法令によって公用語を規定していないが、法令その他の公用文は全て日本語で記述され、各種法令において日本語を用いることが定められ、学校教育においては「国語」として教えられるなど、事実上、唯一の公用語となっている。使用人口について正確な統計はないが、日本国内の人口、および日本国外に住む日本人や日系人、日本がかつて統治した地域の一部住民など、約１億３千万人以上と考えられている。統計によって前後する可能性はあるが、この数は世界の母語話者数で上位１０位以内に入る人数である。日本で生まれ育ったほとんどの人は、日本語を母語とする。日本語の文法体系や音韻体系を反映する手話として日本語対応手話がある。２０１７年4月現在、インターネット上の言語使用者数は、英語、中国語、スペイン語、アラビア語、ポルトガル語、マレー語に次いで７番目に多い。";

void setup(void) {
  u8g2.begin();
  Serial.begin(9600);
  while (!Serial) {
    ;
  }
  u8g2.setFont(u8g2_font_b12_t_japanese3);  // set the font for the terminal window
  u8g2.enableUTF8Print();
  Serial.print(message);
}

void loop(void) {

  Serial.println("Changing message..........");
  u8g2.clearBuffer();
  current_line = 1;
  voffset -= scrollDelta;
  for (int i = 0; i < message.length(); i += (bytesPerChar * line_width)) {
    Serial.println(message.substring(i, i + (bytesPerChar * line_width)));
    u8g2.setCursor(0, (current_line * line_height) + voffset);
    u8g2.println(message.substring(i, i + (bytesPerChar * line_width)));
    current_line++;
  }
  Serial.println("Done!");
  u8g2.sendBuffer();

  if ((current_line * line_height) + voffset < 0) {
    voffset = 0;
  }
}
```