olikraus/u8g2 #615: Chip Select in serial mode

IMAN4K, 2018-05-15:
Hi @olikraus 

I'm wonder if we could remove `cs` arg in LCDs like `ST7920` and in serial mode of operation, e.g `U8G2_ST7920_128X64_F_SW_SPI`
Because documentation stays that chip select or `cs` in serial mode act like chip enable and we could simply connect it to logic 1 (`VCC`)
Am i right ?

Regards.
IMAN.

olikraus, 2018-05-15:
If I remember correctly, the chip enable input still works as expected also in serial mode. You may connect it to 1, but it is still better to let U8g2 control the chip enable input: The chip enable also resets the communication and puts it onto a defined state.

IMAN4K, 2018-05-16:
OK
This might be different between manufacturers of LCD but for now i'm able to working with LCD as normal when `cs` is connected to `3.3`.
And we've 1 `GPIO` free for other use.

Thanks Anyway

closed 2018-05-16 by IMAN4K