olikraus/u8g2 #619: Is it possible to add an SH1108 based display?

feistyz, 2018-05-22:
I have an SH1108 based display. Is it possible to add support for it? I am a bit thin on documentation but should be able to provide the necessary info.

olikraus, 2018-05-22:
I assume it is possible, but I need more information.

feistyz, 2018-05-22:
Which info do you require?

olikraus, 2018-05-22:
The datasheet for your display itself would be a good starting point. In order to add something I also need the dimensions of the display.
Is there any link available to your display?
I also need the datasheet for the sh1108 itself.

feistyz, 2018-05-23:
I don't have the datasheet for the display, however, I do have relevant info. Here is a [zip file](https://github.com/olikraus/u8g2/files/2029636/Display.info.zip) containing powerup sequence, SH1108 datasheet and segment info. The display is 160x160.

Let me know if there is something else that is needed.

Thanks!

olikraus, 2018-05-23:
What kind of interface do you need? SPI?

feistyz, 2018-05-23:
Good point. It supports both. But SPI to start would be great.

<https://github.com/olikraus/u8g2/issues/619#issuecomment-391435563>, or mute
the thread

olikraus, 2018-05-23:
I have added all the interfaces. A constructor with hardware SPI is now part of the examples. New beta 2.23.4 created. Please have a look at the sh1108 constructor: Will the FlipMode.ino work correctly? How does the U8g2 logo look like?
 
You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Install the zip file via Arduino IDE, add zip library menu.

feistyz, 2018-05-23:
Thanks! Will test now and let you know.

feistyz, 2018-05-23:
This is great! Everything works as expected. Flip works correctly. Logo looks right.

Thanks!