olikraus/u8g2 #621: u8x8 for esp8266 nonos sdk

andrushchak, 2018-05-23:
Hello!

Is it possible to use plain C version of the u8x8 library with ESP8266 non OS SDK?
Are cannot find any example about SPI interface configuration for OLED 128x64 noname.
Could you please point me to some demo project?

Thank you,
Alex

olikraus, 2018-05-23:
Plain C is possible. There are several issues regarding the port of U8g2 to plain C and how to write a low level driver. There is also a Wiki page on this topic which also includes links to several projects: https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform

andrushchak, 2018-05-23:
Thank you for response. u8x8_Setup signagure on the wiki page looks wrong:
https://github.com/olikraus/u8g2/wiki/u8x8setupc#ssd1306-128x64_noname

Could you please give a sample in introduction, the same as for u8g2?

Best regards,
Alex

olikraus, 2018-05-23:
Here is an example:

andrushchak, 2018-05-23:
I tried to implement simple gpio and delay callback for esp8266:
uint8_t ICACHE_FLASH_ATTR u8x8_gpio_and_delay_esp8266(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
	console_printf("msg=%d, arg_int=%d\n", msg, arg_int);

    switch(msg){
		case U8X8_MSG_GPIO_AND_DELAY_INIT:
            PIN_FUNC_SELECT(SCL_GPIO_MUX, SCL_GPIO_FUNC);
	        PIN_PULLUP_EN(SCL_GPIO_MUX);
            PIN_FUNC_SELECT(MOSI_GPIO_MUX, MOSI_GPIO_FUNC);
	        PIN_PULLUP_EN(MOSI_GPIO_MUX);
            PIN_FUNC_SELECT(RES_GPIO_MUX, RES_GPIO_FUNC);
	        PIN_PULLUP_EN(RES_GPIO_MUX);
            PIN_FUNC_SELECT(DC_GPIO_MUX, DC_GPIO_FUNC);
	        PIN_PULLUP_EN(DC_GPIO_MUX);
            PIN_FUNC_SELECT(CS_GPIO_MUX, CS_GPIO_FUNC);
	        PIN_PULLUP_EN(CS_GPIO_MUX);
			break;

		case U8X8_MSG_DELAY_MILLI:
            os_delay_us(arg_int * 1000);
			break;

		case U8X8_MSG_DELAY_10MICRO:
			os_delay_us(10);
			break;

		case U8X8_MSG_GPIO_SPI_CLOCK:
		    GPIO_OUTPUT_SET(SCL_GPIO, arg_int);
			break;

		case U8X8_MSG_GPIO_SPI_DATA:
		    GPIO_OUTPUT_SET(MOSI_GPIO, arg_int);
			break;	

		case U8X8_MSG_GPIO_RESET:
		    GPIO_OUTPUT_SET(RES_GPIO, arg_int);
			break;

		case U8X8_MSG_GPIO_DC:		
		    GPIO_OUTPUT_SET(DC_GPIO, arg_int);
			break;		

		case U8X8_MSG_GPIO_CS:	
		    GPIO_OUTPUT_SET(CS_GPIO, arg_int);
			break;		

		default:
            u8x8_SetGPIOResult(u8x8, 1);
            break;
	}
							
	return 1;
}

It obtain some messages, but my screen is fully dark.

Initialization code is:
    u8x8_Setup(&u8x8, u8x8_d_ssd1306_128x64_noname, u8x8_cad_001, u8x8_byte_4wire_sw_spi, u8x8_gpio_and_delay_esp8266);
    u8x8_InitDisplay(&u8x8);
    u8x8_ClearDisplay(&u8x8);
    u8x8_SetPowerSave(&u8x8, 0);
    u8x8_SetFont(&u8x8, u8x8_font_amstrad_cpc_extended_r);  
    u8x8_DrawString(&u8x8, 0,0, "Hello World!");

Do you see what is wrong with it?
Thank you!

olikraus, 2018-05-23:
It should be
```
	case U8X8_MSG_DELAY_10MICRO:
		os_delay_us(10*arg_int);
		break;
```

The rest of the code seems to be ok. Do you see any activity on the pins?

andrushchak, 2018-05-23:
Corrected, but screen is black.

There are a lot of activity in callback:
msg=73, arg_int=1
msg=64, arg_int=0
msg=75, arg_int=1
msg=41, arg_int=100
msg=75, arg_int=0
msg=41, arg_int=100
msg=75, arg_int=1
msg=41, arg_int=100
msg=73, arg_int=0
msg=44, arg_int=20
msg=74, arg_int=0
msg=65, arg_int=1
g=44, arg_int=50
msg=64, arg_int=0
msg=44, arg_int=50
msg=65, arg_int=0
msg=64, arg_int=1
msg=44, arg_int=50
msg=64, arg_int=0
msg=44, arg_int=50
msg=65, arg_int=1
msg=64, arg_int=1
msg=44, arg_int=50
msg=64, arg_int=0
msg=44, arg_int=50
msg=65, arg_int=0
msg=64, arg_int=1
msg=44, arg_int=50
arg_int=50
msg=65, arg_int=0
msg=64, arg_int=1
msg=44, arg_int=50
msg=64, arg_int=0
msg=44, arg_int=50
msg=65, arg_int=1
msg=64, arg_int=1
msg=44, arg_int=50
msg=64, arg_int=0
msg=44, arg_int=50
msg=65, arg_int=0
msg=64, arg_int=1
msg=44, arg_int=50
msg=64, arg_int=0
=50
msg=64, arg_int=0
msg=44, arg_int=50
msg=65, arg_int=1
msg=64, arg_int=1
msg=44, arg_int=50
msg=64, arg_int=0
msg=44, arg_int=50
msg=65, arg_int=1
msg=64, arg_int=1
msg=44, arg_int=50
msg=64, arg_int=0
msg=44, arg_int=50
msg=74, arg_int=0
msg=65, arg_int=1
msg=64, arg_int=1
rg_int=0
msg=44, arg_int=50
msg=65, arg_int=0
msg=64, arg_int=1
msg=44, arg_int=50
msg=64, arg_int=0
msg=44, arg_int=50
msg=65, arg_int=0
msg=64, arg_int=1
msg=44, arg_int=50
msg=64, arg_int=0
msg=44, arg_int=50
msg=65, arg_int=0
msg=64, arg_int=1
...

olikraus, 2018-05-23:
Great, it means, the library works as expected, but the question is, is there some activity at the 4-wire SPI interface?

andrushchak, 2018-05-23:
Yes, i can see different levels on pins, but screen is black

andrushchak, 2018-05-23:
Are there any way to test SSD 1306 without SPI signaling?

olikraus, 2018-05-24:
no :(

andrushchak, 2018-05-24:
Hi again,

I asked seller about screen datasheet and received response with SSD1315 as a driver.
[0.96-7P-LCM.pdf](https://github.com/olikraus/u8g2/files/2035375/0.96-7P-LCM.pdf)

According to wiki this IC is not supported. Do you think i can use any existing driver for this IC?

olikraus, 2018-05-24:
> According to wiki this IC is not supported. Do you think i can use any existing driver for this IC?

Difficult to say. From my experience, there might be huge differences. So you connected a SSD1315 display, but selected a SSD1306 driver? Then yes, I guess we need to study the datasheet first. And yes, indeed the datasheet ot the SSD1315 would be required.

andrushchak, 2018-05-24:
Update. It works with adafruit SSD1306 library, so it 1306 or 1306 compatible. 
Need to figure out what's wrong with u8x8.

#define OLED_MOSI  13   //D1
#define OLED_CLK   5   //D0
#define OLED_DC    12
#define OLED_CS    15
#define OLED_RESET 14

Adafruit_SSD1306 display(OLED_MOSI, OLED_CLK, OLED_DC, OLED_RESET, OLED_CS);

void setup()   {                
//  Serial.begin(9600);
  display.begin(SSD1306_SWITCHCAPVCC);
  display.display();
  delay(1000);
  display.clearDisplay();
  display.setTextSize(1);
  display.setTextColor(WHITE);
}

void loop()
{
  static unsigned long thisMicros = 0;
  static unsigned long lastMicros = 0;
  display.clearDisplay();
  display.setCursor(0,0);
  display.print("Now is the time for all good men to come to the aid the party \n");
  display.print("The quick brown fox jumps over a lazy dog \n");
  display.print(thisMicros - lastMicros);
  display.print(" microseconds");
  display.display();
  lastMicros = thisMicros;
  thisMicros = micros();
}

andrushchak, 2018-05-24:
Compiled sketch from u8g2 sources
  "Hello World" version for U8x8 API

with
U8X8_SSD1306_128X64_NONAME_4W_SW_SPI u8x8(/* clock=*/ 5, /* data=*/ 13, /* cs=*/ 15, /* dc=*/ 12, /* reset=*/ 14);

and ....

it worked as well!

So, what is the problem with c program?

olikraus, 2018-05-24:
ok, well, still difficult to say...

olikraus, 2018-06-17:
no further comments, closing...

closed 2018-06-17 by olikraus