olikraus/u8g2 #624: strange behavior

pppeixoto, 2018-05-25:
I have a arduino system with 2 OLEds, initialization below:
U8G2_SSD1326_ER_256X32_1_4W_SW_SPI OLED_RIGHT(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 7, /* dc=*/ 6, /* reset=*/ 5);         // experimental driver for ER-OLED018-1 10 9 8
U8G2_SSD1326_ER_256X32_1_4W_SW_SPI OLED_LEFT(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);         // experimental driver for ER-OLED018-1 10 9 8
When I write in the first OLED  the second presents a "noise" in the lower right corner, in the form of a square. The reverse does not happen. What could be causing the problem?
https://centi-my.sharepoint.com/:i:/g/personal/ppeixoto_centi_pt/EQZDQdNqn0hGiDYl8D5ZLKgBJxu7E0sFQrWGcGWL0_bEmg?e=9erbKi

pppeixoto, 2018-05-25:
I've checked that when I put the D/C pin of OLEd on which I'm not writing LOW, the problem no longer happens. However I think it is a driver problem.

olikraus, 2018-05-25:
hmm... I think there are also some hints in the FAQ.

closed 2018-06-17 by olikraus