olikraus/u8g2 #628: MAX7219 with a single module (8x8)

mz2212, 2018-05-30:
The library is great, but I only have a single MAX7219 module at the moment, can support for the 8x8 resolution be added?

olikraus, 2018-05-30:
The existing 32x8 doesn't work?

mz2212, 2018-05-30:
It does, but 0 on the x axis starts on the what would be the 4th module. Meaning that for a single module, you need to start at 24 to get the first column of pixels. It's not super important, but it would be handy to start at 0 to get at those pixels.

I suppose it would be possible to use a different rotation in the initializer, but that would also cause the (0, 0) point to be in an unexpected place.

olikraus, 2018-05-30:
Ok, I will add this

olikraus, 2018-06-13:
Done, constructor is:
U8G2_MAX7219_8X8_F_4W_SW_SPI

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

olikraus, 2018-08-04:
any feedback?

mz2212, 2018-08-08:
Sorry for forgetting to get back to you for so long on this one.
It works for the cases I use it in (text display), I haven't tested it with the graphics portion of the library.

olikraus, 2018-08-11:
No problem. Thanks for the feedback.

closed 2018-08-23 by olikraus