olikraus/u8g2 #630: Arduino Mega + ST7920 + HW SP

neoblack2, 2018-06-03:
Hello, tell me how can I attach these details?

//U8G2_ST7920_128X64_F_SW_SPI u8g2(U8G2_R0, /* clock=/ 18 / A4 / , / data=/ 16 / A2 /, / CS=/ 17 / A3 /, / reset=/ U8X8_PIN_NONE);
//U8G2_ST7920_128X64_F_SW_SPI u8g2(U8G2_R0, / clock=/ 13, / data=/ 11, / CS=/ 10, / reset=/ 8);
//U8G2_ST7920_128X64_F_HW_SPI u8g2(U8G2_R0, / CS=/ 10, / reset=*/ 8);

I am used
U8G2_ST7920_128X64_F_HW_SPI u8g2(U8G2_R0,53,8);
or
U8G2_ST7920_128X64_F_SW_SPI u8g2(U8G2_R0, /* clock=/ 52, / data=/ 51, / CS=/ 53, / reset=*/ 49);
and it does not work ((

olikraus, 2018-06-03:
Looks correct, check your wiring, power supply and ensure that you use level shifters if required.

olikraus, 2018-06-17:
no further comments, closing...

closed 2018-06-17 by olikraus