olikraus/u8g2 #631: ChipSelect issue with my 192x64 LCD (based on KS0108)

Busa357, 2018-06-06:
Hi,

I'm trying to get my new LCD screen working but I have some troubles.
The screen is an EastRising ERM19264-1. 192x64 pixels, it is drived by a S6A0107 which is supposed to be a KS0108 equivalent. It has 3 chipselect lines (CS1, CS2, CS3) that are active LOW.
After installing the library, I loaded the GraphicsTest sketch to try my new LCD. 
I can read some text and view some graphics but I can clearely view 3 different blocks of 64pix.
The first (Left) stay blank. The second and the third (center and right) are alive but show exactly the same thing.
I immediately think about a Chipselect issue. 
What I have done so far:
   - I have double checked all my wiring.    
   - I tried library v2.22.18 and v2.23.4
   - I tested my screen by forcing Left (CS1=0, CS2=1, CS3=1). Left zone is ok and work! 
   - I tested my screen by forcing center (CS1=1, CS2=0, CS3=1). Center zone is ok and work! 
   - I tested my screen by forcing Right (CS1=1, CS2=1, CS3=0). Right zone is ok and work! 
   - I have tested CS1, CS2, CS3, RS and E signals with my Tektronix. BOOM! CS3 is always LOW! (Library issue?)
  
What can I do know? 
Tell me I you need more info.

Thank you

[image]

olikraus, 2018-06-06:
Can you post your code? How did you setup the constructor?

olikraus, 2018-06-06:
Personal comments:

Constructor:
https://github.com/olikraus/u8g2/blob/master/cppsrc/U8g2lib.h#L9733
The constructor will call u8x8_SetPin_KS0108:
https://github.com/olikraus/u8g2/blob/master/cppsrc/U8x8lib.cpp#L1495

ohhhhhh:
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_byte.c#L336

I think this must be
arg = arg >> 1;

ok, I do not own this display, so I never came across this.
Can you fix the line to ">>1" instead of ">>2"?

Busa357, 2018-06-06:
Hi,

I have not yet making a code from myself.
For now, I just want to try my LCD with demo sketches. They are all make the same problem.
The most interesting sketch to see the problem is GraphicsTest. (U8g2_Arduino-master\examples\page_buffer\GraphicsTest\)
Here is my constructor setup:
U8G2_KS0108_ERM19264_1 u8g2(U8G2_R0, 4, 5, 6, 7, 8, 9, 10, 11, /*enable=*/ SS, /*dc=*/ MOSI, /*cs0=*/ A3, /*cs1=*/ A4, /*cs2=*/ A5, /* reset=*/  U8X8_PIN_NONE); 	// Set R/W to low! 
I use a genuine Arduino Micro and, of course, I already try several pin setup, especially for cs0, cs1, and cs2.

I just fixed u8x8_byte.c#L336
Now, on the oscillo, I have cs1 always up and some activity on cs2 and cs3. 
On the screen, left side stay blank, center and right are alive but not as they should.

[image]

olikraus, 2018-06-06:
u8x8_gpio_SetCS is actually defined here:
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8.h#L734
So it is acually the same as the other two statements, except it uses `GPIO_CS`.

> I have not yet making a code from myself.

You can update the file by yourself. Arduino IDE will recompile it. Just locate u8x8_byte.c on your local drive, update line 336 accordingly.

Busa357, 2018-06-06:
> You can update the file by yourself. Arduino IDE will recompile it. Just locate u8x8_byte.c on your local drive, update line 336 accordingly.

I already make this modication. As I say in the 2nd part of my 1st reply.
I fixed u8x8_byte.c#L336 and recompile the GraphicsTest sketch.
Now, on the oscillo, I have cs1 always up and some activity on cs2 and cs3.
On the screen, left side stay blank, center and right are alive but not as they should. (see my 2nd picture).

Busa357, 2018-06-06:
In the meantime, I did some search...
In the file **u8x8_d_ks0108.c #L213,**
you define  **uint8_t u8x8_d_ks0108_erm19264(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)**
Some line later, I can see _&u8x8_ks0108_**128**x64_display_info_     Mistake?

olikraus, 2018-06-06:
Indeed, this does not make sense.

olikraus, 2018-06-06:
I have created a new beta release 2.23.6:
You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

Busa357, 2018-06-07:
Hello,
Almost good!
After loading the 2.23.36, the screen was again sliced in 3 parts and not in the right order.
It is just an inversion between CS0 and CS2. 
I can have a good result by swapping these two output on the Arduino but it will be cleaner to update the library.
After a quick look, I found that I just have to modify u8x8_d_ks0108.c #L310  
Just reorder u8x8->cad_cb(u8x8, U8X8_MSG_CAD_START_TRANSFER  3, 5, 6 to 6, 5, 3 do the job !
I suppose that I should do the same changes in the part from L248 to L292 but I do not understand the purpose of this part of the code. 
Anyway, thank you for your reactivity.

olikraus, 2018-06-07:
> Just reorder u8x8->cad_cb(u8x8, U8X8_MSG_CAD_START_TRANSFER 3, 5, 6 to 6, 5, 3 do the job !

I thought CS0 is responsible for the left most part of the display. Is this wrong?

Busa357, 2018-06-08:
I tested again this morning by forcing ChipSelect entry to +5 or GND.
  - CS1=0; CS2=+5; CS3=+5 -> only left part show something
  - CS1=+5; CS2=0; CS3=+5 -> only center part show something
  - CS1=+5; CS2=+5; CS3=0 -> only right part show something

I did another test by modifying u8x8_d_ks0108.c
If I put the value 3 in L312, L318 and L324, only the RIGHT part is active.
3 = 011 in binary. So the MSB (most significant bit) is for CS3 and the LSB is for CS1.

olikraus, 2018-06-08:
> Just reorder u8x8->cad_cb(u8x8, U8X8_MSG_CAD_START_TRANSFER 3, 5, 6 to 6, 5, 3 do the job !

ah... yes, you are right.

reference:
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_ks0108.c#L312

has to be fixed.

olikraus, 2018-06-09:
ok, i changed the cs values...

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

Busa357, 2018-06-11:
Hi,

v2.23.7 is ok for me.
Thank you again for your all your work in maintaining this library.
I close this issue.

closed 2018-06-11 by Busa357

olikraus, 2018-06-11:
Great! 
Thanks.