olikraus/u8g2 #632: Mirrored Display

b2cbd, 2018-06-07:
I'm using U8x8 on SSD1306 for showing some normal and inverted text(clock and temperature). It usually works great for few days. But after few days it just mirrors the whole display. Though in the mirrored display, texts are updating perfectly. I need to restart Arduino manually and good to go again.

What can be the cause of mirrored display after few days?

olikraus, 2018-06-07:
It has been reported from several users, that the internal RAM of the display may loose information after some days. 

If you start your board once and let it run several days, than u8g2 will send the required startup code for the display only once. After that the display may forget some parts of the correct startup configuration.

So as a result, you may need to resend the startup code again. You can just call .begin() to refresh the configuration. So I assume it would make sense, to call .begin() once per day or so.

b2cbd, 2018-06-07:
Okay, I will call .begin() once a day, let's see if it resolves. :)

closed 2018-06-17 by olikraus