olikraus/u8g2 #634: Big 7-Segments style font

Busa357, 2018-06-12:
Hi,

I wanted a big font for my clock project. An 7-Segments like style.
Since it does not exist, I created one by myself. The size is 26x42.
To minimise the weight, the font have  only 12 glyphs. (0,1,2,3,4,5,6,7,8,9, : ,space) 
Note that the two dots and the space glyph have exactly the same width.
So you could easiy make the two dots blinking by alternating these two characters.
I hope that this font could be usefull to someone. 
Oli, you can of course and it the the library if you find it interesting.

[7Segments_26x42.zip](https://github.com/olikraus/u8g2/files/2092975/7Segments_26x42.zip)

[image]

olikraus, 2018-06-12:
Nice. Thanks for the contribution.

olikraus, 2018-08-04:
added, font rebuild required.

olikraus, 2018-08-24:
font added, docu updated

closed 2018-08-24 by olikraus

bisra2, 2022-12-01:
Je teste cette belle font sur un afficheur tft 320x240 mais le segment 'd' du digit '1' s'allume!