olikraus/u8g2 #635: Add support for T6963 160x80

zshivers, 2018-06-13:
There is already support in this library for other resolutions of this chip - just need to add correction initializations for this resolution. 

Using for https://www.crystalfontz.com/product/cfag16080c1tfhtz-160x80-graphic-lcd-module#specs

I am able to do testing for this module.

olikraus, 2018-06-13:
done... please have a look. Constructors are:
U8G2_T6963_160X80_1_8080
U8G2_T6963_160X80_2_8080
U8G2_T6963_160X80_F_8080

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

zshivers, 2018-06-21:
I believe these lines `chip_enable_level` and `chip_disable_level` should be inverted. 
https://github.com/olikraus/u8g2/blob/52679cdcfa7808540d422222bb422e744a229d17/csrc/u8x8_d_t6963.c#L464

They currently read:
```
  /* chip_enable_level = */ 1,
  /* chip_disable_level = */ 0,
```

But they should be 
```
  /* chip_enable_level = */ 0,
  /* chip_disable_level = */ 1,
```

After that change, this LCD module works as expected.

olikraus, 2018-06-22:
Thanks for the feedback. I will modify the code.

olikraus, 2018-08-04:
fixed with head  f6197cbd

closed 2018-08-04 by olikraus