olikraus/u8g2 #636: SSD1306 setting brightness problem

andrasfeher, 2018-06-14:
I am using the [Geekcreit® 0.91 Inch 128x32 IIC I2C Blue OLED LCD Display DIY Oled Module SSD1306 Driver IC DC 3.3V 5V For Arduino PIC](https://www.banggood.com/0_91-Inch-128x32-IIC-I2C-Blue-OLED-LCD-Display-DIY-Oled-Module-SSD1306-Driver-IC-DC-3_3V-5V-p-1140506.html?rmmds=search&cur_warehouse=CN) module. Everything is fine except that I am unable to reduce significantly the brightness of the text using the setContrast function.

The setup part of my code:

```
U8G2_SSD1306_128X32_UNIVISION_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);

void setup()
{
   u8g2.begin();
  u8g2.setFontDirection(0);  
  u8g2.setContrast(8);  
  u8g2.setFont(u8g2_font_helvR14_tf);    
 
  u8g2.clearBuffer();
  u8g2.drawStr(0, 24, "Testing ...");
  u8g2.sendBuffer();  
}
```
How can I reduce the brightness?

Thanks,
Andras

olikraus, 2018-06-14:
First of all: Unexpectedly, this is a very complicated problem. Controlling an OLED requires a lot of intelligence, which is put into the SSD1306 controller. This controller includes a lot of configuration options, which influences the brightness. 

The setContrast function on the other side, just programs one of the most important registers of the controller, which unfortunately does not cover a huge brightness range. This actually different from LCDs, where the contrast control allows to change from no pixel visible to all pixel visible. 

To make it more complicated, changing the configuration to allow larger brightness ranges is ot always possible, because it may not work with all OLEDs, as a result, U8G2_SSD1306_128X32_UNIVISION_F_HW_I2C includes a standard setup, which (most important) should show something an almost all available OLEDs. The drawback is, controlling the brightness is difficult.

After the discussion here #98, I have introduced an alternative setup for 128x64 display.
In your case, my suggestion is to put the following two functions into your ino code and play around with the values:

```
// value from 0 to 7, higher values more brighter
void setSSD1306VcomDeselect(uint8_t v)
{	
  u8x8_cad_StartTransfer(u8g2.getU8x8());
  u8x8_cad_SendCmd(u8g2.getU8x8(), 0x0db);
  u8x8_cad_SendArg(u8g2.getU8x8(), v << 4);
  u8x8_cad_EndTransfer(u8g2.getU8x8());
}

// p1: 1..15, higher values, more darker, however almost no difference with 3 or more
// p2: 1..15, higher values, more brighter
void setSSD1306PreChargePeriod(uint8_t p1, uint8_t p2)
{	
  u8x8_cad_StartTransfer(u8g2.getU8x8());
  u8x8_cad_SendCmd(u8g2.getU8x8(), 0x0d9);
  u8x8_cad_SendArg(u8g2.getU8x8(), (p2 << 4) | p1 );
  u8x8_cad_EndTransfer(u8g2.getU8x8());
}
```

You could start with a value of 0 for VCOM. But please be aware, that your display me not work reliable any more with this value.

olikraus, 2018-06-17:
no further comments, closing...

closed 2018-06-17 by olikraus