olikraus/u8g2 #637: Support for e-paper partial refresh

FrankX0, 2018-06-14:
E-paper controllers like SSD1607 support partial refresh. This allows for local updates, resulting in a smoother refresh appearance.
For the SSD1607 this can be implemented using a different LUT. I use following with good results:
```
0x10, 0x18, 0x18, 0x08, 0x18, 0x18, 0x08, 0x00, 
0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
0x00, 0x00, 0x00, 0x00, 0x13, 0x14, 0x44, 0x12, 
0x00, 0x00, 0x00, 0x00, 0x00, 0x00
```
I would like to propose to extend `u8x8_RefreshDisplay` with an extra `mode` option:
```
void u8x8_RefreshDisplay(u8x8_t *u8x8, uint8_t mode)
{
  u8x8->display_cb(u8x8, U8X8_MSG_DISPLAY_REFRESH, mode, NULL);  
}
```

- Mode = 0: full refresh (current implementation)
- Mode = 1: partial refresh

olikraus, 2018-06-14:
What exactly means partial refresh? Does "partial" refer to a limited range of pixel?

FrankX0, 2018-06-14:
With (the current) full refresh, the entire screen is switches between black and white before displaying the actual content. With a partial refresh, only the affected part of the screen is refreshed, so without the entire screen "flickering".
[This](https://www.youtube.com/watch?v=enzUbiSWenQ) YouTube video shows good examples of a partial refresh (btw: at the start a full refresh is done, showing the switching between full black and white).
Btw: I just finished implementing my suggestion. This works great for displaying things like time and temperature.

olikraus, 2018-06-14:
so it is just a different LUT?

FrankX0, 2018-06-15:
Correct

olikraus, 2018-06-15:
Any other precondition required for partial update except for the changed LUT?

FrankX0, 2018-06-15:
Nope. The only thing I did was creating two versions: one with the current LUT and one with the modified LUT. This one I took from example code from one of the suppliers.

olikraus, 2018-06-15:
ok... I will try this...

RonTheBestKingInTheWorld, 2018-09-06:
@FrankX0 
Could you show how to do it?

RonTheBestKingInTheWorld, 2018-09-06:
I use the same LUT but every second time I refresh the screen the area y<32 isn't being updated.

FrankX0, 2018-09-20:
@RonTheBestKingInTheWorld unfortunately I have issues with my development environment. 
So I am currently not able to see if I can reproduce the issue you are seeing.

FrankX0, 2018-11-08:
Unfortunately I still have issues with my e-paper display with the normal/full refresh.
As you can see from the attached [movie](https://github.com/olikraus/u8g2/files/2563600/IMG_5758.zip), the text is printed first with high contrast, after which a refresh takes place which results in a lower contrast end result.

Anyone experiencing the same?

Code1110, 2018-11-23:
My Waveshare SSD1607 shows similar behavior. Please see attached [movie](https://github.com/olikraus/u8g2/files/2609431/SSD1607_f.zip). After issuing `disp:sendBuffer()` a second time, the image (spider pig) appears with high contrast. This test was done with nodemcu on an ESP8266.

FrankX0, 2018-11-27:
When I first worked with this display (u8g2 with Nodemcu) it was working fine. That's also when I started this issue. Unfortunately I lost that code. Now implementing the same again shows this behaviour. So apparently something changed after May/June. In the Arduino environment I went back to older u8g2 versions, but still with the same result.
I also tried the Waveshare [demo code](https://www.waveshare.com/wiki/E-Paper_ESP8266_Driver_Board#Demo_Code), which runs fine.

@olikraus any ideas what might have happened?

olikraus, 2018-11-28:
The driver for the SSD1607 has not been changed for the last 11 months.

> In the Arduino environment I went back to older u8g2 versions, but still with the same result.

So you say the older u8g2 also did not work. So: Which exact versions did not work?

These are the Arduino Releases:
 2018-02-18	v2.21.8  
 2018-04-30	v2.22.18  
 2018-08-24	v2.23.18  
 2018-10-28	v2.24.3  

So v2.22 and v2.21 are working fine?

Looking into the change log, there was nothing which could have caused a change in the e-paper implementation. 

Hmm.. the last modification is related to this: https://github.com/olikraus/u8g2/issues/454

Is there any simple .ino file to test this?

FrankX0, 2018-11-28:
Well, I tested all of the older releases, down to 2.19.8
And all show the same behaviour. So I have no idea why these even worked back in May/June.

This is the code I am using:
```
#include <Arduino.h>
#include <U8g2lib.h>
#include <SPI.h>
#include <Wire.h>

U8G2_SSD1607_200X200_F_4W_HW_SPI u8g2(U8G2_R0, 15, 4, 5);
int x = 20;

void setup(void) {
  u8g2.begin();
  u8g2.setFont(u8g2_font_ncenB14_tr);
}

void loop(void) {
  u8g2.clearBuffer();
  u8g2.drawStr(0,x,"Hello World!");
  u8g2.sendBuffer();
  delay(4000);
  x = x + 20;
  if (x == 200) {x = 20;}
}
```
This results in the text appearing shortly in high contrast, but then a refresh happens which makes the text very light.

olikraus, 2018-11-28:
thanks, hope i have some time during weekend to test this.

FrankX0, 2018-11-29:
Thanks for the suggestion.
But I'm using an ESP8266, initially with Nodemcu. The spi implementation is already stable for a long time. Furthermore, the behaviour is the same when using the Arduino environment.

FrankX0, 2019-01-02:
@olikraus did you already find some time looking into the issues seen with the e-paper display?

olikraus, 2019-01-03:
I have created the following file: https://github.com/olikraus/u8g2/commit/a2e338ecdb339811a82146dc4a6d208712a291e4#diff-fad34319cb3fe71ba3fa59a68870259e

(page buffer method, because my Uno has not enough RAM)

Everything works fine with the hello world code from above.

What is the actual problem? I do not have the code which generates the pictures shown in the move linked above.

olikraus, 2019-01-03:
This is a video, which I took with my device: https://photos.app.goo.gl/qWbq1TyDJCFB6KnQA

So from my perspective everything seems to be fine.

FrankX0, 2019-01-04:
With your display it works fine! So did mine, but not anymore.
If you look at my [video](https://github.com/olikraus/u8g2/issues/637#issuecomment-437158590), you can see that the order of things happening is different compared to your case:

- With the previous text still visible in low contrast the new text (in high contrast!) is displayed
- Then a refresh is performed
- Resulting in the old text removed, but the new text is now in low contrast

[image]
Because my setup has been working correctly without apparently any code changes, it seems to be caused by an (incorrect/insufficient) initialisation process.
Btw: adding more delays as suggested did not improve the behaviour.

olikraus, 2019-01-04:
ok, but what can we do? 
 * I was not able to see any notable changes in u8g2 code which could have caused this problem.
 * I am not able to reproduce the problem.

Maybe someone can point out, what i need to fix in u8g2 code...

LooksStupid, 2019-01-11:
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_ssd1607_200x200.c#L235
U8X8_CAAAA(0x45, 0, 0, **299**&255, **299**>>8),	/* RAM y start & end, 0..**299** */
I think it's should be 199 not 299

olikraus, 2019-01-12:
The controller internally is a 200x300 pixel controller (although the display is only 200x200).
@FrankX0 If you fix this in your local code, will it help?
If so, I can create a new constructor for your display with the modified code.

FrankX0, 2019-01-12:
Unfortunately this did not improve the behaviour.

I did find another issue though at this [line](https://github.com/olikraus/u8g2/blob/45ac4e0edc76c6bc3d808304fccb73c543535ded/csrc/u8x8_d_ssd1607_200x200.c#L78):
`/* spi_mode = */ 2,		/* active high, rising edge */`
The comment is correct, but the value should be 0 in my view: correct?

But then again, no change in behaviour.
I use this Waveshare [code](https://www.waveshare.com/wiki/E-Paper_ESP8266_Driver_Board#Demo_Code) as a reference, which works fine, if freshly run after power-up of the ESP8266+e-paper.
When I run the same application after running your code, the application starts showing vague and nonuniform pictures. To recover from this the e-paper needs to power cycled.
For me this could be a sign that some registers are incorrectly initialised by your code and not re-initialised by the mentioned Waveshare code.

olikraus, 2019-01-12:
As mentioned, the code is written for a different e-paper display. I could create a complete new constructor for your display, but this will be a bigger task. I could do this if someone provides me the init sequence for the waveshare display.

Then I could just replace the init sequence for the embedded adventures display:
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_ssd1607_200x200.c#L214

FrankX0, 2019-01-12:
I now can confirm that the issue is caused by the initialisation!
I first ran the Waveshare application. Then I disabled the init_seq from your code and ran the rest of the application: perfect!
I will now work on doing things properly and see how we can integrate.

olikraus, 2019-01-12:
sounds good

FrankX0, 2019-01-13:
And I'm again able to make the partial refresh working.
This also allows for a much faster screen update: 300ms versus 1200ms.
[e-paper partial refresh.zip](https://github.com/olikraus/u8g2/files/2753099/e-paper.partial.refresh.zip)

olikraus, 2019-01-13:
Thanks for this. So the best will be  to create a new u8g2 constructor with the above init sequence, correct?

FrankX0, 2019-01-13:
Yes, that seems necessary.
Thanks in advance!

olikraus, 2019-01-20:
I have created beta release 2.25.4

Constructor start with `U8G2_SSD1607_WS_200X200`

Can you confirm, that this constructor works? Thanks!

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

FrankX0, 2019-01-20:
Thanks Oliver!
Two modification are required in u8x8_d_ssd1607_200x200.c

`uint8_t u8x8_d_ssd1607_ws_200x200(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)`
The init sequence is still pointing to the original one.
Please change to:
`u8x8_cad_SendSequence(u8x8, u8x8_d_ssd1607_ws_200x200_init_seq); `

The spi mode [initialisation](https://github.com/olikraus/u8g2/blob/45ac4e0edc76c6bc3d808304fccb73c543535ded/csrc/u8x8_d_ssd1607_200x200.c#L78) is incorrect. Please change mode 2 to mode 0:
/* spi_mode = */ 0,	/* active high, rising edge */

Thanks again for all your work on this library!

olikraus, 2019-01-20:
> The init sequence is still pointing to the original one.

Ups, thanks...

> The spi mode initialisation is incorrect. Please change mode 2 to mode 0:

Which board do you use? I mean, there are some boards, for which the mode implementation is wrong: https://github.com/esp8266/Arduino/issues/2416

FrankX0, 2019-01-20:
The SSD1607 expects mode 0.
The comment u8x8_d_ssd1607_200x200.c also (correctly) states mode 0 (`/* active high, rising edge */`).
In other display constructors this results in spi mode 0.
Only for this display you use 2. Seems to be an error.

FrankX0, 2019-01-20:
Thanks Oliver! Glad I finally have a working display again!

That brings me to the original topic of this thread: what about implementing partial refresh for e-paper displays?
This enhances the user experience quite a bit in my view:
- There is no need for the entire display to flicker black and white during each update
- Update rate is considerably faster: from 1200ms to 300ms

For my display the only thing required is using a different LUT during refresh.
So we need to be able to send a full (default) refresh and partial refresh from u8g2.sendBuffer, as this already implements the refresh of the display.

Please let me know your opinion.

olikraus, 2019-01-24:
> For my display the only thing required is using a different LUT during refresh.
> So we need to be able to send a full (default) refresh and partial refresh from u8g2.sendBuffer, as this already implements the refresh of the display.

Should be possible, but I need some time to look at this.

olikraus, 2019-01-30:
Inside a .ino file you can send a custom command sequence to the display by using the following command (u8g2 is the u8g2 object):
```
u8x8_cad_SendSequence(u8g2.getU8x8(), sequence);
```
The "sequence" variable is a byte array with special commands. It may look like this:
A SSD1606 LUT could look like this:
```
const uint8_t sequence[] = 
  U8X8_START_TRANSFER(),             	/* enable chip, delay is part of the transfer start */
  U8X8_C(0x32),	/* write LUT register, place LUT entries after this command*/

  /* wavefrom part of the LUT: absolute LUT... this will always force the destination color */
  U8X8_A4(0x00,0x00,0x00,0x55),  /* step 0 */
  U8X8_A4(0x00,0x00,0x55,0x55),	/* step 1 */
  U8X8_A4(0x00,0x55,0x55,0x55),
  U8X8_A4(0xAA,0xAA,0xAA,0xAA),
  U8X8_A4(0x15,0x15,0x15,0x15),
  U8X8_A4(0x05,0x05,0x05,0x05),
  U8X8_A4(0x01,0x01,0x01,0x01),
  U8X8_A4(0x00,0x00,0x00,0x00),
  U8X8_A4(0x00,0x00,0x00,0x00),
  U8X8_A4(0x00,0x00,0x00,0x00),
  U8X8_A4(0x00,0x00,0x00,0x00),
  U8X8_A4(0x00,0x00,0x00,0x00),
  U8X8_A4(0x00,0x00,0x00,0x00),
  U8X8_A4(0x00,0x00,0x00,0x00),
  U8X8_A4(0x00,0x00,0x00,0x00),
  U8X8_A4(0x00,0x00,0x00,0x00),
  U8X8_A4(0x00,0x00,0x00,0x00),
  U8X8_A4(0x00,0x00,0x00,0x00),
  U8X8_A4(0x00,0x00,0x00,0x00),
  U8X8_A4(0x00,0x00,0x00,0x00),	/* step 19 */
  
  /* timing part of the LUT */
  U8X8_A8(0x22,0xFB,0x22,0x1B,0x00,0x00,0x00,0x00),
  U8X8_A(0x00),U8X8_A(0x00),
  U8X8_END_TRANSFER(),             	/* disable chip */
  U8X8_END()             			/* end of sequence */
};
```
The sequence must start with `U8X8_START_TRANSFER` and must end with the commands `U8X8_END_TRANSFER` and `U8X8_END`.

Here is a list of all commands, which can be used in the sequence:
```
U8X8_C(c0) 					Send a single command
U8X8_A(a0)					Send a single argument (after a command)
U8X8_CA(c0,a0)				Send command c0 followed by argument a0
U8X8_CAA(c0,a0,a1)			Send one command and two args
U8X8_CAAA(c0,a0,a1, a2)		Send one command and three args
U8X8_CAAAA(c0,a0,a1,a2,a3)		Send one command and four args
U8X8_AAC(a0,a1,c0)			Send two args and then one command
U8X8_D1(d0)					Send a byte as data

U8X8_A4(a0,a1,a2,a3)			Send four args
U8X8_A8(a0,a1,a2,a3,a4,a5,a6,a7)	Send 8 args

U8X8_START_TRANSFER()			Start a transfer
U8X8_END_TRANSFER()			End a transfer
U8X8_DLY(m)					Delay by m Milliseconds
U8X8_END()					End of sequence marker: Must be the last element of a sequence
```

FrankX0, 2019-02-04:
Thanks for this intermediate solution.
But this will probably not work, as `sendBuffer` and `clearBuffer` already contain a (full) `refreshDisplay` call.
To properly implement a partial refresh IMO:

- `sendBuffer` and `clearBuffer` would have to be updated with an optional parameter to specify a partial refresh, or
- `refreshDisplay` should be removed from `sendBuffer` and `clearBuffer` to be able to execute a full or partial refresh.

Please let me have your opinion.

olikraus, 2019-02-09:
added member function `updateDisplay(void)` which is the same as sendBuffer, just without sending the refresh message.

olikraus, 2019-02-09:
will close this for now...

closed 2019-02-09 by olikraus