olikraus/u8g2 #639: AVR SSD1306 I2C 0x78

thorstad, 2018-06-19:
Hi, there are not much documentation about using I2C with AVR. So I tried my best to use the example u8x8_avr_gpio_and_delay with outputing the pins. My problem is that my OLED does not lit up.

`#define DISPLAY_CLK_DIR DDRC
#define DISPLAY_CLK_PORT PORTC
#define DISPLAY_CLK_PIN 5

#define DISPLAY_DATA_DIR DDRC
#define DISPLAY_DATA_PORT PORTC
#define DISPLAY_DATA_PIN 4

uint8_t u8x8_avr_gpio_and_delay(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
	// Re-use library for delays

	switch(msg)
	{
		case U8X8_MSG_GPIO_AND_DELAY_INIT:  // called once during init phase of u8g2/u8x8
		DISPLAY_CLK_DIR |= 1<<DISPLAY_CLK_PIN;
		DISPLAY_DATA_DIR |= 1<<DISPLAY_DATA_PIN;

		break;              // can be used to setup pins
		case U8X8_MSG_GPIO_I2C_CLOCK:        // Clock pin: Output level in arg_int
		if(arg_int)
		DISPLAY_CLK_PORT |= (1<<DISPLAY_CLK_PIN);
		else
		DISPLAY_CLK_PORT &= ~(1<<DISPLAY_CLK_PIN);
		break;
		case U8X8_MSG_GPIO_I2C_DATA:        // Data pin: Output level in arg_int
		if(arg_int)
		DISPLAY_DATA_PORT |= (1<<DISPLAY_DATA_PIN);
		else
		DISPLAY_DATA_PORT &= ~(1<<DISPLAY_DATA_PIN);
		break;

		default:
		if (u8x8_avr_delay(u8x8, msg, arg_int, arg_ptr))	// check for any delay msgs
		return 1;
		u8x8_SetGPIOResult(u8x8, 1);      // default return value
		break;
	}
	return 1;
}

int main (void)
{
	
	CLKPR = (1 << CLKPCE); // enable a change to CLKPR
	CLKPR = 0;
	
	//u8g2_SetI2CAddress(&u8g2, 0x3C*2);
	u8g2_Setup_ssd1306_128x64_noname_f(&u8g2, U8G2_R0, u8x8_byte_sw_i2c, u8x8_avr_gpio_and_delay);
	u8g2_InitDisplay(&u8g2);
	u8g2_SetPowerSave(&u8g2, 0);
	
	u8g2_ClearBuffer(&u8g2);
	u8g2_SetFont(&u8g2, u8g2_font_ncenB14_tr);
	u8g2_DrawStr(&u8g2, 1, 18, "U8g2 on AVR");
	u8g2_SendBuffer(&u8g2);
}
`

Can you tell me what's wrong? I have tried different addresses, even though it says 0x78 on the back of the OLED.
Much appreciated, Thorstad.

olikraus, 2018-06-19:
hmm... everything seems to be correct, except setting the I2C address. This should be done after u8g2_Setup: u8g2_Setup setup procedure will reset the I2C:
```
	u8g2_Setup_ssd1306_128x64_noname_f(&u8g2, U8G2_R0, u8x8_byte_sw_i2c, u8x8_avr_gpio_and_delay);
	u8g2_SetI2CAddress(&u8g2, 0x3C*2);
	u8g2_InitDisplay(&u8g2);
	u8g2_SetPowerSave(&u8g2, 0);
``` 

Your code does not include u8x8_avr_delay() but it is there, right?

thorstad, 2018-06-19:
It is there, but I found out I havent done anything to the DELAY_NANO and DELAY_100NANO.
```
uint8_t u8x8_avr_delay(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
	uint8_t cycles;

	switch(msg)
	{
		case U8X8_MSG_DELAY_NANO:     // delay arg_int * 1 nano 
		// At 20Mhz, each cycle is 50ns, the call itself is slower.
		break;
		case U8X8_MSG_DELAY_100NANO:    // delay arg_int * 100 nano seconds
		// Approximate best case values...
		#define CALL_CYCLES 26UL
		#define CALC_CYCLES 4UL
		#define RETURN_CYCLES 4UL
		#define CYCLES_PER_LOOP 4UL

		cycles = (100UL * arg_int) / (P_CPU_NS * CYCLES_PER_LOOP);

		if(cycles > CALL_CYCLES + RETURN_CYCLES + CALC_CYCLES)
		break;

		__asm__ __volatile__ (
		"1: sbiw %0,1" "\n\t" // 2 cycles
		"brne 1b" : "=w" (cycles) : "0" (cycles) // 2 cycles
		);
		break;
		case U8X8_MSG_DELAY_10MICRO:    // delay arg_int * 10 micro seconds
		for(int i=0 ; i < arg_int ; i++)
		_delay_us(10);
		break;
		case U8X8_MSG_DELAY_MILLI:      // delay arg_int * 1 milli second
		for(int i=0 ; i < arg_int ; i++)
		_delay_ms(1);
		break;
		default:
		return 0;
	}
	return 1;
}
```

thorstad, 2018-06-19:
@olikraus do you have any u8g2 I2C example that might help?

thorstad, 2018-06-19:
[image]

Looks like the timing is off by 25µs, and if I have a led blinking every _delay_ms(1000) is will be 0.05ms faster and it will be noticeable after a few 20 blinks. So this might be the problem why I cant communicate?

olikraus, 2018-06-19:
Ah, yes, you need to implement U8X8_MSG_DELAY_I2C:

See here:
https://github.com/olikraus/u8g2/blob/master/sys/arm/stm32l031x6/u8x8_test/u8x8cb.c#L58

thorstad, 2018-06-19:
Well I do not know if I implementet it correct:
_delay_us((arg_int<=2?5:1));
I get an error "_builtin_avr_delay_cycles expects a compile time integer constant"

olikraus, 2018-06-19:
Maybe you need to change this:
```
if ( arg_int < 2 )
{
 _delay_us(5);
}
else
{
 _delay_us(1);
}
```

thorstad, 2018-06-19:
The screen lits up!! I can only see abit of it, rest of it are only dots.

thorstad, 2018-06-20:
Video of OLED: https://www.youtube.com/watch?v=89tMRId76BQ&feature=youtu.be

olikraus, 2018-06-20:
With your current implementation u8g2 emulates I2C. You need to implement I2C for your HW subsystem. 
https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform#hardware-i2c-communication

There is also a buildin itoa function. 
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_u16toa.c#L62

thorstad, 2018-06-20:
The same HW code were in the library, so I blurred out the other so it uses the one. But now my OLED does not light up.
```
void i2c_transfer(u8x8_t *u8x8, uint8_t adr, uint8_t cnt, uint8_t *data)
{
  uint8_t i;
  i2c_start(u8x8);
  i2c_write_byte(u8x8, adr);
  for( i = 0; i < cnt; i++ )
    i2c_write_byte(u8x8, data[i]);
  i2c_stop(u8x8);  
}

uint8_t u8x8_byte_sw_i2c(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
  static uint8_t buffer[32];		/* u8g2/u8x8 will never send more than 32 bytes between START_TRANSFER and END_TRANSFER */
  static uint8_t buf_idx;
  uint8_t *data;
 
  switch(msg)
  {
    case U8X8_MSG_BYTE_SEND:
      data = (uint8_t *)arg_ptr;      
      while( arg_int > 0 )
      {
	buffer[buf_idx++] = *data;
	data++;
	arg_int--;
      }      
      break;
    case U8X8_MSG_BYTE_INIT:
      /* add your custom code to init i2c subsystem */
	  i2c_init(u8x8);
      break;
    case U8X8_MSG_BYTE_SET_DC:
      /* ignored for i2c */
      break;
    case U8X8_MSG_BYTE_START_TRANSFER:
      buf_idx = 0;
      break;
    case U8X8_MSG_BYTE_END_TRANSFER:
      i2c_transfer(u8x8, u8x8_GetI2CAddress(u8x8) >> 1, buf_idx, buffer);
      break;
    default:
      return 0;
  }
  return 1;
}
```

thorstad, 2018-06-20:
I changed the code a little to this:
```
case U8X8_MSG_BYTE_END_TRANSFER:
     //i2c_transfer(u8x8, u8x8_GetI2CAddress(u8x8) >> 1, buf_idx, buffer);
	 
	 i2c_start(u8x8);
	 i2c_write_byte(u8x8, u8x8_GetI2CAddress(u8x8));
	 for( i = 0; i < buf_idx; i++ )
	 i2c_write_byte(u8x8, buffer[i]);
	 i2c_stop(u8x8);
      break;
    default:
      return 0;
  }
```
My oled is on, but it is the same speed,

olikraus, 2018-06-20:
u8x8_byte.c does not include any hardware accelerated I2C code.
Infact you just call the i2c emulation procedures from u8g2. Excactly the same procedures are called from the SW_I2C constructors.

So you still use sw emulated I2C and obvisously you it will have the same speed.

U8X8_MSG_BYTE_INIT
Turn on master transmit mode of the AVR I2C Subsystem

U8X8_MSG_BYTE_SEND
transfer data via the AVR I2C subsystem.

etc...

thorstad, 2018-06-20:
That makes sense, but I have no idea how to do that.

thorstad, 2018-06-20:
I am using i2cmaster.c and i2cmaster.S

olikraus, 2018-06-20:
You can use the _F_ constructor/ _f_ setup procedure.

thorstad, 2018-06-20:
Im trying to use "u8g2_GetFontAscent" I get undeclared first use in function.

olikraus, 2018-06-21:
You already use this.... noname_f

thorstad, 2018-06-21:
Thank you very much, now it works fast update rate is perfect.
```
Program Memory Usage 	:	7750 bytes   23,7 % Full
Data Memory Usage 	:	1307 bytes   63,8 % Full
```
I dont think the transmitter will use much memory so all is good!

olikraus, 2018-06-21:
Can you share your code? Maybe other AVR are interested...

thorstad, 2018-06-22:
```
#undef F_CPU
#define F_CPU 8000000UL

#include <util/delay.h>
#include <avr/interrupt.h>
#include <avr/io.h>
#include <asf.h>
#include <u8g2/u8g2.h>
#include <i2cmaster.h>
#include "compiler.h"

#define P_CPU_NS (1000000000 / F_CPU)
#define SSD1306  0x78 

#define DISPLAY_CLK_DIR DDRC
#define DISPLAY_CLK_PORT PORTC
#define DISPLAY_CLK_PIN 5

#define DISPLAY_DATA_DIR DDRC
#define DISPLAY_DATA_PORT PORTC
#define DISPLAY_DATA_PIN 4

uint8_t u8x8_avr_delay(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr);
uint8_t u8x8_avr_gpio_and_delay(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr);
uint8_t avr_hw_i2c(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr);

uint8_t u8x8_avr_delay(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
	uint8_t cycles;

	switch(msg)
	{
		case U8X8_MSG_DELAY_NANO:     // delay arg_int * 1 nano
		// At 20Mhz, each cycle is 50ns, the call itself is slower.
		break;
		case U8X8_MSG_DELAY_100NANO:    // delay arg_int * 100 nano seconds
		
		
		// Approximate best case values...
		#define CALL_CYCLES 26UL
		#define CALC_CYCLES 4UL
		#define RETURN_CYCLES 4UL
		#define CYCLES_PER_LOOP 4UL

		cycles = (100UL * arg_int) / (P_CPU_NS * CYCLES_PER_LOOP);

		if(cycles > CALL_CYCLES + RETURN_CYCLES + CALC_CYCLES)
		break;

		__asm__ __volatile__ (
		"1: sbiw %0,1" "\n\t" // 2 cycles
		"brne 1b" : "=w" (cycles) : "0" (cycles) // 2 cycles
		);
		break;
		case U8X8_MSG_DELAY_10MICRO:    // delay arg_int * 10 micro seconds
		for(int i=0 ; i < arg_int ; i++)
		_delay_us(10);
		break;
		case U8X8_MSG_DELAY_MILLI:      // delay arg_int * 1 milli second
		for(int i=0 ; i < arg_int ; i++)
		_delay_ms(1);
		break;
		default:
		return 0;
	}
	return 1;
}

uint8_t u8x8_avr_gpio_and_delay(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
	// Re-use library for delays

	switch(msg)
	{
		case U8X8_MSG_GPIO_AND_DELAY_INIT:  // called once during init phase of u8g2/u8x8
		DISPLAY_CLK_DIR |= 1<<DISPLAY_CLK_PIN;
		DISPLAY_DATA_DIR |= 1<<DISPLAY_DATA_PIN;
		break;              // can be used to setup pins
		case U8X8_MSG_DELAY_I2C:
		if ( arg_int < 2 )
		{
			_delay_us(5);
		}
		else
		{
			_delay_us(1);
		}
		/* delay by 5 us when arg_int is 1, or 1.25us otherwise */
		break;
		case U8X8_MSG_GPIO_I2C_CLOCK:        // Clock pin: Output level in arg_int
		if(arg_int)
		DISPLAY_CLK_PORT |= (1<<DISPLAY_CLK_PIN);
		else
		DISPLAY_CLK_PORT &= ~(1<<DISPLAY_CLK_PIN);
		break;
		case U8X8_MSG_GPIO_I2C_DATA:        // Data pin: Output level in arg_int
		if(arg_int)
		DISPLAY_DATA_PORT |= (1<<DISPLAY_DATA_PIN);
		else
		DISPLAY_DATA_PORT &= ~(1<<DISPLAY_DATA_PIN);
		break;
		default:
		if (u8x8_avr_delay(u8x8, msg, arg_int, arg_ptr))	// check for any delay msgs
		return 1;
		u8x8_SetGPIOResult(u8x8, 1);      // default return value
		break;
	}
	return 1;
}

uint8_t avr_hw_i2c(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
	static uint8_t buffer[32];		/* u8g2/u8x8 will never send more than 32 bytes between START_TRANSFER and END_TRANSFER */
	static uint8_t buf_idx;
	uint8_t *data;
	uint8_t i;
	switch(msg)
	{
		
		case U8X8_MSG_BYTE_SEND:
		data = (uint8_t *)arg_ptr;
		while( arg_int > 0 )
		{
			buffer[buf_idx++] = *data;
			data++;
			arg_int--;
		}
		break;
		case U8X8_MSG_BYTE_INIT:
		/* add your custom code to init i2c subsystem */
		i2c_init();
		
		break;
		case U8X8_MSG_BYTE_SET_DC:
		/* ignored for i2c */
		break;
		case U8X8_MSG_BYTE_START_TRANSFER:
		//i2c_start_wait(SSD1306+I2C_WRITE);
		buf_idx = 0;
		break;
		case U8X8_MSG_BYTE_END_TRANSFER:
		
		i2c_start_wait(SSD1306+I2C_WRITE);     // set device address and write mode
		// i2c_write(&u8x8);
		for( i = 0; i < buf_idx; i++ )
		i2c_write(buffer[i]);
		i2c_stop();
		
		break;
		default:
		return 0;
	}
	return 1;
}

u8g2_t u8g2;

int main (void)
{
	u8g2_Setup_ssd1306_i2c_128x64_noname_f(&u8g2, U8G2_R0, avr_hw_i2c, u8x8_avr_gpio_and_delay);
	u8g2_SetI2CAddress(&u8g2, 0x78);
	u8g2_InitDisplay(&u8g2);
	u8g2_SetPowerSave(&u8g2, 0);

	while(1){

			u8g2_ClearBuffer(&u8g2);
			
			u8g2_SetFont(&u8g2, u8g2_font_fur35_tn);
			u8g2_SetFontRefHeightText(&u8g2);
			u8g2_SetFontPosTop(&u8g2);

			u8g2_DrawStr(&u8g2, 0, 14, "AVR");
	
			u8g2_SendBuffer(&u8g2);	
	}
}

```

olikraus, 2018-06-22:
Thanks a lot. 
Is "i2cmaster.h" part of Atmel Studio? I mean, from where did you get i2cmaster.h?

thorstad, 2018-06-22:
Nope, got it from:
http://homepage.hispeed.ch/peterfleury/doxygen/avr-gcc-libraries/group__pfleury__ic2master.html
But you need to disable one file from the u8x8 thing because of the i2cmaster.

closed 2018-08-04 by olikraus