olikraus/u8g2 #640: Including support for Arabic languages

RobotGB, 2018-06-19:
Hello Oliver,
I would like to include the function of printing Arabic-script text (e.g. Persian, Urdu, etc.) to your library.
I've written some code, which converts the individual characters (which look like ش م ع) into their joined forms (which look like شمع). This means that usual typed Arabic (which isn't joined on its own, it is joined during the display process by the operating system) can be input without the need for prior conversion.
Sorry I'm not a very good programmer so I had to use Strings (which take up lots of memory space). You could replace this with char* if you can find a way to use the equivalent of .replace().
Lastly, can you modify the print() function so that it automatically places diacritics above the letter e.g. ´ a = á should be done during display.
Here is my code (if you can find a way to place the functions together that would be more convenient):
```
const unsigned long joining[23][5] = { //use UTF-8 codes
  //original, isolated, end, beginning, middle
  //hookAfter, hookBefore: FF, FT, TF, TT
  {0xD8A8,0xEFBA8F,0xEFBA90,0xEFBA91,0xEFBA92}, //ba'
  {0xD8AA,0xEFBA95,0xEFBA96,0xEFBA97,0xEFBA98}, //ta'
  {0xD8AB,0xEFBA99,0xEFBA9A,0xEFBA9B,0xEFBA9C}, //tha'
  {0xD8AC,0xEFBA9D,0xEFBA9E,0xEFBA9F,0xEFBAA0}, //jim
  {0xD8AD,0xEFBAA1,0xEFBAA2,0xEFBAA3,0xEFBAA4}, //hard ḥa'
  {0xD8AE,0xEFBAA5,0xEFBAA6,0xEFBAA7,0xEFBAA8}, //kha'
  {0xD8B3,0xEFBAB1,0xEFBAB2,0xEFBAB3,0xEFBAB4}, //sin
  {0xD8B4,0xEFBAB5,0xEFBAB6,0xEFBAB7,0xEFBAB8}, //shin
  {0xD8B5,0xEFBAB9,0xEFBABA,0xEFBABB,0xEFBABC}, //sad
  {0xD8B6,0xEFBABD,0xEFBABE,0xEFBABF,0xEFBB80}, //dhad
  {0xD8B7,0xEFBB81,0xEFBB82,0xEFBB83,0xEFBB84}, //hard ṭa'
  {0xD8B8,0xEFBB85,0xEFBB86,0xEFBB87,0xEFBB88}, //hard dha'
  {0xD8B9,0xEFBB89,0xEFBB8A,0xEFBB8B,0xEFBB8C}, //'ayn
  {0xD8BA,0xEFBB8D,0xEFBB8E,0xEFBB8F,0xEFBB90}, //ghayn
  {0xD981,0xEFBB91,0xEFBB92,0xEFBB93,0xEFBB94}, //fa'
  {0xD982,0xEFBB95,0xEFBB96,0xEFBB97,0xEFBB98}, //qaf
  {0xD983,0xEFBB99,0xEFBB9A,0xEFBB9B,0xEFBB9C}, //kaf
  {0xD984,0xEFBB9D,0xEFBB9E,0xEFBB9F,0xEFBBA0}, //lam
  {0xD985,0xEFBBA1,0xEFBBA2,0xEFBBA3,0xEFBBA4}, //mim
  {0xD986,0xEFBBA5,0xEFBBA6,0xEFBBA7,0xEFBBA8}, //nun
  {0xD987,0xEFBBA9,0xEFBBAA,0xEFBBAB,0xEFBBAC}, //ha'
  {0xD98A,0xEFBBB1,0xEFBBB2,0xEFBBB3,0xEFBBB4}, //ya'
  {0xD8AC,0xEFBA89,0xEFBA8A,0xEFBA8B,0xEFBA8C}  //ya' with hamza above
};

const unsigned long separating[14][3] = { //use UTF-8 codes
  //original, isolated and final
  {0xD8A7,0xEFBA8D,0xEFBA8E}, //'alif
  {0xD8AF,0xEFBAA9,0xEFBAAA}, //dal
  {0xD8B0,0xEFBAAB,0xEFBAAC}, //dhal
  {0xD8B1,0xEFBAAD,0xEFBAAE}, //ra'
  {0xD8B2,0xEFBAAF,0xEFBAB0}, //zay
  {0xD988,0xEFBBAD,0xEFBBAE}, //waw
  {0xD8A2,0xEFBA81,0xEFBA82}, //'alif maddah
  {0xD8A9,0xEFBA93,0xEFBA94}, //ta' marbuṭah
  {0xD989,0xEFBBAF,0xEFBBB0}, //'alif maksurah
  {0xD9B1,0xEFAD90,0xEFAD91}, //'alif waslah
  {0xD8A1,0xEFBA80,0xEFBA80}, //hamza
  {0xD8A3,0xEFBA83,0xEFBA84}, //'alif with hamza above
  {0xD8A4,0xEFBA85,0xEFBA86}, //waw hamza
  {0xD8A5,0xEFBA87,0xEFBA88}  //'alif with hamza below
};

const unsigned long diacritics[22] { //use utf8 codes
//  0xD890, //sallallahou alayhe wassallam
//  0xD891, //alayhe assallam
//  0xD892, //rahmatullah alayhe
//  0xD893, //radhi allahou anhu
  0xD894, //takhallus
  0xD895, //small high tah
  0xD896, //'alif lam ya'
  0xD897, //small high zay
  0xD898, //fatha
  0xD899, //dhamma
  0xD900, //kasra
  0xD98B, //fathatan
  0xD98C, //dhammatan
  0xD98D, //kasratan
  0xD98E, //fatha
  0xD98F, //dhamma
[... 207 lines cut ...]
    if (separating[x][0] == toFind) {
      return x;
      break;
    }
  }
  return -1;
}

bool isDiacritic(unsigned long toCheck) {
  if (toCheck == 0) {
    return false;
  }
  else {
    for (int i = 0; i < diacriticSize; i++) {
      if (toCheck == diacritics[i]) {
        return true;
        break;
      }
    }
    return false;
  }
}
```
Lastly, if you want to use the function with other languages e.g. Persian then perhaps someone who knows it can add more characters.
Many thanks.

olikraus, 2018-06-19:
Can you describe more clearly what kind of transformation you do?
If i take latin (and my own language), then I see the following types:

1) Compression/Overlay. Two chars are printed more tight together, like "A" and "W". If combinded they become more close or: "f" and "f", if combinde they become closer: ff 
unicode1, unicode2 --> kerning correction

2) Two characters become a new character: so to say: f and f may become a new unicode (indeed the is a unicode for ff). Same is true for diacritics, where " and a may become ä.
unicode1, unicode2 --> unicode3

What are other types in your program? Can we have a list of transformations which are required?

RobotGB, 2018-06-19:
In languages like Arabic, the letter is modified depending on where it should go in a word.
Usually, when you print to the display ordinary Arabic text, each character is printed individually and is **not joined together**, whereas a computer would join it automatically on the screen but not modifying the actual data of each character.
What my code does is that it **modifies each character of the string and it changes it so that it is joined** (what a computer would show if you were to type a word). This uses the Unicode block Arabic Presentation Forms-B, whereas the input string uses the block Arabic.
This is different from kerning because kerning **only changes the distance** between the characters - the code changes the **appearance** as well.
The transformations are expressed as the UTF-8 values at the top of the code. You can see them at https://en.wikipedia.org/wiki/Arabic_script_in_Unicode.
The appearance changes like in the picture.
[image]
You can see that the character changes shape depending on the position it takes:
Final = at the end of the word
Medial = at the middle of the word
Initial = at the beginning of the word
Isolated = the character alone (not part of a word).

The only time that kerning is needed is in order to overlay diacritics (like the image below, the diacritics are the small characters at the top of the letter).
[image]

There are also some ligatures in the same way that f + f + i = ffi, ل + ا = لا.
By the way, Arabic text is written in right-to-left whereas numbers are written in the same way as English (left-to-right). Sorry if that complicates things.

Thanks.

idreamsi, 2018-06-20:
Hi,
Thanks @RobotGB 
I've already created a library that's easy to use.
https://github.com/idreamsi/arduino-persian-reshaper

olikraus, 2018-06-20:
@idreamsi 
Nice, but it looks at least as complicated as the code from @RobotGB 

But actually I think about a more generic solution and i wanted to know the specification for your.
I mean, I just see the code, but what is the purpose of your code. How would you describe what you do in the code.

idreamsi, 2018-06-20:
@olikraus
Unfortunately, Persian, Arabic, and ... are very complex and their implementation is time consuming.
You need to be familiar with these languages.

Also, there is a bug in your library and I can not use u8g2.drawPixel(x, y)
Function written by me, requires **drawPixel(x, y)** but my attempt to convert to your library was unsuccessful.

Even the following code has no output!
```
#include <Arduino.h>
#include <U8g2lib.h>

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif

U8G2_SSD1306_128X64_NONAME_1_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 12, /* dc=*/ 4, /* reset=*/ 6);	// Arduboy (Production, Kickstarter Edition)

void setup(void) {
u8g2.begin();  
u8g2.firstPage();
u8g2.setColorIndex(1);
u8g2.drawPixel(100,10);
u8g2.nextPage();
}
```

RobotGB, 2018-06-20:
What my (and I'm assuming @idreamsi's too) code does is that it converts each letter into its joined form (which takes up a different Unicode code point).
You can imagine it like writing joined up handwriting - each letter joins to the previous and next character differently depending on the position in the word (if the letter is alone, at the beginning, at the middle or at the end).
If the character is at the beginning of the word, it is not joined before, but if at the middle, it is joined before and after. Likewise, if the character is at the end, it is joined before and not after.
Of course, it is more complex than this, because different letters behave differently, some might not join after etc.

A simplified process would be:
1. Create a table which has different categories of character

- letters which join to other characters (which is called _joining_ in my code)
- letters which join only before and not after (called _separating_ in my code)
- diacritics which serve as short vowels (called _harakat_) and others which serve to aid in reading (called _i'jam_)

2. Read each character, the one before it and the one after it, then increment to the next set of characters (ignore diacritics when finding the characters because they do not serve in attaching each character, they go above and are not part of the text).
I've named these as _'previous'_, _'letter'_ and _'next'_ in my code.

3. Look into the table to find if the previous letter is a joining character or a separating character (using functions _foundJoining_ and _foundSeparating_). If so, replace the current character _letter_ so that it is joined correctly based on whether it is joined before and after (_hookBefore_ and hookAfter).
Then add the current modified letter to the end of the buffer, which will be returned when the variable is called.

4. Repeat for the next character until the end of the String is reached.

RobotGB, 2018-06-20:
Also @idreamsi, the setPixel code works on my display (SSD1306) but you will need to change `setPixel(x, y, color)` in your code to 
```cpp
u8g2.setDrawColor(color); //white is 1 and black is 0
u8g2.drawPixel(x, y);
```

olikraus, 2018-06-20:
> Function written by me, requires drawPixel(x, y) but my attempt to convert to your library was unsuccessful.

In your example, the loop itself is missing, so it can not work. As an alternative: Use the clear buffer/send buffer approach with the F constructor

> You need to be familiar with these languages.

Well, yes infact I see that his will be too much for u8g2 project.

Thanks a lot for all your explainations. 
Actually what I look for is something like this: A procedure, which gets unicode glyphs as input and outputs a new list of unicode glyphs.
In between the arabic/persian transformation happens.
In order to add this to u8g2 it should be some plain c code without side effects.

Anyhow, my knowledge is indeed not deep enough... so I keep this issue here so that other people can benefit from your solutions.

olikraus, 2018-06-20:
Links: https://github.com/font-store/persian-reshaper

RobotGB, 2018-07-01:
The only code that would complete the process is reversing all Latin characters. Because the code is designed to handle right-to-left text, this means that Latin text will also be displayed in reverse e.g. Hello becomes olleH when displayed with this code.
Also, apparently, 0xEF as a char in String becomes 0xFFFFFFEF on boards like the Arduino Uno and Mega. Can anyone find a workaround?
Thanks.

closed 2018-08-04 by olikraus

idreamsi, 2019-08-18:
> Also, after passing the String through the function arabicToPresentationForms, I've created some code to test the display results. Sorry, it's a hodgepodge at best, and it only works with constant width fonts.
> 
> 1. Set the value of width to be the constant width of the font
> 2. Set the value of y to be the height that the text is drawn at.
>    I might add a function which finds the width of the text afterwards, and definitely clean this up.
>    (the code that I updated is at the bottom, and this includes the function from before)
> 
> ```c++
> #include <SPI.h>
> 
> #include <Wire.h>
> 
> 
> #include <U8x8lib.h>
> #include <U8g2lib.h>
> U8G2_SSD1306_128X64_NONAME_F_SW_I2C u8g2(U8G2_R0, /* clock=*/ SCL, /* data=*/ SDA, /* reset=*/ U8X8_PIN_NONE);   // All Boards without Reset of the Display
> 
> 
> const int joiningSize PROGMEM = 22; //change this when expanding joining table to include
> //length of joining
> const int separatingSize PROGMEM = 14;
> const int diacriticSize PROGMEM = 18;
> int x = 118;
> 
> const unsigned long joining[][5] PROGMEM = { //use UTF-8 codes
>   //original, isolated, end, beginning, middle
>   //hookAfter, hookBefore: FF, FT, TF, TT
>   {0xD8A8, 0xEFBA8F, 0xEFBA90, 0xEFBA91, 0xEFBA92}, //ba'
>   {0xD8AA, 0xEFBA95, 0xEFBA96, 0xEFBA97, 0xEFBA98}, //ta'
>   {0xD8AB, 0xEFBA99, 0xEFBA9A, 0xEFBA9B, 0xEFBA9C}, //tha'
>   {0xD8AC, 0xEFBA9D, 0xEFBA9E, 0xEFBA9F, 0xEFBAA0}, //jim
>   {0xD8AD, 0xEFBAA1, 0xEFBAA2, 0xEFBAA3, 0xEFBAA4}, //hard ḥa'
>   {0xD8AE, 0xEFBAA5, 0xEFBAA6, 0xEFBAA7, 0xEFBAA8}, //kha'
>   {0xD8B3, 0xEFBAB1, 0xEFBAB2, 0xEFBAB3, 0xEFBAB4}, //sin
>   {0xD8B4, 0xEFBAB5, 0xEFBAB6, 0xEFBAB7, 0xEFBAB8}, //shin
>   {0xD8B5, 0xEFBAB9, 0xEFBABA, 0xEFBABB, 0xEFBABC}, //sad
>   {0xD8B6, 0xEFBABD, 0xEFBABE, 0xEFBABF, 0xEFBB80}, //dhad
>   {0xD8B7, 0xEFBB81, 0xEFBB82, 0xEFBB83, 0xEFBB84}, //hard ṭa'
>   {0xD8B8, 0xEFBB85, 0xEFBB86, 0xEFBB87, 0xEFBB88}, //hard dha'
>   {0xD8B9, 0xEFBB89, 0xEFBB8A, 0xEFBB8B, 0xEFBB8C}, //'ayn
>   {0xD8BA, 0xEFBB8D, 0xEFBB8E, 0xEFBB8F, 0xEFBB90}, //ghayn
>   {0xD981, 0xEFBB91, 0xEFBB92, 0xEFBB93, 0xEFBB94}, //fa'
>   {0xD982, 0xEFBB95, 0xEFBB96, 0xEFBB97, 0xEFBB98}, //qaf
>   {0xD983, 0xEFBB99, 0xEFBB9A, 0xEFBB9B, 0xEFBB9C}, //kaf
>   {0xD984, 0xEFBB9D, 0xEFBB9E, 0xEFBB9F, 0xEFBBA0}, //lam
>   {0xD985, 0xEFBBA1, 0xEFBBA2, 0xEFBBA3, 0xEFBBA4}, //mim
>   {0xD986, 0xEFBBA5, 0xEFBBA6, 0xEFBBA7, 0xEFBBA8}, //nun
>   {0xD987, 0xEFBBA9, 0xEFBBAA, 0xEFBBAB, 0xEFBBAC}, //ha'
>   {0xD98A, 0xEFBBB1, 0xEFBBB2, 0xEFBBB3, 0xEFBBB4}, //ya'
>   {0xD8AC, 0xEFBA89, 0xEFBA8A, 0xEFBA8B, 0xEFBA8C} //ya' with hamza above
> };
> 
> const unsigned long separating[][3] PROGMEM = { //use UTF-8 codes
>   //original, isolated and final
>   {0xD8A7, 0xEFBA8D, 0xEFBA8E}, //'alif
>   {0xD8AF, 0xEFBAA9, 0xEFBAAA}, //dal
>   {0xD8B0, 0xEFBAAB, 0xEFBAAC}, //dhal
>   {0xD8B1, 0xEFBAAD, 0xEFBAAE}, //ra'
>   {0xD8B2, 0xEFBAAF, 0xEFBAB0}, //zay
>   {0xD988, 0xEFBBAD, 0xEFBBAE}, //waw
>   {0xD8A2, 0xEFBA81, 0xEFBA82}, //'alif maddah
>   {0xD8A9, 0xEFBA93, 0xEFBA94}, //ta' marbuṭah
>   {0xD989, 0xEFBBAF, 0xEFBBB0}, //'alif maksurah
>   {0xD9B1, 0xEFAD90, 0xEFAD91}, //'alif waslah
>   {0xD8A1, 0xEFBA80, 0xEFBA80}, //hamza
>   {0xD8A3, 0xEFBA83, 0xEFBA84}, //'alif with hamza above
>   {0xD8A4, 0xEFBA85, 0xEFBA86}, //waw hamza
>   {0xD8A5, 0xEFBA87, 0xEFBA88} //'alif with hamza below
> };
> 
[... 520 lines cut ...]
    }
    return false;
}

void setup() {
  Serial.begin(9600);
  u8g2.begin();
  String msg = "سلام";
  msg = Persianize(msg);
  
  u8g2.begin();
  u8g2.enableUTF8Print();    // enable UTF8 support for the Arduino print() function
  //u8g2.setFont(u8g2_font_iranian_sans_12_t_all);
  u8g2.setFont(u8g2_font_samim_fd_14_t_all);
  u8g2.firstPage();
  do {
    u8g2.setCursor(10, 40);
    u8g2.print(ms2);
  
  } while ( u8g2.nextPage() );

}

```
Source: https://www.codeproject.com/Articles/3529/Arabic-Support-for-Windows-CE