olikraus/u8g2 #641: Help needed

Nomatiti, 2018-06-21:
I have this round 1.2" inch oled EA W128128-XRLW; [https://www.mouser.fi/ProductDetail/ELECTRONIC-ASSEMBLY/EA-W128128-XRLW?qs=sGAEpiMZZMshRHSPqbouvJWDKQb1KqZ10IgVqkv8Swn4rpO3VjZ11w%3d%3d](url)

I wired it with i2c to esp8266 and tried to test it with the hello word example.
for the constructor I got from this GitHub the
 `U8G2_SSD1327_MIDAS_128X128_F_SW_I2C(rotation, clock, data [, reset])` and used it like this;
`U8G2_SSD1327_MIDAS_128X128_F_SW_I2C u8g2(U8G2_R0, /* clock=*/ 5, /* data=*/ 4, /* reset=*/ U8X8_PIN_NONE);` but the screen shows only random dots on the top left corner. 

Then I tried to use the `U8G2_SSD1327_SEEED_96X96_F_SW_I2C u8g2(U8G2_R0, /* clock=*/ 5, /* data=*/ 4, /* reset=*/ U8X8_PIN_NONE);` and with it I got text on screen. How do I get the screen working with full 128x128 area? I even tried to connect the reset pin but it didn't do anything. 

I hope you could help, because I don't know what I'm doing wrong. Thanks! (sorry for my bad English)

olikraus, 2018-06-22:
https://www.lcd-module.de/fileadmin/eng/pdf/grafik/W128128-XR.pdf

Can you attach a picture of both the SEED_96X96 result and the MIDAS_128X128 result?
Which code do produce these pictures?

> I hope you could help, because I don't know what I'm doing wrong.

Probably nothing: Although the chip (SSD1327) is the same, the displays still might have huge differences. We need to create a new constructor for  the EA_W128128.

olikraus, 2018-06-22:
I have added the following constructors
```
U8G2_SSD1327_EA_W128128_F_SW_I2C u8g2(U8G2_R0, /* clock=*/ 5, /* data=*/ 4, /* reset=*/ U8X8_PIN_NONE);
U8G2_SSD1327_EA_W128128_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);
```

olikraus, 2018-06-22:
I have created a new beta release 2.23.9

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

olikraus, 2018-06-22:
One more note: I strongly suggest to use Hardware I2C constructor. SW emulated I2C will be very slow.

olikraus, 2018-06-22:
One more request: Can you test the "Flipmode" example? Will the display correct for the normal and the flipped version? Thanks for testing...

Nomatiti, 2018-06-22:
I tried it but the text is cut from the bottom like this; 
[image]

So I tried to draw a 128x128 box which should fill the entire screen and this came up on the screen;
[image]

I also tested the same box with the SEED 96x96 and it also showed similar cutout at the bottom;
[image]

olikraus, 2018-06-22:
Which constructor did you use?  What is your Arduino board?

Nomatiti, 2018-06-22:
I used `U8G2_SSD1327_EA_W128128_F_SW_I2C u8g2(U8G2_R0, /* clock=*/ 5, /* data=*/ 4, /* reset=*/ U8X8_PIN_NONE);` and the board is nodemcu esp8266

Nomatiti, 2018-06-22:
I don’t know which pins are hw i2c and I didn’t google the pin out so I didn’t try hw i2c

Nomatiti, 2018-06-22:
Oh they’re the same pins :)

olikraus, 2018-06-22:
hmmm... maybe the sw emulated i2c is to fast on the esp8266.

The ESP8266 should allows pin remapping for I2C. So this may work and will include proper timing for the I2C:
`U8G2_SSD1327_EA_W128128_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE, /* clock=*/ 5, /* data=*/ 4 );`

if the pins are the same then
`U8G2_SSD1327_EA_W128128_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE );` will also work.

Will the result be better?

Nomatiti, 2018-06-22:
It still doesn’t draw the whole screen when I used `U8G2_SSD1327_EA_W128128_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8G2_PIN_NONE);` but it is better
[image]

olikraus, 2018-06-22:
hmmm... could be a different problem. I think this has to be fixed in the code.

olikraus, 2018-06-22:
... give me a minute...

olikraus, 2018-06-22:
I have created beta 2.23.10.... will this be better?

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

Nomatiti, 2018-06-22:
Now it does this;
[image]

olikraus, 2018-06-22:
strange... ok, then I need your help. please locate file u8x8_d_ssd1327.c on your harddisk. It is part of the U8g2_Arduino library, maybe in some Arduino subfolder.

Goto line 386:
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_ssd1327.c#L386

Change value 0x000 of that line `U8X8_CA(0x0a2, 0x000),` to something else.
For example to `U8X8_CA(0x0a2, 0x010),`
Will this look better? Will your box cover the complete screen?
Try other values if it does not fit. The value actually is something like the y offset. But the correct y value is not really described...

Nomatiti, 2018-06-24:
I found the file but for some reason nothing happens when I modify it. I even tried to reinstall the library after modifying the file, but nothing happens. I must be doing something wrong, I’m sorry.

olikraus, 2018-06-24:
no problem. I have created beta 2.23.12

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

cyberkeiler, 2018-08-03:
Thanks @olikraus.
This beta with `U8G2_SSD1327_EA_W128128_F_HW_I2C` definition works fine for my 1.5" OLED Module sold as "MSP1503".

SSD1327_EA_W128128 wasn't defined in the standard Arduino Library Managers version, but your linked version solved all problems ;)

john--, 2019-03-10:
@nomatiti do you have any info on what harness / wiring you used to get this display working? You've got an interesting board in the back there.

Nomatiti, 2019-03-11:
The board is just a simple breakout board which has the needed 12v power boost converter. The pin header has all the data lines, rd, w/r, d/c, reset and cs lines. The three configuration lines are connected to the dip switch. I just designed it to test the display fast and allow access to every possible line to test both i2c and spi. I don't remember anymore if I did get the display to work fully, because I played with it so long time ago and I haven't had any time after that to test it because i'm still in school in Finland and have my matriculation examination coming.

john--, 2019-05-11:
@Nomatiti  Thanks! Are things less busy for you now? Would be interested in your design to test out some screens I have around.

Nomatiti, 2019-05-11:
@john-- I have done all my exams (results coming next week 😬) and now on "holiday" so I want also start to experiment on the screen myself. But if you want I would happily send the files to you if you are interested and feel like they would help!

john--, 2019-05-11:
@nomatiti that would be great if you could! I enjoy assembling circuit boards :)

Romashka360, 2021-01-16:
> [image]
> This is the screen (the black line is just the screen updating) with SEED with this code;
> 
> ```
> #include <Arduino.h>
> #include <U8g2lib.h>
> 
> #ifdef U8X8_HAVE_HW_SPI
> #include <SPI.h>
> #endif
> #ifdef U8X8_HAVE_HW_I2C
> #include <Wire.h>
> #endif
> 
> // Please UNCOMMENT one of the contructor lines below
> // U8g2 Contructor List (Frame Buffer)
> // The complete list is available here: https://github.com/olikraus/u8g2/wiki/u8g2setupcpp
> // Please update the pin numbers according to your setup. Use U8X8_PIN_NONE if the reset pin is not connected
> 
> U8G2_SSD1327_SEEED_96X96_F_SW_I2C u8g2(U8G2_R0, /* clock=*/ 5, /* data=*/ 4, /* reset=*/ U8X8_PIN_NONE);	// Seeedstudio Grove OLED 96x96
> 
> //U8G2_SSD1327_MIDAS_128X128_F_SW_I2C u8g2(U8G2_R0, /* clock=*/ 5, /* data=*/ 4, /* reset=*/ U8X8_PIN_NONE);
> 
> // End of constructor list
> 
> 
> void setup(void) {
>   u8g2.begin();
> }
> 
> void loop(void) {
>   u8g2.clearBuffer();					// clear the internal memory
>   u8g2.setFont(u8g2_font_ncenB08_tr);	// choose a suitable font
>   u8g2.drawStr(10,20,"Hello World!");	// write something to the internal memory
>   u8g2.sendBuffer();					// transfer internal memory to the display
>   delay(1000);  
> }
> ```
> 
> [image]
> This is with Midas. the code is the same except the SEED is commented out and the Midas is used.

HE-HE. HOW DID YOU GET THE GRAY-SCALE IMAGE ON THE BLACK-AND-WHITE SCREEN???

michaelhay, 2022-01-31:
> The board is just a simple breakout board which has the needed 12v power boost converter. The pin header has all the data lines, rd, w/r, d/c, reset and cs lines. The three configuration lines are [...]

Could you provide the schematics/pcb design for your breakout board?