olikraus/u8g2 #642: Support for LC7981_240X64

vk3alb, 2018-06-23:
I have PG 24064 display that uses a LC7981. It works OK with U8G2_LC7981_240X128_1_6800 but I understand it could be faster. Is it possible to add support for 240x64 pixels please?

olikraus, 2018-06-23:
New constructor: 
`U8G2_LC7981_240X64_1_6800`
`U8G2_LC7981_240X64_2_6800`
`U8G2_LC7981_240X64_F_6800`

olikraus, 2018-06-23:
I have created beta 2.23.11

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

vk3alb, 2018-06-23:
Thanks,

The first two libraries work and they are faster. I cannot try U8G2_LC7981_240X64_F_6800 because it creates too many variables for my UNO. I suspect it would work ok in a MEGA. Thanks for the quick turn around.

Lou

olikraus, 2018-06-23:
Thanks for the feedback. There's no need to test F variant.

vk3alb, 2018-06-23:
OK great. Case closed.

closed 2018-06-23 by vk3alb