olikraus/u8g2 #643: Feature Request/Advice please

stuveri, 2018-06-26:
Hi Oli,

Thanks for this library - it has saved me a lot of time. 
I am developing on a 128x128 display and need clear visibility from a distance so I'm
using the u8g2_font_logisoso92_tn font set which is great however I want to display
up to 15 using hex characters A-F as well as 0-9. What would be the easiest way to
extend this group or make a new set of alpha charaters with A-F? I don't have a lot
of memory so would not want the full alphabet.
thanks,
Stu

stuveri, 2018-06-26:
Yes - that would be all I need to display values up to 15 using one large character.

olikraus, 2018-06-26:
I have used the following commands to generate the font:
```
../otf2bdf/otf2bdf -a -r 72 -p 131 -o tmp.bdf ../ttf/Logisoso.ttf
./bdfconv -v -f 1 -m '32,42-57,65-70' tmp.bdf -o u8g2_font_logisoso92_tx.c  -n u8g2_font_logisoso92_tx -d ../bdf/helvB18.bdf 
```

Result (paste this into your .ino file):
```
const uint8_t u8g2_font_logisoso92_tx[2253] U8G2_FONT_SECTION("u8g2_font_logisoso92_tx") = 
  "\27\0\6\6\6\7\6\10\7\65o\0\372\134\351[\0\6\6\0\0\10\260 \7\0\0\4\334\3*["
  "gJ\5\336\77\11/\371\244A\222H\6\6\1I \204\342`$\214\3\221lX\14\11\201\21\70\202"
  "D\360\3:\302\310\370$\234&\3U\250\10U\201\62\14G\371\30#\341\7t\202D\340\10\26CB"
  "`\14\11\42\211\70\30\11\343P\20\220\4B\60\6I\42\31H\11/\371\36\0+\27\343\210\225\334/"
  "\214\213\371\37\363\3\376\7\250\26\306\305\374\307\0\0,\36\214d\325\23\27\202\343@$\6E\260\24\134"
  "\202\205\240\30\22\304\1!\206\14\222\241\1\0-\13#\203\355\334\3\377\7\250\0.\31\14c\5\24\27"
  "\202\343@$\6E\260\24\134\202\205\240\30\20E\201\0\0/\204+\27\5\334\177L^\223\307\360\232<"
  "\206\327\360\30\37\303kx\214\217\341\65<\206\327\344\61\274&\217\341\65y\14\257\311cx\15\217\361\61"
  "\274\206\307\370\30^\303cxM\36\303k\362\30^\223\307\360\232<\206\327\360\30\37\303kx\214\217\341"
  "\65<\306\307\360\32\36\303k\362\30^\223\307\360\232<\206\327\360\30\37\303kx\214\217\341\65<\306\307"
  "\360\32\36\343cx\15\217\341\65y\14\257\311cx\15\217\361\61}\0\60b.\367\4\344K\212\20\302"
  "Q\60\14\305\211<\17\250\21-\244\304\204\230\16\224\1A \305\203y\20\16\307\61\70\242\206\320\220\32"
  "\2\203b\4\14\13\373\377\377\377\377\377\377Y\214\200A\61\204\206\324\20\32Rch@\15\342\301<\12"
  "\10\2\61P\207\11\61%\244E\324\200\36O\304\241\60\30\205\203\204\12\11\0\61\60\27\227\6\334/\13"
  "\206\261\60)\315\22'\344q@\214H!!&\302\17\370\3\62\302\204\210\30\15\205\261\60\377\377\377\377"
  "\377\377\377\177\313\42\0\0\62\226.\367\4\344O\211\20\302Q\60L\245\221<\17\250\21-\244\304\204\230"
  "\16\224\1A \305\203y\20\16\307\61\64\42\207\320\220\32\2\203b\4\214\212\21\60,\354\317b\4\14"
  "\212!\60\246\206\240\250\32B\242j\20\206\253A\24\256\206!\300\34`\3t|\16\260\1:>\7\330"
  "\0\35`\3t|\16\260\1:>\7\330\0\35\237\3l\200\216\317\1:>\7\330\0\35\237\3l\200"
  "\216\317\1:>'\320\361\71\201\216\317\1\66@\307\347\0\35\237\23\350\370\234\300\37\360\377\0\30\63z"
  ".\367\374\333\3\377\37\0\3\35\237\23\350\370\234@\307\347\0\33\240\343s\200\15\320\361\71\300\6\350\370"
  "\34`\3t|\16\260\1:>\7\350\370\36_\244\233l\24\255\222Y\260\313\205\261\62\225\246\213|\20"
  "\320\21:\302\206\330\20\231F\206\310\374\377\27\62\222\201\305\70\12U\243@L\216a\21\71\204\207\363\20"
  "\42\12\204\230*P\306\11))\244E\324\200\236\316\244\261,\232\344cH\0\64o.\367\4\344C\14"
  "\221\21\62Df!Cd\26\62Df!Cd\12\31\42\263\220!\62\13\31\42\263\220!\62\13\31\42"
  "\263\220!\62\211\13\221Y\310\20\231\216\5\304\70\30\16\3a\70\14\204\341\60\20\206\303@\30\15#a"
  "\64\214\204\321\60\22F\303H\30\14Ca\60\14\205\301\60\24\6\303P\30\13s\313\17\370\177\0\354b"
  "\210\314\377\237\270`\0\65n.\367\4\344\3\377\37\300\23\231\377\377\216\204\304\30\42\216\210!)\245\304"
  "\204\234\16\224\221*T\204\212X\15\23\4\62D\230\207\360\200\32\202Cj\4\215\251\21\64(F\300\260"
  "\260\263\30\42\363\377\23\61\26v\26#`P\14\201\61\65\204\206\324\20\34P\203x\60\17\42\202D\12"
  "\324aBL\11i\21\65\240\307\23q(\14V\331 \241B\2\0\66\223.\327\4\334K\11\21\262U"
  "\60\14\305\211<\17\250\21-\244\304\204\230\16\224\1A\42\304\203y\20\15\250\61\64\244\206\320\230\30\2"
  "\203b\4\14\13\273\353\42\62\377C\16\23\203x<\214\200\322\214\226RbBN\7\312H\25\251BE"
  "\254\6\212\21\31\42\314C\200\70\16\341\1\71\2\207\324\10\32\24#hP\214\200aa\377\377W\65\2"
  "\6\305\20\32RCp@\216\301\341\70\210\7\363( \10\304@\35&\304\224\220\26Q\3z<\21\207"
  "\302`\24\16\22*$\0\67\202.\367\4\344\3\377\37\300\243\64T#`P\215\200A\65\2\306\344\10"
  "\30SC`L\15\201\61\65\4\206\344\20\30Rc`H\215\260\1:\300F\330\4\66\302&\320\1\66"
  "\302&\320\1\66\302&\320\1\66\302&\320\1\66\302&\260\21\66\201\16\260\21\66\201\16\260\21\66\201\16"
  "\260\21\66\201\16\260\21\66\201\215\260\11t\200\215\260\11t\200\215\260\11t\200\215\260\11t\200\215\260\11"
  "l\204M\240\3l\204M\25\0\70\240.\367\4\334K\212\20\302Q\60\14\305\211<\17\250\21-\244\304"
  "\204\230\16\224\1A \305\203y\20\16\307\61\64\244\206\320\220\32\2\203b\4\14\13\373\377U\215\200A"
  "\61\204\206\324\20\34\220cp\70\16\342\301<\12\10\2\61P\207\11\61%\244E\324\200\34P#ZH"
  "\12)\61\35(#\252x\60\17\302\341\70\6\7\344\20\32RC`P\214\240A\61\2\206\205\375\177U"
  "#`P\14\241!\65\204F\344\30\32\217\203x\60\17\42\202@\14\324aBL\11i\21\65\240\307\23"
  "q(\14F\341 \241B\2\0\71\223.\367\4\344K\212\20\302U.\14\305\211<\17\250\21-\244\304"
  "\204\230\16\224\1A \305\203y\20\215\307\61\64\244\206\320\220\32\2\203b\4\14\13\373\377\77+C\65"
  "\2\6\325\10\32\222#p@\17\301\341@\204\7\23\31 \7eX\21jE\312@\35'\304\224\224\226"
  "\21S\11\30\317\203`L\206\204!\62\377\211\227\13ca\263\30\1\203b\10\214\251!\64\244\206\340\200"
  "\32\304\203y\20\21$R\240\16\23bJH\213\250\1=\236\210Ca\260\312\6\11\25\22\0A\266-"
  "\367\4\334G\13q\21\66\300\346\276\207\367\234\7\341\242\267M\264\351\262\12V\35f\261\256W\61\2\6"
  "\305\10\30\324bXP\213aA-\206\305\304\30\30\22c`H\214\201!-\212\205\264(\26\21\243`"
  "@\214\202\1\61\12\6\264\70\26\320\342X@\213c\361\60\16\206\303\70\30\316\42Y\70\213d\341,\222"
  "E\303H\30\14#a\60\214\204\301,\226\5\263X\26\13\223\302X\30\12ca(\343\222\22R\336q"
[... cut ...]
WARNING: I have create the font in the new font format, which will be introduced in the next release. You may need to install new u8g2 beta...

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

stuveri, 2018-06-27:
Perfect. I deleted the old u8g2 and added the new .ZIP library to my IDE then included the 
new font array into my code and it worked fine. 
Thanks so much for your quick response and resolution!

olikraus, 2018-06-27:
:-)

closed 2018-08-04 by olikraus