olikraus/u8g2 #644: u8g2 and screen oled i2c 128x64 (screen empty) and 128x128 soon

Freepl, 2018-06-27:
Hello 

I have this screen
[https://fr.aliexpress.com/item/Free-shipping-Yellow-blue-double-color-128X64-OLED-LCD-LED-Display-Module-For-Arduino-0-96/32665937977.html?spm=a2g0s.9042311.0.0.X7TUdD
](url)

I connect my screen like this
VCC / 5v
GND / GND
SCL / A5
SDA / A4

I'll receveid this screen too soon
[https://fr.aliexpress.com/item/Free-shipping-Yellow-blue-double-color-128X64-OLED-LCD-LED-Display-Module-For-Arduino-0-96/32665937977.html?spm=a2g0s.9042311.0.0.X7TUdD](url)

I have this code. It compile but there is nothing on  screen.
can you see this and say modify this  for the 2 screens ?
I see this for  10 days !
Thank you  

Hello 

I have this screen
https://fr.aliexpress.com/item/Free-shipping-Yellow-blue-double-color-128X64-OLED-LCD-LED-Display-Module-For-Arduino-0-96/32665937977.html?spm=a2g0s.9042311.0.0.X7TUdD
I connect my screen like this
VCC / 5v
GND / GND
SCL / A5
SDA / A4

I'll receveid this screen too
https://fr.aliexpress.com/item/Free-shipping-Yellow-blue-double-color-128X64-OLED-LCD-LED-Display-Module-For-Arduino-0-96/32665937977.html?spm=a2g0s.9042311.0.0.X7TUdD

I have this code. It compile but there is nothing on  screen.
can you see this and say modify this  for the 2 screens ?

It has been one week since I look
Thank you  

My code don't work ( screen empty)
`// Pour l'écran
#include <Arduino.h>
#include <U8g2lib.h>
//Pour les sondes
#include <DallasTemperature.h>     //DS18B20 pour température de l'eau et de l'air
#include <Wire.h>

  //les librairies de l'écran
// pour les fontes  https://github.com/olikraus/u8g2/wiki/fntlistall
// poue l'écran     https://github.com/olikraus/u8g2/wiki/u8g2setupc

// Déclaratipon de l'écran 128x64
//U8G2_SSD1306_128X64_NONAME_2_SW_I2C u8g2(rotation, clock, data [, reset])
                                         // U8G2_R0,SCL  A5, SDA  A4
U8G2_SSD1306_128X64_NONAME_2_SW_I2C u8g2 (U8G2_R0, A5, A4)

//Déclaration de l'écran 128x128
/* 
U8G2_SSD1327_EA_W128128_1_SW_I2C u8g2(U8G2_R0, A5, A4); 
ou le 
U8G2_SSD1327_MIDAS_128X128_1_SW_I2C u8g2(U8G2_R0, A5, A4);
*/

// les 2 sondes DS18B20
#define ONE_WIRE_BUS_1 2               // sonde pour l'eau pin 2 
#define ONE_WIRE_BUS_2 4               // sonde pour l'air pin 4
OneWire oneWire_in(ONE_WIRE_BUS_1);    // sonde pour l'eau
OneWire oneWire_out(ONE_WIRE_BUS_2);   // sonde pour l'air
DallasTemperature sensor_eau(&oneWire_in);     // sonde pour l'eau
DallasTemperature sensor_air(&oneWire_out);    // sonde pour l'air

void setup(void)
{
  /* U8g2 Project: SSD1306 Test Board */
  pinMode(A4, OUTPUT);
  digitalWrite(A4, 0);
  u8g2.begin();
  u8g2.setPowerSave(0);

// start serial port
  Serial.begin(9600);
  Serial.println("Dallas Temperature Control Library Demo - TwoPin_DS18B20");

  // Start up the library
  sensor_eau.begin();    // sonde pour l'eau
  sensor_air.begin();    // sonde pour l'air

  // déclaration de la pin 2 et 4
  pinMode(2, INPUT);
  pinMode(4, INPUT);

  int readValEau = analogRead(2);// met à zéro la sonde EAU
  int readValAir = analogRead(4);// met à zéro la sonde AIR

}

void loop(void)
{
// AFFICHAGE SUR SERIAL
  Serial.print("Recherche des temperatures...");
  sensor_eau.requestTemperatures();
  sensor_air.requestTemperatures();
  Serial.println(" done");

  // sonde EAU
  Serial.print("Température eau: ");
  Serial.println(sensor_eau.getTempCByIndex(0));

  // sonde AIR
  Serial.print("Température air: ");
  Serial.println(sensor_air.getTempCByIndex(0));
  delay ( 5000);

  /*/modele d'écriture
    u8g2.setFont(u8g2_font_ncenB14_tr);
    u8g2.setCursor(0, 15);
    u8g2.print("Hello World!");
  */

  // Affichage Niveau batterie
  u8g2.setFont(u8g2_font_ncenB14_tr);
  u8g2.setCursor(0, 0);
  u8g2.println("NIVEAU BATTERIE");
  //delay ( 1000);

  // pour le voltage de la batterie
  // Conversion du signal (which goes from 0 - 1023) to a voltage (0 - (V):
  int sensorValue = analogRead(A0);//*******************************************************  A3 pour la voiture bateau
  // écrit la valeur lue
  float voltage = sensorValue * (5.0 / 1023.0);
  u8g2.setCursor(0, 1);// position 0, sur la 1 ere ligne,
  u8g2.print("        ");
  u8g2.print(voltage);
  u8g2.print("V");

  //Affichage  températures Air
  u8g2.setFont(u8g2_font_chroma48medium8_r);
  u8g2.setCursor(0, 3);
  u8g2.println("TEMP. AIR ");
  u8g2.print("       ");
  u8g2.print(sensor_air.getTempCByIndex(0));
  u8g2.print(" ");
  u8g2.print("C");

  // Affichage  températures Eau
  u8g2.setFont(u8g2_font_ncenB14_tr);
  u8g2.setCursor(0, 6);
  u8g2.println("TEMP. EAU ");
  u8g2.print("       ");
  u8g2.print(sensor_eau.getTempCByIndex(0));
  u8g2.print(" ");
  u8g2.print("C");

}`

olikraus, 2018-06-27:
All the u8g2 commands have to be encapsulated in a while loop:

```
void loop(void) {
  ...
  u8g2.firstPage();
  do {
    <your code>
  } while ( u8g2.nextPage() );
 ...
}
```

Freepl, 2018-06-27:
Hello
I am tall beginning with arduino
1 What declaration of screen I have to take 128x64 and 128x128 ?
2 I change my code but i have error

Code error
`Arduino : 1.8.4 (Windows 7), Carte : "Arduino/Genuino Uno"
Ecran_u8G2_128x64_128x128:27: error: expected ',' or ';' before 'OneWire'
 OneWire oneWire_in(ONE_WIRE_BUS_1);    // sonde pour l'eau
 ^
Ecran_u8G2_128x64_128x128:29: error: 'oneWire_in' was not declared in this scope
DallasTemperature sensor_eau(&oneWire_in);     // sonde pour l'eau
                               ^
exit status 1
expected ',' or ';' before 'OneWire'

`

Here my code modify
`// Pour l'écran
#include <Arduino.h>
#include <U8g2lib.h>
//Pour les sondes
#include <DallasTemperature.h>     //DS18B20 pour température de l'eau et de l'air
#include <Wire.h>

  //les librairies de l'écran
// pour les fontes  https://github.com/olikraus/u8g2/wiki/fntlistall
// poue l'écran     https://github.com/olikraus/u8g2/wiki/u8g2setupc

// Déclaratipon de l'écran 128x64
U8G2_SSD1306_128X64_NONAME_2_SW_I2C u8g2 (U8G2_R0, A5, A4)

//Déclaration de l'écran 128x128
/* 
U8G2_SSD1327_EA_W128128_1_SW_I2C u8g2(U8G2_R0, A5, A4); 
ou le
U8G2_SSD1327_MIDAS_128X128_1_SW_I2C u8g2(U8G2_R0, A5, A4);
*/

// les 2 sondes DS18B20
#define ONE_WIRE_BUS_1 2               // sonde pour l'eau pin 2 
#define ONE_WIRE_BUS_2 4               // sonde pour l'air pin 4
OneWire oneWire_in(ONE_WIRE_BUS_1);    // sonde pour l'eau
OneWire oneWire_out(ONE_WIRE_BUS_2);   // sonde pour l'air
DallasTemperature sensor_eau(&oneWire_in);     // sonde pour l'eau
DallasTemperature sensor_air(&oneWire_out);    // sonde pour l'air

void setup(void)
{
  /* U8g2 Project: SSD1306 Test Board */
  pinMode(A4, OUTPUT);
  digitalWrite(A4, 0);
  u8g2.begin();
  u8g2.setPowerSave(0);

// start serial port
  Serial.begin(9600);
  Serial.println("Dallas Temperature Control Library Demo - TwoPin_DS18B20");

  // Start up the library
  sensor_eau.begin();    // sonde pour l'eau
  sensor_air.begin();    // sonde pour l'air

  // déclaration de la pin 2 et 4
  pinMode(2, INPUT);
  pinMode(4, INPUT);

  int readValEau = analogRead(2);// met à zéro la sonde EAU
  int readValAir = analogRead(4);// met à zéro la sonde AIR

}

void loop(void)
{
// AFFICHAGE SUR SERIAL
  Serial.print("Recherche des temperatures...");
  sensor_eau.requestTemperatures();
  sensor_air.requestTemperatures();
  Serial.println(" done");

  // sonde EAU
  Serial.print("Température eau: ");
  Serial.println(sensor_eau.getTempCByIndex(0));

  // sonde AIR
  Serial.print("Température air: ");
  Serial.println(sensor_air.getTempCByIndex(0));
  delay ( 5000);

  /*/modele d'écriture
    u8g2.setFont(u8g2_font_ncenB14_tr);
    u8g2.setCursor(0, 15);
    u8g2.print("Hello World!");
  */
u8g2.firstPage();
  do {
  // Affichage Niveau batterie
  u8g2.setFont(u8g2_font_ncenB14_tr);
  u8g2.setCursor(0, 0);
  u8g2.println("NIVEAU BATTERIE");
  //delay ( 1000);

  // pour le voltage de la batterie
  // Conversion du signal (which goes from 0 - 1023) to a voltage (0 - (V):
  int sensorValue = analogRead(A0);//*******************************************************  A3 pour la voiture bateau
  // écrit la valeur lue
  float voltage = sensorValue * (5.0 / 1023.0);
  u8g2.setCursor(0, 1);// position 0, sur la 1 ere ligne,
  u8g2.print("        ");
  u8g2.print(voltage);
  u8g2.print("V");

  //Affichage  températures Air
  u8g2.setFont(u8g2_font_ncenB14_tr);
  u8g2.setCursor(0, 3);
  u8g2.println("TEMP. AIR ");
  u8g2.print("       ");
  u8g2.print(sensor_air.getTempCByIndex(0));
  u8g2.print(" ");
  u8g2.print("C");

  // Affichage  températures Eau
  u8g2.setFont(u8g2_font_ncenB14_tr);
  u8g2.setCursor(0, 6);
  u8g2.println("TEMP. EAU ");
  u8g2.print("       ");
  u8g2.print(sensor_eau.getTempCByIndex(0));
  u8g2.print(" ");
  u8g2.print("C");
} while ( u8g2.nextPage() );

}`

closed 2018-06-27 by Freepl

olikraus, 2018-06-27:
Actually it looks better. 
Optimization: Move statements like `int sensorValue = analogRead(A0);` before the loop.

Is it working now? It not, then please test your hardware setup with one of the u8g2 examples, which are shipped with the u8g2 library.

Freepl, 2018-06-27:
I try this this night
SCL / A5
SDA / A4

The code is like this ?
`void setup(void)
{
.....
}

int sensorValue = analogRead(A0);           //  A4 or A5 for me ?
void loop(void)
{
  u8g2.firstPage();
  do {`
....
}

olikraus, 2018-06-27:
```
void loop(void)
{
int sensorValue;

sensorValue = analogRead(A0); 
u8g2.firstPage();
do {`
....
}
```

Freepl, 2018-06-27:
A GREAT thank you .

The code works.
[image]

Can you confirm the screen for oled 128 x 128 please .
I'l receveid it soon.
Are you ok with this choice ?

< //Déclaration of screen 128x128
/*
  U8G2_SSD1327_EA_W128128_1_SW_I2C u8g2(U8G2_R0, A5, A4);
  ou le
  U8G2_SSD1327_MIDAS_128X128_1_SW_I2C u8g2(U8G2_R0, A5, A4);
*/>

The good code
<// Pour l'écran
#include <Arduino.h>
#include <U8g2lib.h>
//Pour les sondes
#include <DallasTemperature.h>     //DS18B20 pour température de l'eau et de l'air
#include <Wire.h>

//les librairies de l'écran
// pour les fontes  https://github.com/olikraus/u8g2/wiki/fntlistall
// poue l'écran     https://github.com/olikraus/u8g2/wiki/u8g2setupc

// Déclaratipon de l'écran 128x64
U8G2_SSD1306_128X64_NONAME_2_SW_I2C u8g2 (U8G2_R0, A5, A4);

//Déclaration de l'écran 128x128
/*
  U8G2_SSD1327_EA_W128128_1_SW_I2C u8g2(U8G2_R0, A5, A4);
  ou le
  U8G2_SSD1327_MIDAS_128X128_1_SW_I2C u8g2(U8G2_R0, A5, A4);
*/

// les 2 sondes DS18B20
#define ONE_WIRE_BUS_1 2               // sonde pour l'eau pin 2 
#define ONE_WIRE_BUS_2 4               // sonde pour l'air pin 4
OneWire oneWire_in(ONE_WIRE_BUS_1);    // sonde pour l'eau
OneWire oneWire_out(ONE_WIRE_BUS_2);   // sonde pour l'air
DallasTemperature sensor_eau(&oneWire_in);     // sonde pour l'eau
DallasTemperature sensor_air(&oneWire_out);    // sonde pour l'air

void setup(void)
{
  /* U8g2 Project: SSD1306 Test Board */
  pinMode(A4, OUTPUT);
  digitalWrite(A4, 0);
  u8g2.begin();
  u8g2.setPowerSave(0);

  // start serial port
  Serial.begin(9600);
  Serial.println("Dallas Temperature Control Library Demo - TwoPin_DS18B20");

  // Start up the library
  sensor_eau.begin();    // sonde pour l'eau
  sensor_air.begin();    // sonde pour l'air

  // déclaration de la pin 2 et 4
  pinMode(2, INPUT);
  pinMode(4, INPUT);

  int readValEau = analogRead(2);// met à zéro la sonde EAU
  int readValAir = analogRead(4);// met à zéro la sonde AIR
}

void loop(void)
{

  int sensorValue = analogRead(A0); //A0

  u8g2.firstPage();
  do {
    // AFFICHAGE SUR SERIAL
    Serial.print("Recherche des temperatures...");
    sensor_eau.requestTemperatures();
    sensor_air.requestTemperatures();
    Serial.println(" done");

    // sonde EAU
    Serial.print("Température eau: ");
    Serial.println(sensor_eau.getTempCByIndex(0));

    // sonde AIR
    Serial.print("Température air: ");
    Serial.println(sensor_air.getTempCByIndex(0));
    delay ( 500);

    /*/modele d'écriture
      u8g2.setFont(u8g2_font_ncenB14_tr);
      u8g2.setCursor(0, 15);
      u8g2.print("Hello World!");
    */
    /* fonte 8 pixels u8g2_font_t0_11_tf
       fonte 10 pixels  u8g2_font_7x14_mr X11
       fonte 12 pixels u8g2_font_9x15_mn X11
       fonte 14 pixels u8g2_font_t0_18_mn UW ttyp0
        fonte 16 pixels u8g2_font_crox3c_tn crox
      fonte 20 pixels u8g2_font_fur20_tf Free Universal

    */

    // Affichage Niveau batterie
    u8g2.setFont(u8g2_font_t0_11_tf);///(u8g2_font_ncenB14_tr)
    u8g2.setCursor(0, 8);
    u8g2.println("NIVEAU BATTERIE");
    //delay ( 1000);

    // pour le voltage de la batterie
    // Conversion du signal (which goes from 0 - 1023) to a voltage (0 - (V):
    int sensorValue = analogRead(A0);//*******************************************************  A3 pour la voiture bateau
    // écrit la valeur lue
    float voltage = sensorValue * (5.0 / 1023.0);
    u8g2.setCursor(0, 24);// position 0, sur la 1 ere ligne,
    u8g2.print(voltage);
    u8g2.print("V");

    //Affichage  températures Air
    u8g2.setFont(u8g2_font_t0_11_tf);
    u8g2.setCursor(0, 44);
    u8g2.println("TEMP. AIR ");
    u8g2.print("  ");
    u8g2.print(sensor_air.getTempCByIndex(0));
    u8g2.print(" ");
    u8g2.print("C");

    // Affichage  températures Eau
    u8g2.setFont(u8g2_font_t0_11_tf);
    u8g2.setCursor(0, 64);
    u8g2.println("TEMP. EAU ");
    u8g2.print("  ");
    u8g2.print(sensor_eau.getTempCByIndex(0));
    u8g2.print(" ");
    u8g2.print("C");
  } while ( u8g2.nextPage() );

}>

Freepl, 2018-06-28:
Bonjour olikraus,

thank you for all

Patrick Lacroix, patrick.lacroix18@free.fr
28/06/2018 
----- Message reçu ----- 
De : olikraus 
À : olikraus/u8g2 
Date : 2018-06-28, 00:26:15
Sujet : Re: [olikraus/u8g2] u8g2 and screen oled i2c 128x64 (screen empty)and 128x128 soon (#644)

The W128128 is still under construction. The Midas variant should be ok.


Freepl, 2018-06-28:
Hello
Which code is  good ?
(This code works)
<void loop(void)
{

int sensorValue = analogRead(A0); //A0

u8g2.firstPage();
do {>

or this code
<void loop(void)
{
int sensorValue;

sensorValue = analogRead(A0); 
u8g2.firstPage();
do { >
Which is better ?

olikraus, 2018-06-28:
Question 1: Both forms should be the same. There is no difference.

Question 2:
Not clear what you mean? Is it about A0 and A4? I can not tell this, you should know this better.

In general: You should NOT use any analogRead(...) inside the u8g2.NextPage() while loop.
So both of your analogReads should happen outside.

Freepl, 2018-06-28:
> Question 2:
Not clear what you mean?

my origin code is for print on screen the level of battery.
This code use 
`int sensorValue = analogRead(A0);`

and says me use this too befor my code.
I would know if this make one problem and  I asked you if I should not mark A4 in your analogRead instead of A0?

olikraus, 2018-06-28:
As I said, there should be no analogRead(..) inside the u8g2-do-while loop at all.

Freepl, 2018-06-28:
And then where I do have to put this code?

`// pour le voltage de la batterie
// Conversion du signal (which goes from 0 - 1023) to a voltage (0 - (V):
int sensorValue = analogRead(A0);//*******************************************************  A3 pour la voiture bateau
// écrit la valeur lue
float voltage = sensorValue * (5.0 / 1023.0);
u8g2.setCursor(0, 24);// position 0, sur la 1 ere ligne,
u8g2.print(voltage);
u8g2.print("V");
`

olikraus, 2018-06-28:
Outside the u8g2-do-while loop