olikraus/u8g2 #645: [Q] how to connect a JLX256128 LCD?

klero2, 2018-07-02:
Sorry to open an Issue, but i dont know other way to ask a question.

Can you tell us the pin names to connect the arduino to a JLX256128, long time ago I bought one of this, but only now i find the library to drive it and want to give it a try.

the screen has the pin ROM-SCK and ROM-CS I dont know if this are the one to connect or are for something else, JLX dont give much information in their website.

olikraus, 2018-07-02:
Ignore all pins, which are prefixed with ROM. Other pins should have proper labels.

klero2, 2018-07-02:
thank you for your reply.
I am using an Arduino UNO

I have this:
RS (Resgister selection signal) = Arduino Pin 9
CS (Chip Select) = Arduino Pin 10
RW (Read/Write, Data line) = Arduino Pin 11
RST = Arduino Pin 8 
RD (E/Clock)= Arduino Pin 13

and this is the definition in the scketch:
U8G2_ST75256_JLX256128_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);	// Uno, Enable U8g2 16 bit mode for this display

Not working, maybe am missing something or wrong connection.

olikraus, 2018-07-02:
Datasheet link: https://www.crystalfontz.com/controllers/Sitronix/ST75256/409/

Have a look at page 25, the clock and data lines are explaind there.
RD --> connect to high (3.3V)
RW --> connect to high (3.3V)
D0 --> clock
D1, D2, D3 --> data
D4, D5, D6, D7 --> connect to high (3.3V)

Additionally several preconditions are required: 
- Your display needs to be configured for 4-wire SPI
- Your display is a 3.3V device, this means, there needs to be a levelshifter between Arduino and Display. There are some other ordering options (e.g. 5V) which I have not tested.

You could also attach a picture of the back side of your display so what we can crosscheck your setup.

klero2, 2018-07-02:
This is the back of my screen I set the connection to SPI as described in the silk print.
Tomorrow I will try the connection as you said. 

Thank you again
[image]

olikraus, 2018-07-02:
Looks like there are no resistors on the data lines, so I strongly suggest a level shifter between Uno an display (e.g. 74HC4050). 

The lower right is not good enough to identify the configured interface type. 

... Is this really a picture of your LCD? I mean a picture from the shopping site will not help because then I might suggest something which is not true for your display.

klero2, 2018-07-02:
This is a picture from the shop, I don't have it with me right now, but it is exactly the same. Probably i got an unfinished product from a cheap Chinese shop

olikraus, 2018-07-02:
> The lower right has a resistor labeled '0' in R3, R5 and R7.

Well, the, once you rewired the display according to the data sheet, everything should work.

klero2, 2018-07-03:
Well, after try many times I give up, probably I got a defective screen. Not way to make it work.

Thank you for your help my friend.

olikraus, 2018-07-03:
Electronics is a difficult topic and to connect things is not always simple.
Actually I am very sure that a level shifter is required. You can not connect the display directly to an Uno board. So either you use a 3.3V Arduino board (Due, MKR) or a level shifter.