olikraus/u8g2 #647: Change C-prototype of setPageCurrTileRow() to u8g2_SetBufferCurrTileRow in the wiki

ribasco, 2018-07-05:
Perhaps this needs to be updated in the wiki

I did not find any references to **setPageCurrTileRow()** in the C-implementation (exists only in C++) as per the [wiki](https://github.com/olikraus/u8g2/wiki/u8g2reference#setpagecurrtilerow). However in the code, I see that this points to **u8g2_SetBufferCurrTileRow** instead.

olikraus, 2018-07-05:
hmm... yes indeed this is confusing.

actually setPageCurrTileRow is obsolete and it should be setBufferCurrTileRow.
In the C++ you see both functions defined:
https://github.com/olikraus/u8g2/blob/master/cppsrc/U8g2lib.h#L162
and
https://github.com/olikraus/u8g2/blob/master/cppsrc/U8g2lib.h#L164

But yes, I need to update the wiki.

Todo: Rename setPageCurrTileRow to setBufferCurrTileRow and getPageCurrTileRow to getBufferCurrTileRow

ribasco, 2018-07-05:
Thanks for the clarification

olikraus, 2018-08-04:
done

closed 2018-08-04 by olikraus