olikraus/u8g2 #649: Minor proposed changes for sharing the database of supported controllers (for wrapper libraries)

ribasco, 2018-07-07:
Hello again Oliver,

I'm in the process of developing a java wrapper library for U8G2 and I'm currently thinking of ways to automate the code generation for the supported controllers. 
Looking at your codebuild source, I see that the list of all controllers are hard coded into one array. That said, I would like to ask if you are open to any of these proposed changes below: 

1. Move the array to it's own header file which I can then use as a reference for my c++ utility.

2. Move the array to a file in plain-text form (csv/tab-delim/json/yaml/ini etc) so it could be easily be parsed in other languages 

3. Provide an option in codebuild to output the list of controllers (along with thier meta data):

    sample output (tab delimited): 
    
	| Controller | Display Name    | Supported Modes       | Display Size | Setup Procedure                          | 
	|------------|-----------------|-----------------------|--------------|------------------------------------------|
	| ssd1305    | 128x32_noname   | 4WSPI,6800,8080       | 128x32       | u8g2_Setup_ssd1305_128x32_noname_f       | 
	| ssd1305    | 128x32_noname   | I2C                   | 128x32       | u8g2_Setup_ssd1305_i2c_128x32_noname_f   | 
	| ssd1305    | 128x64_adafruit | 4WSPI,6800,8080       | 128x64       | u8g2_Setup_ssd1305_128x64_adafruit_f     | 
	| ssd1305    | 128x64_adafruit | I2C                   | 128x64       | u8g2_Setup_ssd1305_i2c_128x64_adafruit_f | 
	| ssd1306    | 128x64_noname   | 4WSPI,3WSPI,6800,8080 | 128x64       | u8g2_Setup_ssd1306_128x64_noname_f       | 
	| ssd1306    | 128x64_vcomh0   | 4WSPI,3WSPI,6800,8080 | 128x64       | u8g2_Setup_ssd1306_128x64_vcomh0_f       | 
	| ssd1306    | 128x64_alt0     | 4WSPI,3WSPI,6800,8080 | 128x64       | u8g2_Setup_ssd1306_128x64_alt0_f         |
	
Note: The first is ok with me as that would not require any development on your part but I could also help in the changes

Would like to know your thoughts, thanks.

olikraus, 2018-07-07:
You mean this array?
https://github.com/olikraus/u8g2/blob/master/tools/codebuild/codebuild.c#L70

Actually this is not a simple array, because the "Display Name" column in your example may appear multiple times:
https://github.com/olikraus/u8g2/blob/master/tools/codebuild/codebuild.c#L755

I could pull out the array to a .h file. I could also just prefix and postfix the array with a special comment. You could then search for the comment and parse the array in between.

ribasco, 2018-07-07:
Yes that one. I'm fine with the comment approach.

olikraus, 2018-07-07:
```
/* display_controller_list_start */
...
/* display_controller_list_end */
```

olikraus, 2018-08-04:
I think this can be closed.

closed 2018-08-04 by olikraus