olikraus/u8g2 #650: Oled LED Fluctuate with interface i2c using eclipse

pankajverma190, 2018-07-12:
//I am trying to use this code OLED is displaying but hourglass animation, full-half-empty is fluctuating and it is not stable.

#include <driver/gpio.h>
#include <driver/spi_master.h>
#include <driver/i2c.h>
#include <esp_log.h>
#include <freertos/FreeRTOS.h>
#include <freertos/task.h>
#include <stdio.h>
#include <string.h>
#include "u8x8.h"
#include "u8g2.h"

#include "u8g2_esp32_hal.h"
#include "icons.h"

#define PIN_SDA 21
#define PIN_SCL 22

void app_main() {

	// initialize the u8g2 hal
	u8g2_esp32_hal_t u8g2_esp32_hal = U8G2_ESP32_HAL_DEFAULT;
	u8g2_esp32_hal.sda = PIN_SDA;
	u8g2_esp32_hal.scl = PIN_SCL;
	u8g2_esp32_hal_init(u8g2_esp32_hal);

	// initialize the u8g2 library
	u8g2_t u8g2;
	u8g2_Setup_ssd1306_128x64_noname_f(
		&u8g2,
		U8G2_R0,
		u8g2_esp32_i2c_byte_cb,
		u8g2_esp32_gpio_and_delay_cb);
//	u8g2_Setup_ssd1306_i2c_128x32_univision_f(
//		&u8g2,
//		U8G2_R0,
//		u8g2_esp32_i2c_byte_cb,
//		u8g2_esp32_gpio_and_delay_cb);

	// set the display address  0x78
	u8x8_SetI2CAddress(&u8g2.u8x8, 0x78);

	// initialize the display
	u8g2_InitDisplay(&u8g2);

	// wake up the display
	u8g2_SetPowerSave(&u8g2, 0);

	// loop
	while(1) {

		// draw the hourglass animation, full-half-empty
		u8g2_ClearBuffer(&u8g2);
		u8g2_DrawXBM(&u8g2, 34, 2, 60, 60, hourglass_full);
		u8g2_SendBuffer(&u8g2);
		vTaskDelay(1000 / portTICK_RATE_MS);

		u8g2_ClearBuffer(&u8g2);
		u8g2_DrawXBM(&u8g2, 34, 2, 60, 60, hourglass_half);
		u8g2_SendBuffer(&u8g2);
	//	vTaskDelay(1000 / portTICK_RATE_MS);

		u8g2_ClearBuffer(&u8g2);
		u8g2_DrawXBM(&u8g2, 34, 2, 60, 60, hourglass_empty);
		u8g2_SendBuffer(&u8g2);
		vTaskDelay(1000 / portTICK_RATE_MS);

		// set font and write hello world
		u8g2_ClearBuffer(&u8g2);
		u8g2_SetFont(&u8g2, u8g2_font_timR14_tf);
		u8g2_DrawStr(&u8g2, 2,17,"Hello World!");
		u8g2_SendBuffer(&u8g2);
		vTaskDelay(5000 / portTICK_RATE_MS);
	}
}

olikraus, 2018-07-12:
everything looks correct... 
Some discussion is going on, that the I2C speed might be too high. I do not know u8g2_esp32_hal_t, but maybe you need to reduce I2C speed.

pankajverma190, 2018-07-12:
// below is my u8g2_esp32_hal.h file I am using esp32 with eclipse I tried to change it but still no progress(400KHz to 800KHz)normally.

/*
 * u8g2_esp32_hal.h
 *
 *  Created on: Feb 12, 2017
 *      Author: kolban
 */

#ifndef U8G2_ESP32_HAL_H_
#define U8G2_ESP32_HAL_H_
#include "u8g2.h"

#include "driver/gpio.h"
#include "driver/spi_master.h"
#include "driver/i2c.h"

#define U8G2_ESP32_HAL_UNDEFINED (-1)

#define I2C_MASTER_NUM I2C_NUM_1           //  I2C port number for master dev
#define I2C_MASTER_TX_BUF_DISABLE   0      //  I2C master do not need buffer
#define I2C_MASTER_RX_BUF_DISABLE   0      //  I2C master do not need buffer
#define I2C_MASTER_FREQ_HZ          50000  //  I2C master clock frequency
#define ACK_CHECK_EN   0x1                 //  I2C master will check ack from slave
#define ACK_CHECK_DIS  0x0                 //  I2C master will not check ack from slave

typedef struct {
	gpio_num_t clk;
	gpio_num_t mosi;
	gpio_num_t sda; // data for I²C
	gpio_num_t scl; // clock for I²C
	gpio_num_t cs;
	gpio_num_t reset;
	gpio_num_t dc;
} u8g2_esp32_hal_t ;

#define U8G2_ESP32_HAL_DEFAULT {U8G2_ESP32_HAL_UNDEFINED, U8G2_ESP32_HAL_UNDEFINED, U8G2_ESP32_HAL_UNDEFINED, U8G2_ESP32_HAL_UNDEFINED, U8G2_ESP32_HAL_UNDEFINED, U8G2_ESP32_HAL_UNDEFINED, U8G2_ESP32_HAL_UNDEFINED }

void u8g2_esp32_hal_init(u8g2_esp32_hal_t u8g2_esp32_hal_param);
uint8_t u8g2_esp32_spi_byte_cb(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr);
uint8_t u8g2_esp32_i2c_byte_cb(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr);
uint8_t u8g2_esp32_gpio_and_delay_cb(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr);
#endif /* U8G2_ESP32_HAL_H_ */

olikraus, 2018-07-12:
Similar discussion here: https://github.com/olikraus/u8g2/issues/103

See also here: https://github.com/espressif/arduino-esp32/issues/1588

closed 2018-08-04 by olikraus