olikraus/u8g2 #651: ST7567 Arduino Nano

HelloGuys1, 2018-07-12:
Hallo,
mit Bezug auf Deine Nachricht https://www.mikrocontroller.net/topic/454919?goto=5481046 melde ich mich hier nochmals. Ich möchte das Programm Hello World (ggf auch ein anderes, simples) ausführen.

Die folgende Zeile habe ich auskommentiert. Pins sind wie im Thread belegt.

U8G2_ST7567_JLX12864_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 7, /* dc=*/ 9, /* reset=*/ 8);  

D7 (cs)                  - 1 (CSB)
D8 (reset)               - 2 (RSTB)
D9 (display channel, dc) - 3 (A0)
D11 (data)               - 5 (serial Input, SI)
D13 (Clock)              - 4 (Serial Clock, SCL)
6,7 an Versorgungsspannung, 8,9 sowie 10,11 jeweils über 100nF verbunden, 12 über 100nF auf GND.

https://www.google.de/search?q=LX-12864B11&hl=de&gbv=2&tbm=isch&tbo=u&source=univ&sa=X&ved=0ahUKEwiqt5Kf9JHcAhXKAMAKHX6bCjoQsAQIKQ&biw=1461&bih=901#imgrc=RdR0Ft2bn2FyOM:

Beim Hochladen selbst ist das Display kurz komplett ohne Anzeige, danach sowie davor zeigt er hin und wieder Striche oder schwarze Blöcke an. 

Hast du noch eine Idee? 

Grüße

olikraus, 2018-07-12:
> Hast du noch eine Idee?

Nun, zunächst fällt mir auf, dass das Bild nicht zum controller ST7567 passt. Das Bild, auf das du verlinkst zeigt die Pins eines ST7565 controllers.

Die Frage ist also zunächstmal: Um was für ein Display handelt es sich eigentlich.
Ist es tatsächlich ein ST7567, dann ist die Beschaltung falsch. Ist es ein ST7565 ist der U8g2 constructor falsch.

Sinnvoll wäre es also, wenn Du mehr Informationen zu deinem Display schicken könntest: Wo hast du es gekauft? Gibt es ein Datenblatt? Gibt es Fotos vom Display? Etc...

olikraus, 2018-07-18:
Weil hier parallel grade ein anderes issue zum controller aufgemacht wurde: #654 

Dort wurde auch ein Datenblatt verlinkt: https://datasheet.octopart.com/RX12864H-BIW-Raystar-datasheet-17800594.pdf

Ist Dein Display ähnlich? Wie man dort sieht sind die pins ganz andere...

Und noch eine Frage: Wenn sich das Thema erledigt hat, dann kurze antwort damit ich den issue zumachen kann.

HelloGuys1, 2018-07-19:
Hallo, danke für deine Antwort und sorry für die späte Antwort meinerseits. 

Ein Datenblatt gibt es leider nicht zu dem Display, nur das Bild mit den chinesischen Schriftzeichen - deshalb habe ich auch das im Link herangezogen. Ist in der Tat ein ziemlich unseriöses Teil. Wenn du das Disply einmal bei Google eingibst, wirst du auch nicht viel mehr finden als die paar spärlichen Bilder auf EBay. Wenn das nicht zum Controller passt, würde das zumindest einiges erklären. Es soll angeblich einen ST7567 Controller haben.

Bei nochmaligen Tests viel auf, dass die Pixelstreifen zu den Seiten hin sehr verblassen, weshalb ich auch einen Schaden am Bauteil nicht mehr ausschließe. 

Ich setze mich voraussichtlich Anfang der nächsten Woche nochmals daran und melde mich dann ggf nochmals. 

Danke!

Grüße

olikraus, 2018-07-19:
Also, wenn ich beispielsweise dieses hier nehme, scheint mir alles plausibel:
https://www.ebay.com/itm/1-2-inch-96x64-LCD-Display-COG-Module-Serial-SPI-8-bit-Parallel-Blue-Backlight/113026785708

Welches ist denn dein konkretes display? Gibt es einen Link?

olikraus, 2018-07-20:
Wow, das ist wirklich wenig. Da hilft wohl nur ausprobieren...

olikraus, 2018-08-04:
nun ja, ich werde den punkt mal schließen... Wenn es noch was gibt, gerne hier posten...

closed 2018-08-04 by olikraus