olikraus/u8g2 #652: max7912 big fonts and table

wchpikus, 2018-07-13:
hello
is it possible to add driver for max7912, but connected to highest display?
e.g. 24x32 normal is 8x32.
I thing about send one part of fonts to display 1 and second part do x further from display one.
if all display will be connected like this (16 displays long and 4 height), next part will be start upper 128 pixel)
1st counted from down matrix left to right, 2'th matrix upper first, 3'th upper 2'th ect..
------3'th----->
|
<----------------
                       |
------2'th----->
|
<----------------
                       |
-----1st------->

olikraus, 2018-07-13:
You could simply use three different u8g2 objects. Connect the led matrix to three different IO lines and use SW SPI.

wchpikus, 2018-07-15:
I have some question..
Function nextpage=clear lcd?
Clearbuffer+sendbuffer also?

olikraus, 2018-07-15:
U8g2 does not use dynamic memory, so RAM es never allocated or freed.
first/next page is more portable and will work with 1, 2 and F constructors.
clearbuffer/sendbuffer will work only with F constructors, see also the manual for more details.

wchpikus, 2018-07-16:
Hi
Next question..
Is it possible to scroll text using u82g.print?
I have to use this print becouse i have one string with variable text inside and i can not using u2gt.sendstr..error with compiling.
String test =""; and when i use u2gt.sendStr(0,0 test); i have error...so i using print

olikraus, 2018-07-16:
Using print is fine. Just change the x position by one and rebuild the complete picture with print.

closed 2018-08-04 by olikraus