olikraus/u8g2 #653: SSD 1327 I2C

fmarzocca, 2018-07-18:
Hi,

I am buying a 128x128 OLED SSD1327 I2C, but I can't find it in your ampatibility list.
Which contructor line should I use?

Thanks

morsisko, 2018-07-18:
Don't know if I'm right person, but there are already 2 entries on the supported controllers list 

- SSD1327 EA_W128128
- SSD1327 MIDAS_128X128

both of them support the I2C.
The list can be found here: [Click](https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#ssd1327-midas_128x128)
So you might try for examply this line: 
`U8G2_SSD1327_EA_W128128_F_SW_I2C(rotation, clock, data [, reset])`

fmarzocca, 2018-07-18:
Hi,

yes I was wrongly watching only at the constructor list in the examples! Thank you I will test those.
Fabio

closed 2018-07-18 by fmarzocca