olikraus/u8g2 #654: XV0, V0 and VG in the ST7567 controlled LCD

morsisko, 2018-07-18:
Hi, I know that it probably isn't best place for questions like that, but I have found that in the [Gallery](https://github.com/olikraus/u8g2/wiki/gallery#19-may-2018--st7657-transparent-lcd-enh-dg128064) someone has got an LCD with similar pinout, i mean with the XV0, V0 and VG, however he writed only about the capacitors, but not about the voltages I should connect to those pins. So there is my question, can I connect it to raw 3.3V? Because the datasheet doesn't specify it, or I just can't see that, it also says that the Vop is typically 10V, which i also don't understand, because well the LCD is powered via 3.3V, isn't it?

And the last thing I don't understand is to how can i know the correct resistor value if the forward voltage isn't specified in the datasheet?

LCD Model: RX12864H
LCD Datasheet: [Click](https://datasheet.octopart.com/RX12864H-BIW-Raystar-datasheet-17800594.pdf)

Thanks in advance.

olikraus, 2018-07-18:
For the transparent LCD on the gallery page, I mentioned the 3.0V LDO. In fact, the this LCD had required a 3.0V supply. In your datasheed, 3.3V are mentioned on chapter 4 and 9.

> the Vop is typically 10V,

Yes, but this can be ignored. The LCD will generate this voltage internally (with the help of the two caps).

>  how can i know the correct resistor value if the forward voltage

You mean for the backlight? I think it is a documentation error. You could assume that the specified supply voltage is the forward voltage. 3.5V sounds good as forward voltage.

morsisko, 2018-07-19:
Ok, thanks for the answer, so to be sure, the only thing I should connect to the XV0 and V0 pins is the capacitor with the + leg at XV0 and - leg at V0(because V0 is the negative frame)? And also if I would hope the forward voltage for the backlight is ~3.3V, and I would supply just the 3.3V can I connect it without any resistor?

olikraus, 2018-07-19:
You need two caps and you should follow the instructions of page 7 from your datasheet.
Do NOT use electrolyt caps, instead use ceramic (multilayer) caps.

> I connect it without any resistor?

No, you must use a resistor to limit the current.

morsisko, 2018-08-02:
Thanks, everything seems to work fine (except the backlight, I didn't test it yet because it looks fine without it). Already I tested it only for the SPI. The only constructor that works fine for me is U8G2_ST7567_JLX12864, others with the same controller gives me strage results or shifted x/y pixels. I use the setContrast with 190 as parameter.

[image]

olikraus, 2018-08-02:
Yes, looks nice already. Thanks for sharing the picture.

> U8G2_ST7567_JLX12864, others with the same controller gives me strage results or shifted x/y pixels

Of course, each display has its own characteristics how the mapping between the controller RAM and the display pixels is done.

closed 2018-08-04 by olikraus

Ajunehra, 2023-02-21:
To display character only, does we need to connect VG, XV0, V0 pins?

morsisko, 2023-02-21:
Yes, I think so

olikraus, 2023-02-21:
The need to be connected to a 0.1 to 1uF cap .