olikraus/u8g2 #655: SH1106 no Ack

LoaDy588, 2018-07-18:
Hello,
I'm trying to run a SH1106 128x64 display from Raystar Optronics on custom ATTiny1617 board. The Display is set to I2C mode, SDA/SCL are connected to the MCU. Reset/A0/CS are also connected.
The Display doesn't acknowledge/react to any I2C communication. I tried manually setting the A0/CS/Reset lines from MCU with no effect. Is there any glaring mistake in the code below or should I start looking at hardware/PCB design error?

```c
#define F_CPU 1000000

#include <avr/io.h>
#include <util/delay.h>
#include <u8g2.h>

u8g2_t u8g2;

void init(void)
{
	//clock at 1.0MHz
	CCP = 0xD8;  //enable protected register write for next 4 instructions
	CLKCTRL.MCLKCTRLB = 0b00000111;

	//all ports input and pull up -> save power
	for (uint8_t i = 0; i < 8; i++) {
		*((uint8_t *)&PORTA + 0x10 + i) |= 1 << PORT_PULLUPEN_bp;
	}

	for (uint8_t i = 0; i < 8; i++) {
		*((uint8_t *)&PORTB + 0x10 + i) |= 1 << PORT_PULLUPEN_bp;
	}

	for (uint8_t i = 0; i < 8; i++) {
		*((uint8_t *)&PORTC + 0x10 + i) |= 1 << PORT_PULLUPEN_bp;
	}

	PORTB.DIRSET = PIN2_bm;
	PORTB.OUTCLR = PIN2_bm;

	PORTB.DIRCLR = PIN3_bm;
	PORTB_PIN3CTRL &= ~PORT_PULLUPEN_bm;

	PORTB.DIRSET = PIN4_bm | PIN5_bm | PIN6_bm | PIN7_bm | PIN0_bm | PIN1_bm;

	PORTB.OUTCLR = PIN4_bm | PIN5_bm;
	PORTB.OUTSET = PIN6_bm;
}

void twi_init(void)
{
	TWI0.CTRLA = TWI_SDASETUP_4CYC_gc |TWI_SDAHOLD_OFF_gc | 0 << TWI_FMPEN_bm;

	TWI0.MCTRLA = 0 << TWI_ENABLE_bp
	| 0 << TWI_RIEN_bp
	| 0 << TWI_WIEN_bp
	| 0 << TWI_QCEN_bp
	| 0 <<TWI_SMEN_bp
	| TWI_TIMEOUT_DISABLED_gc;

	TWI0.MBAUD = 1;  //max frequency of TWI is 1/10 of F_CPU

	TWI0.MCTRLA |= TWI_ENABLE_bm;
}

unsigned int i2c_master_write(uint8_t slave, uint8_t *data, uint8_t count)
{
	while ((TWI0.MSTATUS & TWI_BUSSTATE_gm) != TWI_BUSSTATE_IDLE_gc);
	TWI0.MADDR = slave;								//Request for write when the bus is idle
	while (!(TWI0.MSTATUS & TWI_WIF_bm));
	if (TWI0.MSTATUS & (TWI_RXACK_bm | TWI_BUSERR_bm | TWI_ARBLOST_bm)) {    //check for errors/NACK
		TWI0.MCTRLB |= TWI_MCMD_STOP_gc;
		return 1;
	}
	for (uint8_t i = 0; i < count; i++) {
		TWI0.MDATA = data[i];									//Writing data
		TWI0.MCTRLB &= ~TWI_ACKACT_bm;
		while (!(TWI0.MSTATUS & TWI_WIF_bm));
		if (TWI0.MSTATUS & (TWI_RXACK_bm | TWI_BUSERR_bm | TWI_ARBLOST_bm)) {   //check for errors/NACK
			TWI0.MCTRLB |= TWI_MCMD_STOP_gc;
			return 1;
		}
	}
	TWI0.MCTRLB |= TWI_MCMD_STOP_gc;					//Execute ACK succeeded by issuing a STOP
	return 0;
}

uint8_t disp_i2c_cb(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr) {
	uint8_t *data = (uint8_t *)arg_ptr;
	switch(msg)
	{
		case U8X8_MSG_BYTE_SEND:
		break;
		case U8X8_MSG_BYTE_INIT:
		break;
		case U8X8_MSG_BYTE_SET_DC:
		break;
		case U8X8_MSG_BYTE_START_TRANSFER:
		break;
		case U8X8_MSG_BYTE_END_TRANSFER:

		if ((i2c_master_write(u8x8_GetI2CAddress(u8x8), data, arg_int)) == 1) {
			return 0;
		}
		break;
		default:
		return 0;
	}
	return 1;
}

uint8_t disp_gpio_cb(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
	switch(msg)
	{
		case U8X8_MSG_GPIO_AND_DELAY_INIT:	// called once during init phase of u8g2/u8x8
		break;							// can be used to setup pins
		case U8X8_MSG_DELAY_NANO:			// delay arg_int * 1 nano second
		break;				//  1/F_CPU is 1uS, no need for nano second delays
		case U8X8_MSG_DELAY_100NANO:		// delay arg_int * 100 nano seconds
		break;
		case U8X8_MSG_DELAY_10MICRO:		// delay arg_int * 10 micro seconds
		for (uint8_t i = arg_int; i > 0; i--) {
			_delay_us(10);
		}
		break;
		case U8X8_MSG_DELAY_MILLI:			// delay arg_int * 1 milli second
		for (uint8_t i = arg_int; i > 0; i--) {
			_delay_ms(1);
		}
		break;
		case U8X8_MSG_GPIO_CS:				// CS (chip select) pin: Output level in arg_int
		if (arg_int == 1) {
			PORTB.OUTSET |= PIN5_bm;
			} else {
			PORTB.OUTCLR |= PIN5_bm;
		}
		break;
		case U8X8_MSG_GPIO_DC:				// DC (data/cmd, A0, register select) pin: Output level in arg_int
		if (arg_int == 1) {
			PORTB.OUTSET |= PIN4_bm;
			} else {
			PORTB.OUTCLR |= PIN4_bm;
		}
		break;
		case U8X8_MSG_GPIO_RESET:			// Reset pin: Output level in arg_int
		if (arg_int == 1) {
			PORTB.OUTSET |= PIN6_bm;
		} else {
			PORTB.OUTCLR |= PIN6_bm;
		}
		break;
		u8x8_SetGPIOResult(u8x8, 1);			// default return value
		break;
	}
	return 1;
}

int main(void)
{
	init();
	twi_init();
	TWI0.MSTATUS |= TWI_BUSSTATE_IDLE_gc;

	u8g2_Setup_sh1106_i2c_128x64_noname_f(&u8g2, U8G2_R0, disp_i2c_cb, disp_gpio_cb);
	u8g2_InitDisplay(&u8g2);
	u8g2_SetPowerSave(&u8g2, 0);

	while (1)
	{
		u8g2_ClearBuffer(&u8g2);
		u8g2_SetFont(&u8g2, u8g2_font_ncenB14_tr);
		u8g2_DrawStr(&u8g2, 0,15, "Hello World!");
		u8g2_SendBuffer(&u8g2);

		_delay_ms(200);
	}

}

```

olikraus, 2018-07-18:
There is at least one problem and the might be one more problem.

1. The disp_i2c_cb is implemented wrongly.
```
uint8_t u8x8_byte_i2c(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
  static uint8_t buffer[32];		/* u8g2/u8x8 will never send more than 32 bytes between START_TRANSFER and END_TRANSFER */
  static uint8_t buf_idx;
  uint8_t *data;
 
  switch(msg)
  {
    case U8X8_MSG_BYTE_SEND:
      data = (uint8_t *)arg_ptr;      
      while( arg_int > 0 )
      {
	buffer[buf_idx++] = *data;
	data++;
	arg_int--;
      }      
      break;
    case U8X8_MSG_BYTE_INIT:
      /* add your custom code to init i2c subsystem */
      break;
    case U8X8_MSG_BYTE_SET_DC:
      /* ignored for i2c */
      break;
    case U8X8_MSG_BYTE_START_TRANSFER:
      buf_idx = 0;
      break;
    case U8X8_MSG_BYTE_END_TRANSFER:

		if ((i2c_master_write(u8x8_GetI2CAddress(u8x8), data, buf_idx)) == 1) {
			return 0;
		}
      break;
    default:
      return 0;
  }
  return 1;
}
```
Whether you can use u8x8_GetI2CAddress(u8x8) or whether to use u8x8_GetI2CAddress(u8x8)/2 depends on your implementation if i2c_master_write (i have not checked this)

2. If the display really does not support ACK, then hardware I2C will NOT work at all. In this case you have to implement software I2c has discussed here:
https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform#the-uc-specific-gpio-and-delay-callback

--> u8x8_byte_sw_i2c

LoaDy588, 2018-07-19:
Thanks for the help.

The I2C hardware on ATTiny1617 can be forced to push data onto the bus even when the slave doesn't Ack, so with the fixed disp_i2c_cb function, Im getting pretty meaningful looking data(apart from the init data, which look to be just bunch of zeroes). The display still doesn't work. Tried another display with same results.

Currently working on the SW i2c version, so far getting only garbage data out, I need to fix the timing and pin setup.(1MHz clock speed might not be enough for this?).

Will report back with results from SW i2c version.

olikraus, 2018-07-19:
I think there is no lower frequency limit on I2C. In fact it must not be faster than 400KHz for fast I2C and it should be only 100KHz for normal I2C.

Remember to implement the I2C delay message in the u8x8 gpio and delay callback for software I2C.

LoaDy588, 2018-07-22:
SW version of I2C doesn't work too, the output on the bus is identical to HW version. Found some errors on the PCB design, so I have to make new revision for the PCB, will try switching to 8080 protocol.

closed 2018-07-22 by LoaDy588