olikraus/u8g2 #657: How to drive nice 64x32 LCD with ST7567Ai  (I2C)

Sebart, 2018-07-18:
That nice very small LCD have ST7567Ai which is internaly set to use only I2C  (0x3F address)  

[http://www.artronic.com.pl/o_produkcie.php?id=1532?](http://www.artronic.com.pl/o_produkcie.php?id=1532?)

How to modify one of constructors to use Your librarry for that display ?

olikraus, 2018-07-18:
Did you try one of the existing ST7567 constructors? Do you see something? You may need to change the contrast (setContrast). If there is any existing constructor, which shows up something, then this constructor would be a good starting point.

olikraus, 2018-07-19:
> There is no I2C constructor for ST7567. 

Ah, yes, I have to check this....

olikraus, 2018-07-19:
ok, there is no I2C, but I do not remember why I have not implemented this. Needs to be analyzed.

Sebart, 2018-07-19:
I drive some test today

[image]

olikraus, 2018-07-20:
nice, looks good already

olikraus, 2018-08-04:
The I2C protocol seems to be the same as ST7588, which is the same as SSD1306: `u8x8_cad_ssd13xx_i2c`.
Unfortunately the documentation is not really clear about the protocol.

olikraus, 2018-08-04:
The constructor is U8G2_ST7567_64X32_...
I have created version U8g2 2.23.14:
You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

olikraus, 2018-08-04:
Can you test the new device? Feedback would be great. Thanks!

olikraus, 2018-08-04:
New constructor: 
```
U8G2_SSD1317_96X96_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);  
```

olikraus, 2018-08-04:
I have created beta 2.23.15. Can you do some tests? Especially I am interessted in the output of the flip mode examples.

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

Sebart, 2018-08-06:
no success (nothing displayed) with U8G2_ST7567_64X32_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE) on Arduino Pro Mini.

olikraus, 2018-08-06:
oh... are you able to see any activity on the I2C bus?

olikraus, 2018-08-23:
any feedback on this?

olikraus, 2018-08-24:
Closing this for now, please comment of possible.
Please reopen if required.

closed 2018-08-24 by olikraus