olikraus/u8g2 #658: Start up time

fmarzocca, 2018-07-19:
Hi,

I am using a display in the pagination mode. I noticed that upon switching on the Arduino, the display remains black for a couple of seconds before lighting up. Isn’t there a way to speed up this time or to display a logo during the startup?

JamesGKent, 2018-07-26:
the way arduinos are programmed means they spend a short time in "bootloader" mode on initial power up, where it listens for a connection from a PC to program it, as far as i'm aware the only ways around this would be to customise and burn a modified version of the bootloader where the wait time is reduced (may make it difficult to reprogram) or there is an option in the arduino IDE to compile without the bootloader, meaning the MCU will boot straight into your code, however both of these methods require the use of an external programmer through the ICSP header. the second method also means you can no longer program it the "normal" way until you burn a bootloader back in.

fmarzocca, 2018-07-26:
James, thanks for answering.

closed 2018-07-26 by fmarzocca

JamesGKent, 2018-07-26:
you may want to have a read of this thread:
https://forum.arduino.cc/index.php?topic=470086.0

and there are a fair few similar threads, turns out the optiboot bootloader is much better than the standard one in this regard